genodelabs/genode #4931: pc_nic_drv: initialization of e1000 on Qemu takes very long sometimes - DDE timing bug?

chelmuth, 2023-06-23:
This effect can be experienced in roughly 50 percent of runs of `make run/lwip KERNEL=pistachio`. I traced it down to a unreasonably slow EEPROM checksumming in *drivers/net/ethernet/intel/e1000/e1000_hw.c*. The following two log snippets reflect the timing behavior in a good and a base case, where the latter looks quite strange: The large numbers are `ktime_get()` and `jiffies_64` and should (as I understand it) change synchronously.

good case
```
[2023-06-23 07:45:08.164443] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 31 1
[2023-06-23 07:45:08.167219] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 1199269000 4294892594
[2023-06-23 07:45:08.170625] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935 
[2023-06-23 07:45:08.177896] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938 
[2023-06-23 07:45:08.182486] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944 
[2023-06-23 07:45:08.186936] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 1218742000 4294892600
[2023-06-23 07:45:08.190346] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 1221912000 4294892601
[2023-06-23 07:45:08.193012] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 706
[2023-06-23 07:45:08.195439] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 32 1
[2023-06-23 07:45:08.200355] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 1229371000 4294892602
[2023-06-23 07:45:08.207880] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935 
[2023-06-23 07:45:08.211598] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938 
[2023-06-23 07:45:08.218694] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944 
[2023-06-23 07:45:08.225383] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 1253994000 4294892609
[2023-06-23 07:45:08.228763] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 1259674000 4294892610
[2023-06-23 07:45:08.231803] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 1008
[2023-06-23 07:45:08.233961] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 33 1
[2023-06-23 07:45:08.236824] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 1267186000 4294892611
[2023-06-23 07:45:08.240183] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935 
[2023-06-23 07:45:08.243775] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938 
[2023-06-23 07:45:08.255691] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944 
[2023-06-23 07:45:08.259296] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 1289251000 4294892618
[2023-06-23 07:45:08.261643] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 1291794000 4294892618
[2023-06-23 07:45:08.264536] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 0
[2023-06-23 07:45:08.266638] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 34 1
[2023-06-23 07:45:08.269403] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 1299209000 4294892619
[2023-06-23 07:45:08.275457] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935 
[2023-06-23 07:45:08.279313] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938 
[2023-06-23 07:45:08.284043] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944 
[2023-06-23 07:45:08.287315] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 1316865000 4294892625
[2023-06-23 07:45:08.289992] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 1319410000 4294892625
[2023-06-23 07:45:08.292985] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 f04
[2023-06-23 07:45:08.295198] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 35 1
[2023-06-23 07:45:08.297762] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 1327178000 4294892626
[2023-06-23 07:45:08.300985] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935 
[2023-06-23 07:45:08.308551] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938 
[2023-06-23 07:45:08.312943] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944 
[2023-06-23 07:45:08.316367] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 1345358000 4294892632
[2023-06-23 07:45:08.318942] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 1347916000 4294892632
[2023-06-23 07:45:08.322107] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 7fff
```

bad case
```
[2023-06-23 08:01:20.533526] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 31 1
[2023-06-23 08:01:20.536048] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 26877699000 4294899010
[2023-06-23 08:01:20.746506] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935
[2023-06-23 08:01:21.125932] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938
[2023-06-23 08:01:21.730569] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944
[2023-06-23 08:01:21.805401] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 27858669000 4294899258
[2023-06-23 08:01:21.810167] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 27858669000 4294899260
[2023-06-23 08:01:21.850419] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 706
[2023-06-23 08:01:21.852638] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 32 1
[2023-06-23 08:01:21.855518] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 27887464000 4294899267
[2023-06-23 08:01:22.014500] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935
[2023-06-23 08:01:22.359711] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938
[2023-06-23 08:01:22.964662] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944
[2023-06-23 08:01:23.009783] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 28746034000 4294899482
[2023-06-23 08:01:23.012957] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 28746034000 4294899482
[2023-06-23 08:01:23.033714] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 1008
[2023-06-23 08:01:23.035864] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 33 1
[2023-06-23 08:01:23.038911] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 28763530000 4294899486
[2023-06-23 08:01:23.227952] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935
[2023-06-23 08:01:23.518553] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938
[2023-06-23 08:01:24.117684] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944
[2023-06-23 08:01:24.219687] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 29630771000 4294899697
[2023-06-23 08:01:24.225471] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 29630771000 4294899703  <-- 0 ktime but 6 jiffies?
[2023-06-23 08:01:24.240740] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 0
[2023-06-23 08:01:24.242841] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 34 1
[2023-06-23 08:01:24.245646] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 29641783000 4294899705
[2023-06-23 08:01:24.436197] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935
[2023-06-23 08:01:24.751458] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938
[2023-06-23 08:01:25.394659] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944
[2023-06-23 08:01:25.463659] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 30555295000 4294899934
[2023-06-23 08:01:25.466408] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 30555295000 4294899934
[2023-06-23 08:01:25.492637] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 f04
[2023-06-23 08:01:25.494575] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3853 35 1
[2023-06-23 08:01:25.499998] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3930 30581101000 4294899939
[2023-06-23 08:01:25.640758] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3935
[2023-06-23 08:01:25.986469] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3938
[2023-06-23 08:01:26.586230] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3944
[2023-06-23 08:01:26.663567] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3946 31427929000 4294900152
[2023-06-23 08:01:26.666863] [init -> drivers -> nic_drv] ---- e1000_do_read_eeprom:3948 31427929000 4294900152
[2023-06-23 08:01:26.689596] [init -> drivers -> nic_drv] ---- e1000_read_eeprom:3857 7fff
```

m-stein, 2023-07-11:
This seems to affect also the ping run scripts. When it happens, the driver gets stuck at this point:
```
[init -> drivers -> nic_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256
[init -> drivers -> nic_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526s
[init -> drivers -> nic_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 764504178510000s
[init -> drivers -> nic_drv] time-clocksource: Switched to clocksource dde_counter
[init -> drivers -> nic_drv] e1000-e1000_main: Intel(R) PRO/1000 Network Driver
[init -> drivers -> nic_drv] e1000-e1000_main: Copyright (c) 1999-2006 Intel Corporation.
[init -> drivers -> nic_drv] e1000 00:02.0 eth0: (PCI:33MHz:32-bit) 52:54:00:12:34:56
[init -> drivers -> nic_drv] e1000 00:02.0 eth0: Intel(R) PRO/1000 Network Connection
[init -> drivers -> nic_drv] e1000e-netdev: Intel(R) PRO/1000 Network Driver
[init -> drivers -> nic_drv] e1000e-netdev: Copyright(c) 1999 - 2015 Intel Corporation.
```

It seems to appear more often with background CPU load.

chelmuth, 2023-07-12:
> This seems to affect also the ping run scripts. When it happens, the driver gets stuck at this point:

Did you really really check that the driver *got stuck*? In the cases I describe above it just took about 30s to finish the EEPROM checksum but finished initialization afterwards and kind of worked.

m-stein, 2023-07-12:
Sorry for my rash analysis! The driver indeed continues, but after about 3 minutes in my case. So, if I raise the test timeout accordingly and apply 66c268a2bb, the test succeeds.

ssumpf, 2023-07-20:
I have a little more insight. The behavior is caused by `udelay`. The EEPROM checksum verification requires a lot of bit bang operations with many calls to `udelay(50)`. These calls end up in `__udelay` (_repos/dde_linux/src/lib/lx_emul/shadow/lib/delay.c_). Where `lx_emul_time_counter` is called - which calls `curr_time` on Genode's timer session and returns us - until 50 us have passed. In the good case this looks like this when the body of the `while` loop is executed:

```
current: 949231 end: 949280 
current: 950124 end: 950173 
```
note the values are close together and updated nicely.

The bad case:
```
current: 690084 end: 690134
... many more
current: 690084 end: 690134
current: 721676 end: 721726
... many more
current: 721676 end: 721726
current: 752028 end: 752078
```

So `curr_time` is updated every 20 to 30ms which explains the 6 jiffies adjusted by `cpu_relax`. This of course leads to an immense amount of delay caused by `udelay(50)` which has become `udelay(25000)`. So what is wrong here, as I gather the timer connection goes directly to the pit implementation (interpolation seems not to be involved here)? Wrong timer calibration or missing IRQ in the Qemu case?

This requires more investigation.

ssumpf, 2023-07-20:
Update: Okay, interpolation is involved. The problem occurs in `Timer::Connection::curr_time()` (_repos/base/src/lib/timeout/timer_connection_time.cc_). https://github.com/genodelabs/genode/blob/7aa301361d1717c032a1f5afae039c4cc64eb112/repos/base/src/lib/timeout/timer_connection_time.cc#L162 

When `_interpolation_quality == MAX_INTERPOLATION_QUALITY` the delay does **not** occur.

Otherwise, `elapsed_us` is called, which calls `curr_time` within the timer component. In case of the PIT the `_curr_time_us ` is only updated on an IRQ when a timeout is programmed. This does not seem to happen often and the delay **does** occur. Why we hit this path in the bad is case remains to be investigated.

ssumpf, 2023-07-21:
Last note: If I update `_curr_time_us` in the PIT implementation on every call, everything is working as expected. The commit that changed the behavior is 80778b267dfb8a88b0ce547c074b372915c52f22 in issue genodelabs/genode#2579. 
It looks like there is a conflict of interest, first we need precise times for us resolutions, on the other hand, we don't want to call `curr_time` too often.

m-stein, 2023-08-03:
I'd like to sum up yesterdays off-line discussion regarding this topic:

The issue can be broken down to a very simple formula:
There are timer clients (like pc_nic_drv) that expect ``curr_time`` to be us accurate as soon as they come up. However, our timing stack has an initial stage of low accuracy due to having to calibrate interpolation with the local counter. There is no synchronization that prevents the former from running into the latter.

Although not elementary for the described problem, there is an additional issue that different time sources apply different rate limitation strategies: PIT updates time only on IRQs, EPIT only on IRQs under 1000 Hz, NOVA on IRQs **or** under 4000 Hz, others don't apply any. This lowers the accuracy in the startup stage even further.

My test results confirm the observations @ssumpf made for x86_32+qemu+okl4+run/ping. The pc_nic_drv wants to do around 4500 calls to ``__udelay(50)`` before becoming operational. If the driver starts up right away, it often catches the timing startup stage which causes the loop in ``__udelay`` to call ``curr_time`` around 500-1000 times with the same result before before time makes one leap far beyond the desired timeout. The following ouput shows the remaining us for each iteration in ``__udelay`` while ``|`` marks the end of a ``__udelay`' count followed by the call count and the iteration count:
```
[init -> init -> drivers -> nic_drv] NR_IRQS: 2304, nr_irqs: 2304, preallocated irqs: 256
[init -> init -> drivers -> nic_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[init -> init -> drivers -> nic_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[init -> init -> drivers -> nic_drv] time-clocksource: Switched to clocksource dde_counter
[init -> init -> drivers -> nic_drv] e1000-e1000_main: Intel(R) PRO/1000 Network Driver
[init -> init -> drivers -> nic_drv] e1000-e1000_main: Copyright (c) 1999-2006 Intel Corporation.
[init -> init -> drivers -> nic_drv] 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50  
[init -> init -> drivers -> nic_drv] 0 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 55
[init -> init -> drivers -> nic_drv]  50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 500
[init -> init -> drivers -> nic_drv] 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50  
[init -> init -> drivers -> nic_drv] 0 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 55
[init -> init -> drivers -> nic_drv]  50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 500
[init -> init -> drivers -> nic_drv] 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50  
[init -> init -> drivers -> nic_drv] 0 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 55
[init -> init -> drivers -> nic_drv]  50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 500
[init -> init -> drivers -> nic_drv] 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50  
[init -> init -> drivers -> nic_drv] 0 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 55
[init -> init -> drivers -> nic_drv]  50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 500
[init -> init -> drivers -> nic_drv] 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50  
[init -> init -> drivers -> nic_drv] 0 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 55
[init -> init -> drivers -> nic_drv]  0 | 2 1013
[init -> init -> drivers -> nic_drv] 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50 50  
[... cut ...]
[init -> init -> drivers -> nic_drv] 49 48 47 46 44 43 42 41 40 39 38 37 36 35 34 33 32 31 30 29 28 27 26 25 24 23 22 21 20 19 18 17 16 15 14 12 11 10 9 8 7 6 5 4 3 2 1 0 | 4608 47
[init -> init -> drivers -> nic_drv] 49 47 46 45 44 43 42 41 40 39 38 37 36 35 34 33 32 31 30 29 28 27 26 25 24 23 22 21 20 19 18 17 16 15 14 13 12 11 10 9 8 7 6 5 4 3 2 1 0 | 4609 48
[init -> init -> drivers -> nic_drv] e1000-e1000_main: eth0 NIC Link is Up 1000 Mbps Full Duplex, Flow Control: RX
[init -> init -> drivers -> nic_drv] create uplink for net device eth0
[init -> ping_1] 64 bytes from 10.0.2.2: icmp_seq=1 ttl=64 time=3.451 ms
```

m-stein, 2023-08-03:
One point we agreed on is this temporary commit for detecting high accuracy clients accessing low accuracy timing: e4d020601f . This should give us a better picture of the problem. I hope it meets your expectations @chelmuth ?

Another point is that rate limitation should not come in the form of low-level, target-specific solutions. The time source drivers should each only translate simple tasks to hardware, while controlling requests and streamlining results should move one layer up to the generic timer code.

A decision regarding the main issue here, the inaccuracy problem, is still to be made.

chelmuth, 2023-08-08:
The commit produced 1166 log entries in the nightly.

```
> find . -name \*.log | xargs grep -n 'XXXXXXXXXX session' | nl
     1  ./x86_32/x86_32.pc.pistachio.sntp_client.log:161:[2023-08-08 05:09:09] [init -> timer] Warning: XXXXXXXXXX session "sntp_client -> " returned an elapsed time of 145665 us 500 times in a row XXXXXXXXXX
     2  ./x86_32/x86_32.pc.pistachio.sntp_client.log:163:[2023-08-08 05:09:09] [init -> timer] Warning: XXXXXXXXXX session "drivers -> nic_drv -> " returned an elapsed time of 253376 us 500 times in a row XXXXXXXXXX
...
  1165  ./x86_64/x86_64.pc.sel4.ping_nic_router.log:269:[2023-08-08 04:57:44] [init -> timer] Warning: XXXXXXXXXX session "ping_21 -> " returned an elapsed time of 1273465 us 500 times in a row XXXXXXXXXX
  1166  ./x86_64/x86_64.pc.sel4.ping_nic_router.log:270:[2023-08-08 04:57:44] [init -> timer] Warning: XXXXXXXXXX session "ping_23 -> " returned an elapsed time of 1273465 us 500 times in a row XXXXXXXXXX
```