genodelabs/genode #4932: tresor: alignment exceptions / MMU faults

chelmuth, 2023-06-23:
Despite commit b0c673b6a106841ab13afb5894d8e6721de1d795, the *test-file_vault_config_report* still produces CPU exceptions related to unaligned accesses (likely with FPU instructions).

arm_v7a.imx6q_sabrelite.sel4.depot_autopilot
```
[2023-06-23 07:15:49] [34mWarning: alignment fault, addr=0xa765 ip=0x10f814c4[0m
```

arm_v7a.imx53_qsb.hw.depot_autopilot.log
```
[2023-06-23 06:38:42] [32mKernel: init -> dynamic -> test-file_vault_config_report -> file_vault -> tresor_trust_anchor_vfs -> ep raised unhandled MMU fault ip=0x10f814c4 fault-addr=0xa765 type=unknown[0m
```

arm_v7a.imx7d_sabre.hw.depot_autopilot.log
```
[2023-06-23 07:16:49] [32mKernel: init -> dynamic -> test-file_vault_config_report -> file_vault -> tresor_trust_anchor_vfs -> ep raised unhandled MMU fault ip=0x10f814c4 fault-addr=0xa765 type=unknown[0m
```

chelmuth, 2023-06-23:
> arm_v7a.imx6q_sabrelite.sel4.depot_autopilot
> 
> ```
> [2023-06-23 07:15:49] �[34mWarning: alignment fault, addr=0xa765 ip=0x10f814c4�[0m
> ```

Happens also on base-hw and in *gems/src/lib/vfs/tresor_trust_anchor/aes_256.cc:205*.

m-stein, 2023-06-28:
@chelmuth The commits c3190aca1b, 0594a55be5 are two different approaches to solving the AES256 issue. I myself prefer the latter (0594a55be5) but first came up with the former. The former should theoretically be less performant. The latter fits well with my plans to move the different key buffer types to the AES256 header and use them directly as arguments to the AES256 functions. I would do this once I clean up the TA VFS plugin in general. What do you think?

commit 7b02b91f (2023-06-29): vfs/tresor_trust_anchor: fix alignment faults

Fixes alignment faults that occured in the AES256 implementations while
wrapping or unwrapping keys on imx53_qsb, imx6q_sabrelite, and imx7d_sabre.
The problem was that the unwrap_key/wrap_key functions did reinterpret casts
from unsigned char pointers to uint64_t pointers and then directly used the 64
bit values of referenced by the latter. Most probably this caused the compiler
to optimize operations in the assumption that the pointer is 8-byte aligned
which then created alignment faults.

As a solution, this commit changes the interface of the  [... cut]
closed 2023-07-14 by nfeske