genodelabs/genode #4947: monitor: add memory write support

cproc, 2023-06-29:


cproc, 2023-06-29:
Commit ca85abd implements the feature, but currently a lot of the `Memory_accessor` code is duplicated. What would be a good way to reduce the code duplication?

nfeske, 2023-06-29:
I'd leave the public monitor.h header as sparse as possible. Since the new `Gdb_binary_buffer` is only used by the test (and it is relatively simple anyway) I'd leave it local to the test.

> What would be a good way to reduce the code duplication?

Two ideas come to mind:

* The update of the `_curr_view` could go to a new method `_with_curr_view_at(Virt_addr at, auto const &fn)` that first updates the `_curr_view` and - if that succeeded - calls `fn` with a reference to the updated `curr_view`. The read and write methods would each call this method.

* Moving the watchdog and pagefault-related code from the Memory_accessor's `read` and `write` methods to a new method `_with_watched_pagefaults(auto const &fn)` like this (just sketched):
  ```
  void _with_watched_pagefaults(auto const &fn)
  {
    /* give up after 100 milliseconds */
    _watchdog_timer.trigger_once(1000*100);

    /* drain pending signals to avoid spurious watchdog timeouts */
    while (_env.ep().dispatch_pending_io_signal());

    unsigned const orig_page_fault_count = pd.page_fault_count();
    unsigned const orig_timeout_count    = _timeout_count;

    if (!_probe.constructed())
      _probe.construct(_env, _probe_response_handler);

    auto fault_or_timeout_occurred = [&] {
      return (orig_page_fault_count != pd.page_fault_count())
          || (orig_timeout_count != _timeout_count); };

    auto block_fn = [&]
    {
      if (fault_or_timeout_occurred())
        return false; /* cancel read or write */

      _env.ep().wait_and_dispatch_one_io_signal();
        return true; /* keep trying */
    };

    fn(block_fn);

    /* wind down the faulted probe thread, spawn a fresh one on next call */
    if (fault_or_timeout_occurred())
      _probe.destruct();
  }
  ```
  The `read` and `write` methods would then call this function with the read/write-specific part as lambda argument. The lambda function gets the `auto const &block_fn` as argument.

cproc, 2023-06-29:
Thanks, I applied your suggestions in commit 5dffe79.

nfeske, 2023-06-29:
Very nice! I merged the new version to staging.

closed 2023-07-14 by nfeske