genodelabs/genode #4951: vfs_pipe: write fails when RTC is not configured (corner case)

ttcoder, 2023-06-30:
@norman New ticket, as per today's conversation on Genode-Users (i.e. your "spidey sense" tells you it seems this might actually be the sign of something important lurking, and might deserve investigating ;-) :

To be clear, the bug is not about a freeze or I/O stall (my email did discuss a couple issues/freezes/stalls, and things of this nature re. my NTFS plug-in, but I didn't see any freeze or deadlocks with the vfs_pipe plug-in). The observed symptom for vfs_pipe is that write() always fails (it always returns something like -1).

The failure to write occurs in a component called "broker" (in charge of doing 'BMessage' IPC between applications), and other components, for instance (below) the Deskbar.

## Pointers to relevant code

If you'd rather examine the code before trying to replicate the bug, here's where the write() occurs in broker:

- go to https://chiselapp.com/user/ttcoder/repository/genode-haiku/file?udc=1&ln=on&ci=91b37516e8741dae481d0e4087c9c78c9cf96870&name=genode-haiku%2Fhog-services%2Fbroker%2Fbroker_server.cpp
- look for  writeport_chunk()
- look for `sent = write( tx, accu->Buffer(), accu->BufferLength() );`
- tracing occurs a few lines later in the method (remove the "if( debug )" to make it unconditionnal, exhibiting that sent==-1)

And the config(urator):

- https://chiselapp.com/user/ttcoder/repository/genode-haiku/file?name=genode-haiku/hai-src/apps/deskbar/_GeJamfile.inc&ci=91b37516e8741dae481d0e4087c9c78c9cf96870
- look for the comment that goes like this : "make sure to have rtc (...).."
- remove the "Rtc" part to repro, or leave it in to check that the code otherwise works fine

## Repro steps

This can be replicated on Genode 23.02 with my (in)famous HoG library at least :-)... Using a 'simplified' test scenario called "run deskbar"... Though I understand that's still quite far from a minimal repro case! Anyway:

- [chiselapp's Setup:Main ](https://chiselapp.com/user/ttcoder/repository/genode-haiku/wiki?name=Setup:Main) : play along, but stop *before* the "jam hog-demos.run" step, so as to do the below:
- edit the jam config file mentionned previously (deskbar jamfile) to *remove* the Rtc config, which should be enough to reproduce the bug
- execute "Deskbar_FULL_TEST=1  jam hog-deskbar.emu8"

I might also meet you "half way" and provide more data (generated config file, Log generated from running the scenario in qemu myself to check I got all steps right, etc) if you'd rather not build the whole thing and  look for a shortcut instead, which makes sense since it might take an hour to set it all up and reach the "jam.. deskbar..." step.

I seem to remember trying out "jam hog-broker.run", a simplier test case, but it didn't trigger the bug, because it does not actually perform cross-component BMessage IPC, only local "IPC". I can check again if that helps.

nfeske, 2023-07-03:
Thank you for opening the issue.

> The observed symptom for vfs_pipe is that write() always fails (it always returns something like -1).

The `errno` value might be interesting to know.

I presume that the observed symptom is related to either a missing call of `wakeup_vfs_user` (when I/O happened, the VFS user must be woken up to pick up the new state), or a missing call of `wakeup_remote_peers` (whenever the application induces new I/O, the remote peer must be informed to pick it up). In previous versions, the submission/occurrence of I/O with the application code was implicitly coupled whereas 23.02 introduced the dedicated calls.

After version 23.02, a few follow-up changes entered the master branch that might be worth trying to backport to your version. In particular, are you aware of commit https://github.com/genodelabs/genode/commit/c7c95422629982f1fe7fcb9342b2b47a6d241534?

ttcoder, 2023-07-04:
Well that commit is spot-on !

First, I should mention that the observed symptom (now that I'm actually back on my laptop rather than talking "from memory") is/was an fs stall, rather than a -1 status with errno. The very first write() from broker to registrar never completed. Hence, the next step after this ticket is going to be very interesting : I'll see if the fix also fixes my sporadic vfs stalls.

Anyway yep, the stalls cannot be triggered any more now : after applying that commit I launched dual "deskbar" tests again (with and without the RTC config), and now they both work as they should.

And my "jam t8" test-suite is still green across the board. The next few days are going to be interesting. (but, I should improve my "process", the way I follow Genode's timeline/commits, as I had missed the importance of that commit or even missed it altogether).

Closing as duplicate/alrealdy-fixed

ttcoder, 2023-07-04:
I did click "close with comment" but WebPositive missed the boat again, so back on FireFox to _really_ close the ticket.

closed 2023-07-04 by ttcoder

nfeske, 2023-07-05:
I'm relieved. ;-) Thank you Cedric for the positive feedback.