genodelabs/genode #4954: x86_64/hw ping: latency regression

m-stein, 2023-07-03:
When running a ping behind nic_router and drivers_nic-pc with current staging (f71511cbef9dd1637aa9cdc684d36a28bfe95030) I'm seeing an average latency of around 500 ms on both hardware and qemu. Curiously, the actual value significantly oscillates on hardware (10 ms - 1000 ms) while on qemu it doesn't do that.

When running the same setup on current genodelabs/master (79e262921e554baf565156b504ead70782913deb), I get an average latency of around 0.3 ms on hardware and around 4 ms on qemu.

m-stein, 2023-07-03:
I created a small measurement series with my setup:
```
 qemu     hw commit
650ms  500ms f71511cbef9 fixup "lx_kit: support multiple pending IRQs" (virt_linux)
650ms  500ms f550108d880 lx_kit: decouple schedule execution
  6ms 0.37ms dd5fdc70503 intel_fb: disable encoders of disconnected ports
  7ms 0.39ms 838bfc77e53 Replace ipxe_nic_drv by pc_nic_drv in run scripts
  4ms 0.32ms 9fabe7f4c2f pc_nic_drv: ethernet NIC driver based on Linux 6.1
  4ms 0.32ms 2b5fe1d567c register framework: read bitfields to fitting uint
```
The introduction of the new nic driver seems to have decreased latency a bit already but the lx-kit change raises the numbers significantly.

m-stein, 2023-07-03:
The same goes for other x86_64 platforms (hardware):
```
          master   staging
nova      <1ms     501.122ms  (note that the second value was *very* steady)
sel4      <1ms     500ms      (the second value was steady but rarely jumped down to <1ms)
```

Netperf results of hw seem to be affected as well (from >400M to <5M).

chelmuth, 2023-07-04:
I'd bet this is related to my analysis in #4952. Could you please add a call to `genode_uplink_notify_peers()` to *dde_linux/src/lib/legacy/lx_kit/work.cc* before returning from `_schedule()` and test staging again?

m-stein, 2023-07-04:
@chelmuth Thanks for the idea. I tried it, but it didn't change the numbers :(

But I noticed something else.  The typical pattern with staging looks like this:
```
hw
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=41 ttl=64 time=991.477 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=42 ttl=64 time=12.150 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=43 ttl=64 time=991.514 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=44 ttl=64 time=12.151 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=45 ttl=64 time=992.492 ms

nova
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=5 ttl=64 time=501.122 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=6 ttl=64 time=501.124 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=7 ttl=64 time=501.121 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=8 ttl=64 time=501.123 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=9 ttl=64 time=501.123 ms
```

But when I enable verbose_packets in the router, there appear longer periods of relatively low latency on hw while nova only shows a generally raised latency.
```
hw
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=2 ttl=64 time=30.947 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=3 ttl=64 time=15.801 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=4 ttl=64 time=25.684 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=5 ttl=64 time=14.789 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=6 ttl=64 time=30.848 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=7 ttl=64 time=15.783 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=8 ttl=64 time=30.864 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=9 ttl=64 time=14.778 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=10 ttl=64 time=26.167 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=11 ttl=64 time=14.919 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=12 ttl=64 time=30.858 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=13 ttl=64 time=14.660 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=14 ttl=64 time=25.6 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=15 ttl=64 time=201.567 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=16 ttl=64 time=725.100 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=17 ttl=64 time=524.10 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=18 ttl=64 time=89.930 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=19 ttl=64 time=523.281 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=20 ttl=64 time=535.130 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=21 ttl=64 time=711.670 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=22 ttl=64 time=538.933 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=23 ttl=64 time=74.89 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=24 ttl=64 time=507.80 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=25 ttl=64 time=522.981 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=26 ttl=64 time=694.240 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=27 ttl=64 time=524.152 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=28 ttl=64 time=58.219 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=29 ttl=64 time=519.296 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=30 ttl=64 time=535.256 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=31 ttl=64 time=679.340 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=32 ttl=64 time=539.10 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=33 ttl=64 time=42.157 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=34 ttl=64 time=535.196 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=35 ttl=64 time=523.288 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=36 ttl=64 time=662.329 ms
[init -> ping] 64 bytes from 10.0.0.6: icmp_seq=37 ttl=64 time=550.3 ms
```

m-stein, 2023-07-04:
@chnuke had the idea that we might simply forget to push a state machine at a point where it still could make progress and that the typical 500ms delay come from some sort of regular event that triggers the pending state machine as side effect.

cnuke, 2023-07-04:
For documenting purposes: yesterday's commit [f7122ee](https://github.com/cnuke/genode/commit/f7122eeca6b2c801c035997febc017f28a30fb4d) calls `genode_uplink_notify_peers` from the `pc_nic_drv`'s _Main_ object signal-handler somewhat remedies the problem. However, @chelmuth idea, implemented in [bac7fcb](https://github.com/cnuke/genode/commit/bac7fcb6b7e9548d89aeb1710286bcf874cf3d4f), fixes the problem (ping from intra to fuji6 results in ~0.89ms).

chelmuth, 2023-07-04:
*500 ms* is exactly `REAL_TIME_UPDATE_PERIOD_US` in the timeout framework. Just coincidence?

m-stein, 2023-07-04:
If I apply only the corrected advice of @chelmuth https://github.com/cnuke/genode/commit/f7122eeca6b2c801c035997febc017f28a30fb4d, nova latency goes down to a steady 0.4 ms while hw latency still shows an average of 250ms. If I furthermore add [f7122ee](https://github.com/cnuke/genode/commit/f7122eeca6b2c801c035997febc017f28a30fb4d), both hw and nova produce 0.4ms latency.

cnuke, 2023-07-07:
To put the results into perspective: without changing the code between different runs I get _~0.4 ms_ sometimes as well as as _~0.8 ms_ at other times. Since we do not have a dedicated test network - in my case there are even multiple switches involved - I would consider everything below 1 ms as acceptable because we cannot rule-out interference with other network activity.

chelmuth, 2023-10-26:
Please help me to understand the state of this issue as I cannot deduce it from the comments.

m-stein, 2023-10-27:
@chelmuth From looking into the code, I'd say that [f7122ee](https://github.com/cnuke/genode/commit/f7122eeca6b2c801c035997febc017f28a30fb4d) is part of staging while a modified version of [bac7fcb](https://github.com/cnuke/genode/commit/bac7fcb6b7e9548d89aeb1710286bcf874cf3d4f) seems to have found its way to staging as e18c02991e5239f19ad92e7787d96a109888e062 (dates to Jul 6). A quick test on qemu shows ~1ms on nova and 1,5ms on hw. Therefore, I'd say the issue is fixed.

closed 2023-10-27 by m-stein