genodelabs/genode #4957: wireguard: initialize timekeeping with RTC

trimpim, 2023-07-04:
Without this servers will deny connections with  "Invalid handshake initiation" when the client is restarted.

jschlatow, 2023-07-04:
@trimpim That's great. I ran into the handshake-initiation issue myself, took this as an opportunity to create a test scenario in Goa but haven't found the time to investigate this issue any further. Just took your commit out for a walk in my test scenario...the handshake now works like a charm.

m-stein, 2023-07-05:
@trimpim Thanks for having taken the time to fix this important issue with the Wireguard port! As I see, you're already processing my review :)

trimpim, 2023-07-05:
@m-stein I force pushed a new version which addressed your comments. I even was able to sneak in one more `const` while doing it :-) unfortunately I had to do a second push as I forgot to replace one occurence of `current` ;-(

m-stein, 2023-07-06:
@trimpim Thanks for adopting my ideas. I created a small fixup fdbd14bbe36 - merely style preferences and diff reduction. What do you think about it?

trimpim, 2023-07-07:
@m-stein I'm fine with that.

I'm currently creating a depot_autopilot test for wireguard, which is a spiced up version of `wg_fetchurl.run`. Our intent is to run this regularly on our pipeline. It tests restarting the client.

chelmuth, 2023-07-07:
@trimpim it's great that you managed to implement the feature by just some lines of code. Nevertheless, I think the implementation may be rearranged like sketched in the following and the contribution broke up into two commits.

1. lx_emul mechanism
   - provide an API for a custom initial *seconds since 1970* value in *lx_emul/time.h*, e.g.,
     `void lx_emul_time_initial(unsigned long long seconds)`
   - override `read_persistent_clock64()` using the initial value, which is *0 seconds* by default
2. app/wireguard adaption
   - construction of an RTC connection
   - calculation of the *seconds* value with the tm lib
   - configuration of lx_emul via the new mechanism
   - in best case, this feature could optionally be disabled (proposed by @m-stein)

Would you like to change your implementation accrodingly?

trimpim, 2023-07-07:
@chelmuth I'll do this. I'll add a third patch which will be `test-wireguard` for `depot_autopilot`, if you do not mind.

m-stein, 2023-07-07:
@trimpim Thanks for willing to give this feature and our feedback so much time!

First, I don't think that WireGuard fits well into the current Depot-Autopilot setup. The depot_autopilot.run script is meant for very basic framework tests with not much more requirements than a timer driver. If you have additional things to test, then maybe another run script would be the best way to go for now!?

Second, regarding the adaptions to the wireguard port that @chelmuth mentioned: I'd vote for a new and optional configuration attribute:

```
<config use_rtc="yes"></config>
```
The default value is "yes" and yields the behavior you just implemented and that is expected in every productive context. When set to "no", however, the port would act as it used to act, i.e. not requesting an RTC session and not setting the initial time in lx_emul. This way, we don't have to deal with the hassle of providing an RTC on each platform we're testing while the initial time is irrelevant in the tests anyway. With this change, the only adaption to the existing tests would be adding the new attribute.

What do you think?

m-stein, 2023-07-07:
@trimpim Of course, if your Depot-Autopilot test is meant only for your own test infrastructure, I haven't said a thing :) Be aware in this case, that depot_autopilot.run so far has no network support. When we created depot_autopilot.run we thought that it would be better to have a dedicated depot_autopilot_net.run for that later on but didn't approach it yet.

trimpim, 2023-07-07:
@m-stein I will implement the `use_rtc` attribute, as you have described it. I think it is sensible to be able to disable the RTC requirement.

The test would not communicate with the outside world. It is completely self contained. Only two instances and some control logic.

trimpim, 2023-07-10:
@m-stein if you prefer a run script for the test, I gladly will create such a test. Our intention behind using `depot_autopilot` is that it speeds up our pipelines because of less device reboots.

The outline of the test
 - self contained
 - a sub init with the server side (`wireguard` and `lighttpd`)
 - a sub init for the client side that is started multiple times (`wireguard` and `fetchurl`)
   - with disabled RTC it is expected to fail
   - with enabled RTC it is expected to pass

nfeske, 2023-07-11:
> if you prefer a run script for the test, I gladly will create such a test. Our intention behind using depot_autopilot is that it speeds up our pipelines because of less device reboots.

How about keeping the test as a pkg/runtime without adding it to the depot_autopilot, while supporting the execution of the test via `make run/test PKG=test-wireguard`? The run script could then merely set the PKG environment variable before `source`ing `repos/os/run/test.run`. So @trimpim can include the test in his depot-autopilot-based CI.

m-stein, 2023-07-11:
I merely wanted to point out that internal discussions settled on not letting more complex tests like this one enter depot_autopilot.run and I am totally fine with @trimpim creating it as a package ;)

trimpim, 2023-07-13:
@m-stein and @nfeske I have switched to a run script.

Unfortunately I wasn't able to do the negative tests, as the server side of wireguard on Genode doesn't print an error message when the handshake with a client fails.

m-stein, 2023-07-14:
@trimpim One of the maxims of the WireGuard protocol is that "silence is a virtue". So, whenever something doesn't work as expected, WireGuard becomes completely ignorant about it in order to not put further resources into it respectively provide an attacker with additional data. That said, even if this would be desired, the Genode code would have to put a considerable amount of work into detecting such situations.

m-stein, 2023-07-14:
@trimpim It's wonderful how well you organized the code of your test :-) I have added some minor suggestions inline. Furthermore, would you be fine with renaming the test something like ``test-wg_reconnect``, so, that it integrates better into the set of existing test names (wg_ping*, wg_fetchurl, wg_lighttpd, wg_reconfig)?

trimpim, 2023-07-17:
@m-stein thanks for the compliment. 5aa6a03717c6c61e9ce3b6850369ba63cebd20f5 addresses your comments and renames:
 - the run script to `wg_reconnect.run`
 - the depot raw archive to `test-wg_reconnect`

chelmuth, 2023-09-29:
I've to admit that I lost track of this issue. @trimpim @m-stein what's the state?

trimpim, 2023-10-02:
@chelmuth sorry, I have overlooked them for some reason. I created fixup commits for both of your comments. additionally I slightly increased the timeouts in `run/wg_reconnect` this failed on one of my machines when executed in QEMU.

chelmuth, 2023-10-02:
Thanks for the fixups @trimpim. You may squash your topic branch as I plan the merge not until after the upcoming Sculpt release.

m-stein, 2023-10-04:
@chelmuth Looks all very well to me! @trimpim Thanks again for the contribution!

chelmuth, 2023-10-26:
Merged the series to staging.

chelmuth, 2023-10-27:
Unfortunately, *run/wg_fetchurl* failed last night because of the following error.

```
Error: Rtc-session creation failed
```

I'm going to change the default value of `use_rtc` to `false` to restore the original behavior until someone comes up with an RTC driver for pinephone. Also, the new option is undocumentated because wireguard misses any kind of documentation.

m-stein, 2023-10-27:
@chelmuth Shall I schedule writing a README?

chelmuth, 2023-10-27:
Sounds reasonable to me.

m-stein, 2023-11-27:
@chelmuth Does 647de330a7 cover your needs?

chelmuth, 2023-11-28:
Perfect. I merged the commit to staging.

m-stein, 2023-11-28:
@chelmuth Thanks!