genodelabs/genode #4958: dde_linux: update remaining ported USB device drivers to Linux 6.1.20

ssumpf, 2023-07-04:
This concerns drivers that connect to the usb host-controller driver via Genode's USB session interface. Namely, `usb_hid`, `usb_net` (USB ethernet NICs), and `usb_modem` (USB MBIM/WDM modems).

ssumpf, 2023-07-04:
The commit series above enables the `virt_linux` based and updated `usb_hid` driver. `usb_net` and `usb_modem` will follow.

ssumpf, 2023-08-21:
The series above contains the updated `usb_hid_drv` and `usb_net_drv`. `usb_modem_drv` has been removed and the modem functionality is now provided by `usb_net_drv` (branches: https://github.com/ssumpf/genode/tree/issue4958_usb_update and https://github.com/ssumpf/genode-world/tree/issue4958_usb_update).

ssumpf, 2023-09-04:
@jschlatow: Understood, I can set the label to `""` as before.

ssumpf, 2023-09-05:
81a2871 omits the session label

ssumpf, 2023-09-11:
@chelmuth: 0030a89 and 46f2ad3 address your suggestions.

chelmuth, 2023-09-11:
Thanks for your excellent work. Despite my obligatory nitpicking here and there, it's a pleasure to merge the series to staging. Please rebase your topic branch to staging accordingly.

Could we remove *dde_linux/recipes/src/legacy_usb_host_drv* after updating genode-world with the modem fixes?

ssumpf, 2023-09-12:
@chelmuth: 10378c9 fixes `usb_hid_drv` for arm_v8a, 1078e0b fixes `usb_net_drv` (broken by [81a2871](https://github.com/ssumpf/genode/commit/81a28710f0b7cf162cf7cec2dabfe0aae8f3b244)).

ssumpf, 2023-09-12:
@chelmuth: 29d130e removes the label from `usb_net_drv`

chelmuth, 2023-09-12:
Commit 39b4a8ebef9be9c6223775927ff74825ba690cc3 fixes *usb_hid_raw.run* autopilot test for all supported platforms.

chelmuth, 2023-09-27:
The usb_hid_raw test fails on rpi like follows.

```
[2023-09-26 02:44:11] [init -> usb_hid_drv] hid-generic 0003:03EB:204D.0002: input: USB HID v1.11 Mouse [HID 03eb:204d] on usb-usbbus-0/input1
[2023-09-26 02:44:15] [init -> event_dump] Input event #0	PRESS KEY_X 65534	key count: 1
[2023-09-26 02:44:15] [init -> event_dump] Input event #1	RELEASE KEY_X	key count: 0
[2023-09-26 02:44:16] [init -> event_dump] Input event #2	PRESS BTN_LEFT 65534	key count: 1
[2023-09-26 02:44:16] [init -> event_dump] Input event #3	REL_MOTION -1+1	key count: 1
[2023-09-26 02:44:16] [init -> event_dump] Input event #4	RELEASE BTN_LEFT	key count: 0
[2023-09-26 02:44:17] [init -> usb_hid_drv] [31mError: Function usb_clear_halt not implemented yet!
```

For some reason, the HID driver gets a STALL on the device and decides to call this specific endpoint reset, which is a combination of `usb_control_msg_send(USB_REQ_CLEAR_FEATURE, USB_RECIP_ENDPOINT, USB_ENDPOINT_HALT)` and `usb_reset_endpoint()`.

chelmuth, 2023-10-04:
Removed the fixed label until arm_v6 is solved.

tomga, 2023-10-21:
Probably I don't have enough dde_linux knowledge now to fix this issue myself, but it is reproducible on my rpi1 and if there is a need I can test some potential fixes.

chelmuth, 2023-12-05:
> The usb_hid_raw test fails on rpi like follows.
>
> ```
> [2023-09-26 02:44:11] [init -> usb_hid_drv] hid-generic 0003:03EB:204D.0002: input: USB HID v1.11 Mouse [HID 03eb:204d] on usb-usbbus-0/input1 [...]

@ssumpf please also address this issue with virt_linux for arm_v6 - thanks.

ssumpf, 2023-12-05:
@chelmuth: Made a note.

chelmuth, 2024-01-24:
@ssumpf: any news according the behavior above with the current changes?

ssumpf, 2024-01-24:
@chelmuth: I was not able to reproduce this here, yet. I still have to try out disconnect and re-connect, though. I will let you know.

chelmuth, 2024-01-29:
> @chelmuth: I was not able to reproduce this here, yet. I still have to try out disconnect and re-connect, though. I will let you know.

Test fails as before.

```
[2024-01-29 03:29:30] U-Boot 2018.11-rc2-00171-gcf033e04da-dirty (Dec 07 2018 - 11:45:14 +0100)
[2024-01-29 03:29:30] 
[2024-01-29 03:29:30] DRAM:  448 MiB
[2024-01-29 03:29:30] RPI Model B rev2 (0xe)
[2024-01-29 03:29:30] MMC:   sdhci@7e300000: 0
[2024-01-29 03:29:30] Loading Environment from FAT... OK
[2024-01-29 03:29:30] In:    serial
[2024-01-29 03:29:30] Out:   vidconsole
[2024-01-29 03:29:30] Err:   vidconsole
[2024-01-29 03:29:30] Net:   No ethernet found.
[2024-01-29 03:29:30] starting USB...
[2024-01-29 03:29:33] USB0:   scanning bus 0 for devices... 4 USB Device(s) found
[2024-01-29 03:29:33]        scanning usb for storage devices... 0 Storage Device(s) found
[2024-01-29 03:29:35] Hit any key to stop autoboot:  2  1  0 
[2024-01-29 03:29:37] Waiting for Ethernet connection... done.
[2024-01-29 03:29:37] BOOTP broadcast 1
[2024-01-29 03:29:37] DHCP client bound to address 10.0.0.93 (3 ms)
[2024-01-29 03:29:37] Using smsc95xx_eth device
[2024-01-29 03:29:37] TFTP from server 10.0.0.2; our IP address is 10.0.0.93
[2024-01-29 03:29:37] Filename '/tftpboot/hosts/rpi1.scr'.
[2024-01-29 03:29:37] Load address: 0x200000
[2024-01-29 03:29:37] Loading: *##################################################  151 Bytes
[2024-01-29 03:29:37] 	 2.9 KiB/s
[2024-01-29 03:29:37] done
[2024-01-29 03:29:37] Bytes transferred = 151 (97 hex)
[2024-01-29 03:29:37] ## Executing script at 00200000
[2024-01-29 03:29:38] Waiting for Ethernet connection... done.
[2024-01-29 03:29:38] Using smsc95xx_eth device
[2024-01-29 03:29:38] TFTP from server 10.0.0.6; our IP address is 10.0.0.93
[2024-01-29 03:29:38] Filename '/tftpboot/rpi/uImage'.
[2024-01-29 03:29:38] Load address: 0xf000000
[2024-01-29 03:29:40] Loading: *##################################################  3.7 MiB
[2024-01-29 03:29:40] 	 3 MiB/s
[2024-01-29 03:29:40] done
[2024-01-29 03:29:40] Bytes transferred = 3829824 (3a7040 hex)
[2024-01-29 03:29:40] ## Booting kernel from Legacy Image at 0f000000 ...
[2024-01-29 03:29:40]    Image Name:   
[2024-01-29 03:29:40]    Image Type:   ARM Linux Kernel Image (uncompressed)
[2024-01-29 03:29:40]    Data Size:    3829760 Bytes = 3.7 MiB
[2024-01-29 03:29:40]    Load Address: 00800000
[2024-01-29 03:29:40]    Entry Point:  00800000
[2024-01-29 03:29:40]    Verifying Checksum ... OK
[2024-01-29 03:29:40]    Loading Kernel Image ... OK
[2024-01-29 03:29:40] 
[2024-01-29 03:29:40] Starting kernel ...
[2024-01-29 03:29:40] 
[2024-01-29 03:29:40] 
[2024-01-29 03:29:40] kernel initialized
[2024-01-29 03:29:40] Genode 23.11-105-ge4eb679f719
[2024-01-29 03:29:40] 250 MiB RAM and 64535 caps assigned to init
[2024-01-29 03:29:40] [init -> usb_hid_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[2024-01-29 03:29:40] [init -> usb_hid_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[2024-01-29 03:29:40] [init -> usb_hid_drv] usbcore: registered new interface driver usbfs
[2024-01-29 03:29:40] [init -> usb_hid_drv] usbcore: registered new interface driver hub
[2024-01-29 03:29:40] [init -> usb_hid_drv] usbcore: registered new device driver usb
[2024-01-29 03:29:40] [init -> usb_hid_drv] time-clocksource: Switched to clocksource dde_counter
[2024-01-29 03:29:40] [init -> usb_hid_drv] usbcore: registered new interface driver usbhid
[2024-01-29 03:29:40] [init -> usb_hid_drv] usbhid-hid-core: USB HID core driver
[2024-01-29 03:29:40] [init -> usb_hid_drv] Warning: use compatibility mode: will claim all HID devices from USB report
[2024-01-29 03:29:40] [init -> usb_drv -> usb_drv] NR_IRQS: 16, nr_irqs: 16, preallocated irqs: 16
[2024-01-29 03:29:40] [init -> usb_drv -> usb_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[2024-01-29 03:29:40] [init -> usb_drv -> usb_drv] sched_clock: 64 bits at 1000kHz, resolution 1000ns, wraps every 2199023255500ns
[2024-01-29 03:29:41] [init -> usb_drv -> usb_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 19112604462750000 ns
[2024-01-29 03:29:41] [init -> usb_drv -> usb_drv] usbcore: registered new interface driver usbfs
[2024-01-29 03:29:41] [init -> usb_drv -> usb_drv] usbcore: registered new interface driver hub
[... 48 lines cut ...]
[2024-01-29 03:29:43] [init -> usb_drv -> usb_drv] usb 1-1.3: Product: LUFA Mouse and Keyboard Demo
[2024-01-29 03:29:43] [init -> usb_drv -> usb_drv] usb 1-1.3: Manufacturer: Dean Camera
[2024-01-29 03:29:43] [init -> usb_hid_drv] input: HID 03eb:204d as /devices/usb-1-4/0-0:1.0/0003:03EB:204D.0001/input/input0
[2024-01-29 03:29:43] [init -> usb_hid_drv] Connected device: input0 (HID 03eb:204d at usb-usbbus-0/input0)
[2024-01-29 03:29:43] [init -> usb_hid_drv] hid-generic 0003:03EB:204D.0001: input: USB HID v1.11 Keyboard [HID 03eb:204d] on usb-usbbus-0/input0
[2024-01-29 03:29:43] [init -> usb_hid_drv] input: HID 03eb:204d as /devices/usb-1-4/0-0:1.1/0003:03EB:204D.0002/input/input1
[2024-01-29 03:29:43] [init -> usb_hid_drv] Connected device: input1 (HID 03eb:204d at usb-usbbus-0/input1)
[2024-01-29 03:29:43] [init -> usb_hid_drv] hid-generic 0003:03EB:204D.0002: input: USB HID v1.11 Mouse [HID 03eb:204d] on usb-usbbus-0/input1
[... cut ...]
[2024-01-29 03:29:51] [init -> usb_drv -> usb_drv] usb 1-1.3: new full-speed USB device number 5 using dwc_otg
[2024-01-29 03:29:52] [init -> usb_drv -> usb_drv] usb 1-1.3: New USB device found, idVendor=03eb, idProduct=204d, bcdDevice= 0.01
[2024-01-29 03:29:52] [init -> usb_drv -> usb_drv] usb 1-1.3: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[2024-01-29 03:29:52] [init -> usb_drv -> usb_drv] usb 1-1.3: Product: LUFA Mouse and Keyboard Demo
[2024-01-29 03:29:52] [init -> usb_drv -> usb_drv] usb 1-1.3: Manufacturer: Dean Camera
[2024-01-29 03:30:28] Error: Test execution timed out
```

ssumpf, 2024-01-29:
@chelmuth: As said, looks like disconnect is the problem here. This is the next point on my list.

ssumpf, 2024-01-30:
@chelmuth: Out of curiosity, I also get `<no return address found>` even with `-fno-omit-frame-pointer` on the RPI. What am I doing wrong?

ssumpf, 2024-01-30:
@chelmuth: 5342c6c implements `usb_clear_halt` and fixes the problem on RPI for me. The commit can be found on my _issue4958_usb_hid_rpi_ branch.

chelmuth, 2024-01-30:
> @chelmuth: Out of curiosity, I also get `<no return address found>` even with `-fno-omit-frame-pointer` on the RPI. What am I doing wrong?

Works for me. Commit 4b71e19e27322abb6eefccf308f62d7ffff097b5 produces the following output on RPi.

```
kernel initialized
Genode 23.11-107-g4b71e19e273
252 MiB RAM and 64535 caps assigned to init
[init -> test-log] hex range:          [0e00,1680)
[init -> test-log] empty hex range:    [0abc0000,0abc0000) (empty!)
[init -> test-log] hex range to limit: [f8,ff]
[init -> test-log] invalid hex range:  [f8,08) (overflow!)
[init -> test-log] negative hex char:  0xfe
[init -> test-log] positive hex char:  0x02
[init -> test-log] floating point:     1.70
[init -> test-log] multiarg string:    "parent -> child.7"
[init -> test-log] String(Hex(3)):     0x3
[init -> test-log] Very long messages:
[init -> test-log -> log] 1.....................................................................................................................................................................................................................................2
[init -> test-log] 3.....................................................................................................................................................................................................................................4
[init -> test-log] 5.....................................................................................................................................................................................................................................6
[init -> test-log] 
Kernel: fp=0x401feb14
Kernel: width=7
Kernel: width=7
Kernel: width=7
Kernel: width=7
Kernel: width=7
Kernel: fp=0x401feb14
[init -> test-log] backtrace "ep"
[init -> test-log]   401feb14  1000618
[init -> test-log]   401feccc    a822c
[init -> test-log]   401fee34    70630
[init -> test-log]   401fee54    60580
[init -> test-log]   401fefb4    672bc
[init -> test-log] Test done.
```

chelmuth, 2024-02-01:
It seems we're not quite done yet: 1 out of 3 runs fails.

```
[2024-02-01 03:01:39] [init -> usb_hid_drv] Disconnected device: input1
[2024-02-01 03:01:39] Error: illegal READ at address 0x1458f98 by pager_object: pd='init -> usb_drv -> usb_drv' thread='ep' ip=0x106dea4
```

chelmuth, 2024-02-22:
I moved the failure above to #5126.