genodelabs/genode #4959: hw/imx53: strange timing behavior (at least) in depot_autopilot

chelmuth, 2023-07-06:
Several depot_autopilot tests time-out on imx53 while the wall clock progressed just a fraction of the timeout.

imx53_qsb

```
[2023-07-06 06:47:59] [init -> depot_autopilot] --- Run "test-timer" (max 30 sec) ---
[2023-07-06 06:47:59] [init -> depot_autopilot]
[2023-07-06 06:47:59] [init -> depot_autopilot] 0.072 [init -> client] --- timer test ---
[2023-07-06 06:48:00] [init -> depot_autopilot] 0.089 [init -> client] register 4-seconds timeout...
[2023-07-06 06:48:04] [init -> depot_autopilot] 4.097 [init -> client] timeout fired - 0/160/160
[2023-07-06 06:48:04] [init -> depot_autopilot]
[2023-07-06 06:48:04] [init -> depot_autopilot]  test-timer                      failed   133.297  timeout 30 sec
[2023-07-06 06:48:04] [init -> depot_autopilot]
...
[2023-07-06 06:40:54] [init -> depot_autopilot] --- Run "test-clipboard" (max 20 sec) ---
[2023-07-06 06:40:54] [init -> depot_autopilot] 
[2023-07-06 06:40:54] [init -> depot_autopilot] 
[2023-07-06 06:40:54] [init -> depot_autopilot]  test-clipboard                  failed   128.962  timeout 20 sec
...
[2023-07-06 06:41:44] [init -> depot_autopilot] --- Run "test-file_vault_config_report" (max 70 sec) ---
[2023-07-06 06:41:44] [init -> depot_autopilot] 
[2023-07-06 06:41:44] [init -> depot_autopilot] 0.625 [init -> file_vault] Error: failed to deliver Report session with label fs_query -> listing
[2023-07-06 06:41:44] [init -> depot_autopilot] 0.639 [init -> report_rom] report 'file_vault -> ui_report'
[2023-07-06 06:41:44] [init -> depot_autopilot] 0.647 [init -> report_rom]   <ui_report version="step_1_wait" state="uninitialized"/>
[2023-07-06 06:41:50] [init -> depot_autopilot] 6.567 [init -> report_rom] report 'file_vault -> ui_report'
[2023-07-06 06:41:50] [init -> depot_autopilot] 6.576 [init -> report_rom]   <ui_report version="step_2_init" state="initializing"/>
[2023-07-06 06:41:51] [init -> depot_autopilot] 6.866 [init -> file_vault] child "truncate_file" exited with exit value 0
[2023-07-06 06:41:51] [init -> depot_autopilot] 7.140 [init -> file_vault] child "tresor_init_trust_anchor" exited with exit value 0
[2023-07-06 06:41:51] [init -> depot_autopilot] 7.490 [init -> file_vault] child "tresor_init" exited with exit value 0
[2023-07-06 06:41:52] [init -> depot_autopilot] 7.935 [init -> file_vault] child "sync_to_tresor_vfs_init" exited with exit value 0
[2023-07-06 06:41:52] [init -> depot_autopilot] 8.265 [init -> file_vault -> mke2fs] mke2fs 1.46.5 (30-Dec-2021)
[2023-07-06 06:41:52] [init -> depot_autopilot] 8.349 [init -> file_vault -> mke2fs] Error: no plugin found for socket()
[2023-07-06 06:41:52] [init -> depot_autopilot] 8.359 [init -> file_vault -> mke2fs] Error: no plugin found for socket()
[2023-07-06 06:41:52] [init -> depot_autopilot] 8.370 [init -> file_vault -> mke2fs] Creating filesystem with 1024 1k blocks and 128 inodes
[2023-07-06 06:41:52] [init -> depot_autopilot] 8.381 [init -> file_vault -> mke2fs] 
[2023-07-06 06:41:52] [init -> depot_autopilot] 8.390 [init -> file_vault -> mke2fs] Allocating group tables: 0/1   done
[2023-07-06 06:41:52] [init -> depot_autopilot] 8.483 [init -> file_vault -> mke2fs] Writing inode tables: 0/1   done
[2023-07-06 06:41:53] [init -> depot_autopilot] 8.706 [init -> file_vault -> mke2fs] Writing superblocks and filesystem accounting information: 0/1   done
[2023-07-06 06:41:53] [init -> depot_autopilot] 8.720 [init -> file_vault -> mke2fs] 
[2023-07-06 06:41:53] [init -> depot_autopilot] 8.743 [init -> file_vault] child "mke2fs" exited with exit value 0
[2023-07-06 06:41:53] [init -> depot_autopilot] 8.877 [init -> file_vault -> client_fs_fs_query] Error: failed to watch '/'
[2023-07-06 06:41:53] [init -> depot_autopilot] 8.889 [init -> file_vault -> client_fs_fs_query] Error: failed to watch '/data'
[2023-07-06 06:41:53] [init -> depot_autopilot] 8.904 [init -> report_rom] report 'file_vault -> ui_report'
[2023-07-06 06:41:53] [init -> depot_autopilot] 8.912 [init -> report_rom]   <ui_report version="step_2_init" state="unlocked"/>
[2023-07-06 06:41:53] [init -> depot_autopilot] 
[2023-07-06 06:41:53] [init -> depot_autopilot]  test-file_vault_config_report   failed   138.378  timeout 70 sec
```

imx53_qsb_tz
```
[2023-07-06 06:56:55] [init -> depot_autopilot] --- Run "test-timer" (max 30 sec) ---
[2023-07-06 06:56:55] [init -> depot_autopilot] 
[2023-07-06 06:56:55] [init -> depot_autopilot] 0.074 [init -> client] --- timer test ---
[2023-07-06 06:56:55] [init -> depot_autopilot] 0.090 [init -> client] register 4-seconds timeout...
[2023-07-06 06:56:59] [init -> depot_autopilot] 4.099 [init -> client] timeout fired - 0/160/160
[2023-07-06 06:56:59] [init -> depot_autopilot] 
[2023-07-06 06:56:59] [init -> depot_autopilot]  test-timer                      failed   133.178  timeout 30 sec
```

In the last two tests, the measured duration of `138/133` seems off while the wall clock in the first looks okay. :thinking:

m-stein, 2023-07-06:
@chelmuth There is a bug in the epit timer driver (fix: ef3f0718e1e) that was initially masked by an unnecessary cast and that was then unmasked by 13453e3c68c. The fix is ready to be merged and should fix the depot autopilot problems. However, while debugging, I found another issue with the timeout test that first appears with c3b53bc2ec90e1f34b2d38cbc385873ca55fe0da. I'm working at it.

chelmuth, 2023-07-06:
Marvelous, I merged 58d10bde02f2b2d7083bf061ca55d6c0fd411ddb instantly.

Does c3b53bc loosen the former *volatile* semantics too much?

m-stein, 2023-07-06:
@chelmuth From what I've seen so far it should rather be a performance regression. I'll investigate tomorrow.

I also found that I accidentally added a new bug to the epit timer. Here's a fix-up e0a2710ce28.

chelmuth, 2023-07-07:
Does 6432d0adc2c2b5d37930f49cba98ac22bbd8727d reflect the essence of your fix? Could we go with this commit instead?

m-stein, 2023-07-11:
@chelmuth Logically speaking, yes. However, I would at least rename last to initial_cnt or start_timeout_cnt and cnt to curr_cnt in order to make it easier to grasp what's happening there.

chelmuth, 2023-07-12:
> @chelmuth Logically speaking, yes. However, I would at least rename last to initial_cnt or start_timeout_cnt and cnt to curr_cnt in order to make it easier to grasp what's happening there.

I'm open for a standalone follow-up commit that improves the readability. Please consider to add a short comment about the overflow case why `Device::Cnt` is read again (like in the commit message).

m-stein, 2023-07-12:
@chelmuth Like this 3d58d20fd3 ?

chelmuth, 2023-07-12:
> @chelmuth Like this [3d58d20](https://github.com/genodelabs/genode/commit/3d58d20fd3bb6af37d206b1b417fae7734f9bbd8) ?

Perfect, merged the commit to staging.

m-stein, 2023-07-13:
@nfeske I'm sorry for that blooper. This should fix the build errors 59637917e8.

nfeske, 2023-07-13:
Thanks @m-stein for the fixup.

nfeske, 2023-07-14:
Fixed in the master branch.

closed 2023-07-14 by nfeske

chelmuth, 2024-03-15:
Unfortunately, 90ba794e8e02386d40e150217492306291ec530b did not fix the duration calculation in all cases because it depended a kind of strict tick-tock semantic of `_duration()` and `_start_one_shot()`, which is (no longer?) the case. If `_duration()` was called twice consecutively, the implementation interpreted `Ocif` again as roll-over and returned durations larger than `_max_value()` (more than 130s). Commit d8e976524915aaa2f5472f11f03938d85f6daf51 addresses this by replacing the `Ocif` check by a check for `curr_cnt > initial_cnt` to detect roll-over.