genodelabs/genode #4966: nic_uplink: connect a Nic to an Uplink session

m-stein, 2023-07-14:


m-stein, 2023-07-14:
@nfeske With the below listed commits on my nic_uplink branch, Sculpts network dialog receives an extra option for raw access to wired NIC. Could you please have a quick first look at it as soon as your time allows for it? There are 3 limitations that are already on my list in case you're fine with this approach: Only one NIC session is supported; There is no button for raw Wifi access; The nic_uplink config cannot be managed manually.

16be780055 net_test-pc.sculpt: minimal setup for net tests
1630d2ac25 sculpt network: wired without router interception
f978c876f2 ping: archive recipes
cc2e00cb95 nic_uplink: connect a Nic to an Uplink session

nfeske, 2023-07-17:
Thank you for the comprehensive commit for the nic_uplink component.

While causally reviewing it, I noticed a few code repetitions regarding the exception handling (e.g., `free(ram_ds)`), which could be avoided by using a guard object. Have you considered this?

Regarding the Sculpt integration, I think we can get away w/o modifying the Sculpt manager by merely providing a `pkg/pc_nic` package that includes the driver and the `nic_uplink` component in a sub init. So the user can leave Sculpt's network setting to `off` (neither the NIC router nor any driver is started) and use the `pc_nic` instead. This pattern could be expanded to a `pc_wifi_nic` pkg that combines the `nic_uplink` with the wifi driver.

m-stein, 2023-07-17:
@nfeske Thanks for the fast feedback! I'll try to implement your suggestions.

m-stein, 2023-07-18:
@nfeske Thanks for the hint regarding the guard. It was on-spot as the code shows. Is this updated commit list what you had in mind?
262063520f net_test-pc.sculpt: minimal setup for net tests
ae30f77e42 ping: src and pkg recipes
e645bca563 nic_uplink: connect a Nic to an Uplink session
ce85c16ab6 nic_router: clean-up session creation with guard

m-stein, 2023-07-19:
@nfeske I've updated the commits on my merge_to_staging branch according to our offline discussion.

m-stein, 2023-07-20:
@nfeske I had to convert the former guard renaming into fixups while rebasing and added also test fixes for arm and lx and the hex-dump utility we discussed:
 ac00d8c4bd util/formatted_output.h: hex-dump utility
966a38b11d ping: align dhcp client more with nic_router again
be67be82e5 fixup "nic_uplink: connect a Nic to an Uplink session" (fix wakeup calls)
d786ded80f fixup "nic_uplink: connect a Nic to an Uplink session" (no autopilot test on lx)
f4fa22cf2e fixup "nic_uplink: connect a Nic to an Uplink session" (rename Create_session_guard)
95bd93157f fixup "nic_router: clean-up session creation with class" (rename Create_session_guard)

nfeske, 2023-07-21:
Thanks for the fixup commits, which I merged to staging.

I also like the `Hex_dump` utility. Just gave it a try for logging the GDB commands in the debug monitor, which worked very well. I like the two-bytes grouping, which nicely assists with the visual navigation through the hex values.

As a minor detail, I would welcome the compatibility of the `Hex_dump` with `Const_byte_range_ptr`. The fixup fe017f459da480e641458cf710d723c2f6afae35 changes the specific argument type to `auto`. So all types that follow the pattern of `Byte_range_ptr` (hosting a `start` pointer and a `num_bytes` value) can work.

nfeske, 2023-07-21:
On second thought, I also merged https://github.com/genodelabs/genode/commit/3fc26d97d7e9997e9e16061cbe8eacc9242889f7 to staging now because I sense that it would be nice to have those launchers at hand for experimentation. I'm a little bit concerned about `net_test-pc.sculpt` eventually becoming stale if not regularly exercised but let's see how it fares.

m-stein, 2023-07-21:
@nfeske

>  I'm a little bit concerned about net_test-pc.sculpt eventually becoming stale if not regularly exercised but let's see how it fares.

How about making this utility fit for automated testing? Using the "deploy:" option and "LOG=core", I was able to produce the same test as nic_uplink.run but with Sculpt and packages. I don't know exactly how to integrate the two env variables (SCULPT, LOG) in the test infrastructure but its shouldn't be too hard.

Thanks for having tested the Hex_dump utility so fast! I'm glad you like it.

> The fixup https://github.com/genodelabs/genode/commit/fe017f459da480e641458cf710d723c2f6afae35 changes the specific argument type to auto. So all types that follow the pattern of Byte_range_ptr (hosting a start pointer and a num_bytes value) can work.

Thanks for the fixup. I'm surprised that this works. Kind of a cheap interface or template.

m-stein, 2023-07-21:
I tried a bit around with my own drivers package for a headless sculpt. I used "sculpt/drivers/linux"  as starting point and removed fb_sdl. When I set up everything around this and add a deploy file, Sculpt stops doing anything after startup. I couldn't make up a quick way of providing a dummy fb, so, I'll leave it with this for now. Maybe later I find the time to go on.
```
[init] Warning: runtime: configured RAM exceeds available RAM, proceed with 573575104
[init -> runtime] parent provides
[init -> runtime]   service "ROM"
[init -> runtime]   service "PD"
[init -> runtime]   service "RM"
[init -> runtime]   service "CPU"
[init -> runtime]   service "LOG"
[init -> runtime]   service "IO_MEM"
[init -> runtime]   service "IO_PORT"
[init -> runtime]   service "IRQ"
[init -> runtime]   service "Gui"
[init -> runtime]   service "Timer"
[init -> runtime]   service "Block"
[init -> runtime]   service "Report"
[init -> runtime]   service "Platform"
[init -> runtime] parent provides
[init -> runtime]   service "VM"
[init -> runtime]   service "Usb"
[init -> runtime]   service "File_system"
[init -> runtime]   service "Rtc"
[init -> runtime]   service "TRACE"
[init -> runtime]   service "Event"
[init -> runtime]   service "Capture"
[init -> runtime]   service "Gpu"
[init -> runtime]   service "Pin_state"
[init -> runtime]   service "Pin_control"
[init -> runtime] child "runtime_view"
[init -> runtime]   RAM quota:  7944K
[init -> runtime]   cap quota:  166
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "panel_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "menu_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "settings_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "system_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "network_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "popup_view"
[init -> runtime]   RAM quota:  3848K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "file_browser_view"
[init -> runtime]   RAM quota:  7944K
[init -> runtime]   cap quota:  116
[init -> runtime]   ELF binary: menu_view
[init -> runtime]   priority:   0
[init -> runtime] child "ram_fs"
[init -> runtime]   RAM quota:  776K
[init -> runtime]   cap quota:  266
[init -> runtime]   ELF binary: vfs
[init -> runtime]   priority:   0
[init -> runtime]   provides service File_system
[init -> runtime] child "ram_fs" announces service "File_system"
```

m-stein, 2023-07-21:
@nfeske The nic_uplink test has to be further restricted (66097d3c1a). I found that the "foc" filter in os net tests seems to be outdated and thereby unified the restriction code (8213a124ee). As this is quite redundant, I'd rather like to have two central functions (assert_target_supports_nic, assert_target_supports_isolated_nic) but I hesitated as I sensed that further effort into this direction is not appreciated.

nfeske, 2023-07-24:
To avoid the duplicated checks in the run scripts, how do you think about moving the checks to an include file like repos/os/run/assert_nic.inc, which could be `source`d by each run script that depends on a nic driver?

The "isolated" nic is a rather special case (let's remember that it is designated as a mere intermediate stepping stone), which does not call for introducing a pattern, in my opinion.

m-stein, 2023-07-24:
@nfeske Like this?
b9e9cf4527 os net tests: unify & document target restrictions
3add7981d5 fixup "nic_uplink: connect a Nic to an Uplink session" (restrict test targets further)

nfeske, 2023-07-26:
Thank you for the two commits, which I just merged to staging.

m-stein, 2023-07-27:
@nfeske This 05792dd88c should fix imx6q+sel4.

chelmuth, 2023-08-24:
Merged in Genode 23.08.

closed 2023-08-24 by chelmuth