genodelabs/genode #4968: rework vmm library API

atopia, 2023-07-17:
The new API emphasizes control over the vCPU data by granting access through the `Vcpu::with_state()` method, which makes sure that the vCPU is stopped, invokes the supplied function with a reference to the VM state and resumes the vCPU if the function returns true.

The old `Vcpu::run()`, `Vcpu::pause()` and Vcpu::state() methods are removed in favor of the new API. Execution of the vCPU is now interrupted by sending a native Genode Signal to its `Vcpu_handler`, which will run the VMM's exit handling method. When this method retrieves the vCPU state by calling `Vcpu::with_state()`, the outside interruption is detected and on x86 a recall exit is injected into the state to signal the vCPU interruption / pause request to the VMM's vCPU handler.

chelmuth, 2023-09-13:
@atopia I reviewed all commits one by one and added appropriate style fixups/cleanups. I also added a WiP commit for base-nova/port update. Please have a look and squash if you confirm to get to a clean topic branch.

atopia, 2023-09-13:
@chelmuth thanks a lot for your style fixes and simplifications, I've pushed a  [branch with the squashed commits](/atopia/genode/tree/4968-new-vmm-api-squash) as requested.

chelmuth, 2023-09-14:
Before the merge to staging it would be best if @alex-ab reviews and merges the changes in 0489d6705abfaee9a43bc17e7577a05421caced5.

@atopia are there other changes to consider? genode-world?

atopia, 2023-09-14:
@chelmuth yes: [seoul](/atopia/genode-world/tree/4968-new-vmm-api), notably atopia/genode-world@497a330f11674fdaedcc7be8690e449c8d424e6a 
@alex-ab acked the seoul changes via email on Jul 18th. I have just locally rebased the branch to staging, where the commits applied cleanly.

chelmuth, 2023-09-14:
Thanks, I prepared another new-vmm-api branch in my genode-world repo.

chelmuth, 2023-09-14:
I merged the updates series to staging.

Unsurprisingly, `make lib/virtualbox5` now fails like follows. Should we fix it or remove the generic virtualbox5 variant?

```
    COMPILE  sup_vmm.o
In file included from .../genode/repos/ports/src/virtualbox5/generic/sup_vmm.cc:33:
.../genode/repos/ports/src/virtualbox5/vcpu_svm.h: In member function ‘void Vcpu_handler_svm::run_vm()’:
.../genode/repos/ports/src/virtualbox5/vcpu_svm.h:141:41: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘run’
  141 |                 void run_vm()   { _vcpu.run(); }
      |                                         ^~~
.../genode/repos/ports/src/virtualbox5/vcpu_svm.h: In member function ‘virtual void Vcpu_handler_svm::pause_vm()’:
.../genode/repos/ports/src/virtualbox5/vcpu_svm.h:142:41: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘pause’
  142 |                 void pause_vm() { _vcpu.pause(); }
      |                                         ^~~~~
.../genode/repos/ports/src/virtualbox5/vcpu_svm.h: In constructor ‘Vcpu_handler_svm::Vcpu_handler_svm(Genode::Env&, size_t, Genode::Affinity::Location, unsigned int, Genode::Vm_connection&, Genode::Allocator&)’:
.../genode/repos/ports/src/virtualbox5/vcpu_svm.h:164:41: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘state’
  164 |                         _state = &_vcpu.state();
      |                                         ^~~~~
.../genode/repos/ports/src/virtualbox5/vcpu_svm.h:166:31: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘run’
  166 |                         _vcpu.run();
      |                               ^~~
In file included from .../genode/repos/ports/src/virtualbox5/generic/sup_vmm.cc:34:
.../genode/repos/ports/src/virtualbox5/vcpu_vmx.h: In member function ‘void Vcpu_handler_vmx::run_vm()’:
.../genode/repos/ports/src/virtualbox5/vcpu_vmx.h:182:41: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘run’
  182 |                 void run_vm()   { _vcpu.run(); }
      |                                         ^~~
.../genode/repos/ports/src/virtualbox5/vcpu_vmx.h: In member function ‘virtual void Vcpu_handler_vmx::pause_vm()’:
.../genode/repos/ports/src/virtualbox5/vcpu_vmx.h:183:41: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘pause’
  183 |                 void pause_vm() { _vcpu.pause(); }
      |                                         ^~~~~
.../genode/repos/ports/src/virtualbox5/vcpu_vmx.h: In constructor ‘Vcpu_handler_vmx::Vcpu_handler_vmx(Genode::Env&, size_t, Genode::Affinity::Location, unsigned int, Genode::Vm_connection&, Genode::Allocator&)’:
.../genode/repos/ports/src/virtualbox5/vcpu_vmx.h:205:41: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘state’
  205 |                         _state = &_vcpu.state();
      |                                         ^~~~~
.../genode/repos/ports/src/virtualbox5/vcpu_vmx.h:207:31: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘run’
  207 |                         _vcpu.run();
      |                               ^~~
make[2]: *** [.../genode/repos/base/mk/generic.mk:62: sup_vmm.o] Error 1
make[1]: *** [var/libdeps:524: virtualbox5.lib] Error 2
make: *** [Makefile:336: gen_deps_and_build_targets] Error 2
```

chelmuth, 2023-09-15:
@atopia my first runs of Sculpt with current staging show that we're not quite there. The Ubuntu 22.04/20.04 VM does not crash but runs quite slow in some aspects.

- Boot, login, starting programs, and shutdown each take about 3 mins each.
- There were some interesting screen artifacts looking like FPU or cache issues.
- On the other hand, using the GUI is snappy and feels like Sculpt 23.04.

I now have to investigate how staging without the VMM commits appears.

chelmuth, 2023-09-15:
I can confirm that vbox6 works as expected on a22c7e8831d89cc8a0cebe672e0f8b8c442f7672 (right before the changes in this issue).

chelmuth, 2023-09-15:
Unfortunately, the series has a negative impact on *server/tz_vmm* on arm_v7a to.

```
  Program server/tz_vmm/spec/imx53_qsb/imx53_qsb_tz_vmm
    COMPILE  block_driver.o
In file included from .../genode/repos/os/src/server/tz_vmm/include/block_driver.h:22,
                 from .../genode/repos/os/src/server/tz_vmm/block_driver.cc:15:
.../genode/repos/os/src/server/tz_vmm/include/vm_base.h:77:80: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘state’
   77 |                 Vm_state                   &_state       { *(Vm_state *)&_vcpu.state() };
      |                                                                                ^~~~~
.../genode/repos/os/src/server/tz_vmm/include/vm_base.h:77:88: error: invalid initialization of non-const reference of type ‘Genode::Vm_state&’ from an rvalue of type ‘<brace-enclosed initializer list>’
   77 |                 Vm_state                   &_state       { *(Vm_state *)&_vcpu.state() };
      |                                                                                        ^
.../genode/repos/os/src/server/tz_vmm/include/vm_base.h: In member function ‘void Genode::Vm_base::run()’:
.../genode/repos/os/src/server/tz_vmm/include/vm_base.h:100:38: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘run’
  100 |                 void run()   { _vcpu.run(); }
      |                                      ^~~
.../genode/repos/os/src/server/tz_vmm/include/vm_base.h: In member function ‘void Genode::Vm_base::pause()’:
.../genode/repos/os/src/server/tz_vmm/include/vm_base.h:101:38: error: ‘class Genode::Vm_connection::Vcpu’ has no member named ‘pause’
  101 |                 void pause() { _vcpu.pause(); }
      |                                      ^~~~~
make[2]: *** [.../genode/repos/base/mk/generic.mk:62: block_driver.o] Error 1
make[1]: *** [var/libdeps:198: imx53_qsb_tz_vmm.prg] Error 2
make: *** [Makefile:336: gen_deps_and_build_targets] Error 2
```

chelmuth, 2023-09-18:
With the FPU WiP of today applied I could run my guest Ubuntu for updating its software repositories and watching some video streams in parallel. Unfortunately, I ended up in errors I never experienced before.

```
[2023-09-18 16:13:45.226534] [init -> runtime -> upx_vbox6_22_04_mate -> vbox] Warning: out of capability selectors for handling server requests
... many many of these messages ...
[2023-09-18 16:13:45.226555] [init -> runtime -> upx_vbox6_22_04_mate -> vbox] Warning: out of capability selectors for handling server requests
[2023-09-18 16:13:45.226571] [init -> runtime -> upx_vbox6_22_04_mate -> vbox] Error: Thread 'RTThrdPP' died because of an uncaught exception
[2023-09-18 16:13:45.226587] [init -> runtime -> upx_vbox6_22_04_mate -> vbox] Error: Uncaught exception of type 'Genode::Cpu_session::Thread_creation_failed'
[2023-09-18 16:13:45.226604] [init -> runtime -> upx_vbox6_22_04_mate -> vbox] Warning: abort called - thread: RTThrdPP
```

chelmuth, 2023-09-19:
@alex-ab regarding exception 16 in vbox6, I discovered there should be no `fninit` required in `Nova_vcpu::_exit_entry()` after `fxsave` according to the Intel manual entry for fxsave:

> After the FPU state has been saved, the FPU is reset to the same default values it is set to with the FINIT/FNINIT
instructions (see “FINIT/FNINIT—Initialize Floating-Point Unit” in this chapter).

Do you agree? If yes, the quickfix would be pointless and we should change the NOVA API to pass guest FPU state in UTCB for portal entries soon.

alex-ab, 2023-09-19:
> @alex-ab regarding exception 16 in vbox6, I discovered there should be no `fninit` required in `Nova_vcpu::_exit_entry()` after `fxsave` according to the Intel manual entry for fxsave:

Yes, since the suspicion is not valid, the potential quick fix is no one.

> @alex-ab regarding exception 16 in vbox6, I discovered there should be no `fninit` required in `Nova_vcpu::_exit_entry()` after `fxsave` according to the Intel manual entry for fxsave:
> 
> > After the FPU state has been saved, the FPU is reset to the same default values it is set to with the FINIT/FNINIT
> > instructions (see “FINIT/FNINIT—Initialize Floating-Point Unit” in this chapter).
> 
> Do you agree? If yes, the quickfix would be pointless and we should change the NOVA API to pass guest FPU state in UTCB for portal entries soon to eliminate this potential error cause.

Since the suspicion is not valid, the potential quick fix is no one, yes.

atopia, 2023-09-19:
@chelmuth could you please have a look if [971fa5b](https://github.com/atopia/genode/commit/971fa5b2459ff1ffdc8db420bcd89bbb8ffcd134) works on hardware? Possibly the vmm causes a runtime exception by accessing `Vm_state` outside of the `with_state()` lambda.

chelmuth, 2023-09-20:
> @chelmuth could you please have a look if [6d0932b](https://github.com/atopia/genode/commit/6d0932be75295b1c7ad71eb61c0f5a603bfdd865) works on hardware? Possibly the vmm causes a runtime exception by accessing `Vm_state` outside of the `with_state()` lambda.

@atopia please have a look, `make KERNEL=hw BOARD=imx53_qsb_tz core` fails with errors.

chelmuth, 2023-09-20:
@atopia forgot to mention: please rebase on current staging.

atopia, 2023-09-20:
@chelmuth huh?
both
```
make -C build/arm_v7a KERNEL=hw BOARD=imx53_qsb_tz core
```
and
```
 make -C build/arm_v7a KERNEL=hw BOARD=imx53_qsb_tz server/tz_vmm
 ```
 work for me after a `cleanall`.

chelmuth, 2023-09-20:
Commit 91f72a5775510b1d607c96a186b383a2e2ee703e implements the minimum UTCB fixup before returning from a VM exit handler. @alex-ab noted this morning that the new scheme of "remote" handler invocation mixes ordinary RPC handling with VM-exit portals. The *out of capability* warnings mentioned above are effectively prevented for 15 mins - long-term test is running.

atopia, 2023-09-21:
> I see. Because I fixed the immediate issue with `Vcpu_state` , the remaining error becomes apparent on `make run/tz_vmm` only.

Sorry I missed the remaining issue. The build issue should be fixed by [eeaab61](https://github.com/atopia/genode/commit/eeaab61c4cce0944dbaeb9d6ac8d718e17827a08), please give it a spin.

chelmuth, 2023-09-21:
Unfortunately, eeaab61c4cce0944dbaeb9d6ac8d718e17827a08 fails like follows... @skalk any idea how to address this?

```
[init -> imx53_qsb_tz_vmm] Error: Uncaught exception of type 'Genode::Reconstructible<Genode::Vm_base::State_container>::Deref_unconstructed_object'
[init -> imx53_qsb_tz_vmm] Warning: abort called - thread: ep
```

atopia, 2023-09-21:
> Unfortunately, [eeaab61](https://github.com/genodelabs/genode/commit/eeaab61c4cce0944dbaeb9d6ac8d718e17827a08) fails like follows... @skalk any idea how to address this?
> 
> ```
> [init -> imx53_qsb_tz_vmm] Error: Uncaught exception of type 'Genode::Reconstructible<Genode::Vm_base::State_container>::Deref_unconstructed_object'
> [init -> imx53_qsb_tz_vmm] Warning: abort called - thread: ep
> ```

In case that wasn't clear, this is certainly due to my deliberate use of `state_container->ref` (or calls to `state()` which amount to the same) throught the porting. Most likely I missed a constructor using `state()` outside of the `with_state()` scope, which leads to the exception. I'll have a look into this later.

atopia, 2023-09-22:
@chelmuth hopefully  [00bbd88](https://github.com/atopia/genode/commit/00bbd88115c83fc153701e3a034594386efe8426) averts the exception.

chelmuth, 2023-09-22:
@atopia unfortunately your commit does not address the problem at hand. Please see my comments in d15c5b0b7c37c852a43987aef02f5e4bc898759f.

atopia, 2023-09-28:
@chelmuth my [4968-new-vmm-api-rebased](atopia/genode/tree/4968-new-vmm-api-rebased) should provide the requested style cleanups and attribution fixes.

chelmuth, 2023-10-04:
Fixed in master.

closed 2023-10-04 by chelmuth