genodelabs/genode #4970: depot_download_manager: genereate state report even on empty "installation" ROM

Time0o, 2023-07-21:
I'm not sure if I would classify this as a bug but I'm stuck in a situation that could be resolved by slightly modifying `depot_download_manager` in a way that might be generally desirable:

I'm currently writing a cloud service through which a user can declaratively specify a list of depot archives that should be installed on a system running Genode. While these archives are downloaded I want to display a progress indicator to the user (which should display all archives as "done/installed" once they are installed). What I do to implement this is update the  `installation` report that ends up being read by `depot_download_manager` any time the user updates this list of depot archives and then query the state report of `depot_download_manager`. This works fine except for the following:

Say a user specifies that archives A, B and C should be installed on the system, the archives are downloaded and a progress indicator is displayed. Later, the user wants all slices to be uninstalled again and clears the list of archives. Thus an empty `installation` report is written and ideally this would also trigger an empty `depot_download_manager` state report to be written. But this does not happen, the installation report is unaffected and as  a consequence to the user it looks like uninstalling the archives has not been successful.

Now, currently "uninstalling" archives does not actually modify what files are present on the system but this is an implementation detail, I want the "desired state" and what I display to the user to remain in sync. A possible solution would be to make `depot_query` generate its `dependencies` report even the query it receives contains no `dependencies` but I'm not sure if this is the best solution or if it would break something else.

nfeske, 2023-07-24:
Have you considered the use of the `version` attribute to infer the freshness of the reports? The version string as found in the query is reflected in the generated reports, which helps you to correlate each (potentially stale) report with the latest query.

That idea notwithstanding:

> A possible solution would be to make depot_query generate its dependencies report even the query it receives contains no dependencies but I'm not sure if this is the best solution or if it would break something else.

By peeking at the `depot_query` code (https://github.com/genodelabs/genode/blob/master/repos/gems/src/app/depot_query/main.h#L468), adding an empty `<dependencies/>` node to the query could do the trick. But I haven't tried.

Time0o, 2023-09-06:
@nfeske I have worked around this now by simply clearing the download state report I display in the cloud anytime the `installation` report is updated. This seems to work so I think this can be closed as this does not seem to be a bug per se.

nfeske, 2023-09-06:
I was actually a little uncertain how to proceed with this issue. Thank you for the update. I'm happy that you found a workable solution.