genodelabs/genode #4974: Genode::Root_directory instance breaks vfs_lwip

Time0o, 2023-08-08:
I have noticed an odd interaction where when merely instantiating `Genode::Root_directory` in a libc application running on Linux that uses `lwip` breaks libcurl code.

I have created a more or less minimal example in the form of a Genode repository [here](https://github.com/Time0o/genode_repositories). The crux is line 26 in `repos/experiments/src/app/lwip_vfs_issue/main.cc`. When this line is commented out, everything works as expected. This is alternatively also the case when I use `lxip` instead of `lwip` in the provided runscript `repos/experiments/run/lwip_vfs_issue.run`. In both cases the runscript completes successfully. So the combination `lwip` + `Genode::Root_directory` seems to be problematic, running the runscript as-is on my system results in:

```
[init -> lwip_vfs_issue] lwIP Nic interface up address=10.10.10.2 netmask=255.255.255.0 gateway=10.10.10.1
[init -> lwip_vfs_issue] * Could not resolve host: www.google.com
[init -> lwip_vfs_issue] * Closing connection 0
[init -> lwip_vfs_issue] curl returned: 6
```

nfeske, 2023-10-25:
My interpretation is as follows:

Each libc-based component implicitly has a VFS instance. By creating a custom `Root_directory` in your component, a second VFS is created. Since you feed the same configuration to both, the second VFS tries to create an instance of vfs_lwip as did the libc's instance. But since lwIP has global variables, both instances end up stepping on each other's toes.

I don't think that we should aspire the hosting of multiple TCP/IP stacks in one component.

To test my hypothesis, you may try feeding a different config (w/o lwip) to your custom Root_directory. If this works, I suggest closing this issue.

Generally, let me advise against the multiple instantiation of VFSes in one component. Although it may work in a concrete case, the VFS was not ultimately not designed for such usage patterns. To stay on the safe side, I recommend using only one VFS instance per component.

nfeske, 2023-11-30:
I'm closing this issue because there are no further plans attached to it.

closed 2023-11-30 by nfeske