genodelabs/genode #4977: monitor: add more debugging features

cproc, 2023-08-08:
Add more debugging features like stopping and resuming of threads, single-stepping and exception reporting.

chelmuth, 2023-08-16:
The automatic test fails on NOVA each run like follows.

```
[2023-08-16 04:13:16] [init -> monitor -> test-log] Test done.
[2023-08-16 04:13:27] localhost:5555: Connection timed out.
[2023-08-16 04:13:27] Breakpoint 1 at 0xdae10: file /data/genode/repos/base/src/lib/ldso/debug.cc, line 21.
[2023-08-16 04:13:27] The program is not being run.
[2023-08-16 04:13:27] Unknown thread 2.
[2023-08-16 04:13:27] Load new symbol table from "debug/test-monitor_gdb"? (y or n) [answered Y; input not from terminal]
[2023-08-16 04:13:27] Reading symbols from debug/test-monitor_gdb...
[2023-08-16 04:13:27] (gdb) b Main::Main
[2023-08-16 04:13:27] Breakpoint 2 at 0x1000498: file /data/genode/repos/os/src/test/monitor_gdb/main.cc, line 79.
[2023-08-16 04:13:27] (gdb) c
[2023-08-16 04:13:27] The program is not being run.
[2023-08-16 04:13:27] (gdb) *** Error: Breakpoint in Main::Main() did not trigger
```

cproc, 2023-08-16:
Fixup commit a9c97f2 adds the ARM64 support to the monitor recipe and fixup commit 0182795 uses a `tcp_terminal` on real x86_64 hardware.

commit 4a77a4da (2023-08-09): monitor: add more debugging features

Fixes #4977
commit 7000fb86 (2023-08-23): monitor: add more debugging features

Fixes #4977
closed 2023-08-24 by chelmuth