genodelabs/genode #4981: libc: vfs_terminal file wrongly classified as socket

cproc, 2023-08-08:
Currently, `stat()` classifies VFS nodes of type `TRANSACTIONAL_FILE` as socket, which has the effect that a Genode-native GDB treats a `vfs_terminal` file as socket[2], which fails. Can this be fixed in the libc or should GDB get patched instead?

[1] https://github.com/genodelabs/genode/blob/83f3df65602fa5f9d3a91ee4aaf2fead76957dda/repos/libports/src/lib/libc/vfs_plugin.cc#L113

[2] https://github.com/bminor/binutils-gdb/blob/4f3e26ac6ee31f7bc4b04abd8bdb944e7f1fc5d2/gdb/serial.c#L213-L220

chelmuth, 2023-08-09:
I could imagine to return `S_IFSOCK` only for socket files in the configured socket-fs path. But what should `stat()` return for the terminal file (or any other non-socket `TRANSACTIONAL_FILE`)?

cproc, 2023-08-14:
> I could imagine to return `S_IFSOCK` only for socket files in the configured socket-fs path. But what should `stat()` return for the terminal file (or any other non-socket `TRANSACTIONAL_FILE`)?

For the terminal file `S_IFCHR` could be fitting, but I'm not sure if it would fit all other non-socket `TRANSACTIONAL_FILE`s as well. Regarding sockets, currently `fstat()` for the fd returned by `socket()` returns -1, because this fd has no path set and the socket-fs-internal files should probably not be classified as sockets, because they cannot be used with socket functions like `connect()`? So we would need to add a path to the fd returned by `socket()` first to have something that can actually be classified with `S_IFSOCK`, it seems.

chelmuth, 2023-09-28:
@cproc What about `Dirent_type::TRANSACTIONAL_FILE` in the same file?

cproc, 2023-09-28:
Commit a7468c9 changes `Dirent_type::TRANSACTIONAL_FILE` as well. According to [1] `DT_SOCK` means "A local-domain socket" and my understanding is that the pseudo-files of the socket fs are only used within the libc itself, so it seems like there is no code depending on the classification of these files as socket.

[1] https://www.gnu.org/software/libc/manual/html_node/Directory-Entries.html

commit 7644cd39 (2023-09-28): libc: stat: classify transactional files as character devices

Fixes #4981
closed 2023-10-04 by chelmuth