genodelabs/genode #4986: Scroll when adding a component

ya0, 2023-08-23:
I want to add a virtual machine to sculp os. I run sculp on the Oracle VM. I follow the turotial here https://genodians.org/jschlatow/2022-10-27-fresh-vm-on-sculpt.
In the step where the vbox6 from genolabs is added about 20 configurations need to be made. The problem I am facing is that I cannon select the last ones by mouse click, because I cannot scroll.

Have you faced this issue before? Is there a workaround for this? Can I edit the setup for the vbox over a config file? Is the oracle vm the problem with the scrolling?

I am using the latest release of sculpt 23.04 (the turotial uses 22.04 maybe this is an issue?)

Thank you for any help in advance :)

atopia, 2023-08-24:
What is the resolution of your display? Can you maybe post a photo of your screen?

nfeske, 2023-08-24:
This is a duplicate of #3771.

Please also consider the related posting at https://lists.genode.org/pipermail/users/2023-January/008334.html

ya0, 2023-08-24:
Thank you for the quick replies! I have manged to change the screen resolutions. Note that in oracle vm not all resolutions work. i.e. full hd crashes.

There is some blogpost https://genodians.org/nfeske/2020-09-17-sculpt-os that describes what resolutions work in genode in oracle vbox. I have used width `1280` and hight `1024` which worked. This solved this particular problem (additionaly, there is also the setting to change the font size which might help!). 

solution: `vim /config/fb.drv` then change the pixels numbers 

@nfeske @atopia can you point me towards setting up new components manually?

closed 2023-08-24 by ya0

nfeske, 2023-08-25:
> can you point me towards setting up new components manually?

The general documentation applies for the component configuration (aka the deploy configuration) as well:

https://genode.org/documentation/articles/sculpt-23-04#Manual_configuration

Adding a component comes down to the insertion of a new `<start>` node at /config/deploy. Under the hood, when interactively adding a component using the UI, Sculpt adds a new `<start>` node at /config/managed/deploy`. In practice, it is often useful to interactively add a component once, and use the resulting `<start>` node as a template for the manually managed `/config/deploy` or a launcher at `/config/launcher/<name>`.

For further usability hints, I'd like to invite you to the Genode mailing list, which is the most natural place for discussing such topics:  https://genode.org/community/mailing-lists