genodelabs/genode #4987: [vfs_pipe] crashes its host server when remote client restarts

ttcoder, 2023-08-26:
## Summary

I have client apps that open() named pipes owned by a vfs_server (which hosts a vfs_pipe plugin).
When said apps get restarted by a change in 'init' (thus without being given the chance of close()ing their named pipe) that triggers a crash in the vfs server, in Genode::Fifo_element::dequeue()

## Details

I've started implementing an "app launcher" using a sub-init, following Norman's advice on the mailing-list. Works fast and great for things like nano3d and top, but my apps are troublesome : they make use of a vfs server (hosted _outside_ the sub-init), and can crash it simply when being forcibly restarted by init.

With addr2line aided by some tracing I collected this pseudo call stack:

```
repos/gems/src/lib/vfs/pipe/plugin.cc:116 notify_read()
repos/base/include/util/fifo.h:180 dequeue_all()
repos/base/include/util/fifo.h:164 dequeue()
```

Seems like the faulty line goes
```
result->Fifo::Element::_next = nullptr
```

The 'result' pointer (0x230f70) is not null, but not valid either. Maybe it's a use-after-free problem ?

## Log output

Launching "top" instead of "Pulse" (i.e. this forcibly kills Pulse):

```
[init -> dyn]   RAM quota:  130824K
[init -> dyn]   cap quota:  266
 ...
[init -> vfs] vfs_pipe: notify_read
[init -> vfs] vfs_pipe: write
[init -> vfs] vfs_pipe: notify_read
[init -> vfs] vfs_pipe: write
[init -> vfs] vfs_pipe: notify_read
[init -> Deskbar] BRoster::_LaunchApp() done: No error: 0 (0)
[init -> vfs] vfs_pipe: notify_read
no RM attachment (WRITE pf_addr=0x230f70 pf_ip=0x10f4b2cd from pager_object: pd='init -> vfs' thread='ep') 
page fault, pd='init -> vfs' thread='ep' cpu=0 ip=0x10f4b2cd address=0x230f70 stack pointer=0x401fec60 qualifiers=0x6 irUWp reason=1
```

ttcoder, 2023-08-30:
More data : the crash also occurs on (self-initiated) abrupt exit() but not on (self initiated) proper quitting, seemingly confirming it's a resource tracking problem :

- if the app closes the pipes and calls env.exit(), the vfs server keeps going
- if the app calls env.exit() _without_ closing the named pipes, the vfs server crashes with the same backtrace as above

nfeske, 2023-09-01:
Thank you @ttcoder for the detailed report, which I appreciate very much. I'm currently mostly offline but will investigate the problem next week.

ttcoder, 2023-09-04:
At a quick glance, the 'reader' and 'writer' seem to mirror what is done for me here (with 'Pulse' and 'broker' respectively), with the reader being repeatedly restarted indeed. My reader does not use select(), just blocking reads -- see "read_from_broker" here : https://chiselapp.com/user/ttcoder/repository/genode-haiku/file?udc=1&ci=2a880f427bdbbb6460cf421af4072a4569e5d53e&name=genode-haiku%2Fhaiku-on-genode%2Fsystem%2Fkinda-kernel%2Fapi-public_sem-port-thread-etc.cpp . The writer (hog-services/broker/broker_server.cpp) does this whenever a (full) BMessage has been received and must be tx'ed to recipient : `sent = write( tx, accu->Buffer(), accu->BufferLength() );` ; the pipe is opened once in a static (before main()), and to trigger the crash it must be closed in the *writer*, when the BMsgBroker Session is severed, otherwise there's no crash ; one pipe should be enough, since I repro with just one app (Pulse) and I have one pipe per app.

If you're looking for a 'shortcut' to replicate this bug, the latest commit (2a880) here allows to do that, by inoking "jam hog-deskbar.run" and using the Deskbar pull-down menu to launch Pulse a couple times : https://chiselapp.com/user/ttcoder/repository/genode-haiku/timeline  (building the hog/distro should be *much* less painful than you had experienced a couple years ago :-)

Though that build requires building to "layer 8" level (i.e. almost everything in the distro) so you might want to avoid that mess and wait until I come up with a better approach from your end, or at least meet you half-way from my end.
Gotta make a few important commits on some things I've been working on, but I should have some time to investigate soon, possibly next week!

nfeske, 2023-09-12:
Once I removed the select call, I was able to reproduce the issue on base-linux and got hold of the following backtrace:

```

0x0000000010f5313d in Genode::Fifo<Genode::Fifo_element<Vfs_pipe::Pipe_handle> >::dequeue<Vfs_pipe::Pipe::notify_read()::{lambda(Genode::Fifo_element<Vfs_pipe::Pipe_handle>&)#1}>(Vfs_pipe::Pipe::notify_read()::{lambda(Genode::Fifo_element<Vfs_pipe::Pipe_handle>&)#1} const&) (func=..., this=0x7ffff7fe5f98)
    at .../repos/base/include/util/fifo.h:164
164					result->Fifo::Element::_next = nullptr;
(gdb) bt
#0  0x0000000010f5313d in Genode::Fifo<Genode::Fifo_element<Vfs_pipe::Pipe_handle> >::dequeue<Vfs_pipe::Pipe::notify_read()::{lambda(Genode::Fifo_element<Vfs_pipe::Pipe_handle>&)#1}>(Vfs_pipe::Pipe::notify_read()::{lambda(Genode::Fifo_element<Vfs_pipe::Pipe_handle>&)#1} const&) (func=..., this=0x7ffff7fe5f98)
    at .../repos/base/include/util/fifo.h:164
#1  Genode::Fifo<Genode::Fifo_element<Vfs_pipe::Pipe_handle> >::dequeue_all<Vfs_pipe::Pipe::notify_read()::{lambda(Genode::Fifo_element<Vfs_pipe::Pipe_handle>&)#1}>(Vfs_pipe::Pipe::notify_read()::{lambda(Genode::Fifo_element<Vfs_pipe::Pipe_handle>&)#1} const&) (func=..., this=0x7ffff7fe5f98) at .../repos/base/include/util/fifo.h:182
#2  Vfs_pipe::Pipe::notify_read (this=0x7ffff7fe3eb0)
    at .../repos/gems/src/lib/vfs/pipe/plugin.cc:116
#3  Vfs_pipe::Pipe::write (src=..., out_count=@0x401ff8a0: 4, this=0x7ffff7fe3eb0)
    at .../repos/gems/src/lib/vfs/pipe/plugin.cc:214
#4  Vfs_pipe::Pipe::write (out_count=@0x401ff8a0: 4, src=..., this=0x7ffff7fe3eb0)
    at .../repos/gems/src/lib/vfs/pipe/plugin.cc:195
#5  Vfs_pipe::Pipe_handle::write (this=<optimized out>, src=..., out_count=@0x401ff8a0: 4)
    at .../repos/gems/src/lib/vfs/pipe/plugin.cc:260
#6  0x000000000100c13a in Vfs_server::Io_node::_execute_write (write_pos=<optimized out>, src=..., 
    this=0x7ffff7f4fc40) at .../repos/os/src/server/vfs/node.h:412
#7  Vfs_server::File::execute_job (this=0x7ffff7f4fc40) at .../repos/os/src/server/vfs/node.h:809
#8  0x000000000100cf14 in Vfs_server::Session_component::_execute_jobs()::{lambda(Vfs_server::Node&)#1}::operator()(Vfs_server::Node&) const (node=..., __closure=<optimized out>)
    at .../repos/os/src/server/vfs/main.cc:255
#9  Genode::Fifo<Vfs_server::Node>::dequeue<Vfs_server::Session_component::_execute_jobs()::{lambda(Vfs_server::Node&)#1}>(Vfs_server::Session_component::_execute_jobs()::{lambda(Vfs_server::Node&)#1} const&) (func=..., 
    this=0x7ffff7fdc368) at .../repos/base/include/util/fifo.h:168
#10 Genode::Fifo<Vfs_server::Node>::dequeue_all<Vfs_server::Session_component::_execute_jobs()::{lambda(Vfs_server::Node&)#1}>(Vfs_server::Session_component::_execute_jobs()::{lambda(Vfs_server::Node&)#1} const&) (func=..., 
    this=0x7ffff7fdc368) at .../repos/base/include/util/fifo.h:182
#11 Vfs_server::Session_component::_execute_jobs (this=0x7ffff7fda018)
    at .../repos/os/src/server/vfs/main.cc:252
#12 Vfs_server::Session_component::process_packets (this=this@entry=0x7ffff7fda018)
    at .../repos/os/src/server/vfs/main.cc:330
#13 0x000000000100d81a in Vfs_server::Session_component::_handle_packet_stream (this=0x7ffff7fda018)
    at .../repos/os/src/server/vfs/main.cc:367
#14 0x00000000500684d0 in Genode::Entrypoint::Signal_proxy_component::signal (
    this=0x50120c38 <Genode::bootstrap_component(Genode::Platform&)::startup+3352>)
    at .../repos/base/src/lib/base/entrypoint.cc:61
#15 0x000000005006af50 in Genode::Meta::call_member<Genode::Meta::Empty, Genode::Entrypoint::Signal_proxy_component, Genode::Meta::Empty> (
    func=(void (Genode::Entrypoint::Signal_proxy_component::*)(Genode::Entrypoint::Signal_proxy_component * const)) 0x50068430 <Genode::Entrypoint::Signal_proxy_component::signal()>, server=...)
    at .../repos/base/include/util/meta.h:462
#16 Genode::Entrypoint::Signal_proxy::Rpc_signal::serve<Genode::Entrypoint::Signal_proxy_component, Genode::Meta::Empty> (args=<synthetic pointer>..., server=...) at .../repos/base/include/base/entrypoint.h:54
#17 Genode::Rpc_dispatcher<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::_do_serve<Genode::Entrypoint::Signal_proxy::Rpc_signal> (args=<synthetic pointer>..., this=<optimized out>)
    at .../repos/base/include/base/rpc_server.h:147
#18 Genode::Rpc_dispatcher<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::_do_dispatch<Genode::Meta::Type_list<Genode::Entrypoint::Signal_proxy::Rpc_signal> > (in=..., out=..., opcode=..., 
    this=<optimized out>) at .../repos/base/include/base/rpc_server.h:181
#19 Genode::Rpc_dispatcher<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::dispatch (
    out=..., in=..., opcode=..., this=<optimized out>)
    at .../repos/base/include/base/rpc_server.h:235
#20 Genode::Rpc_object<Genode::Entrypoint::Signal_proxy, Genode::Entrypoint::Signal_proxy_component>::dispatch (
    this=0x50120c38 <Genode::bootstrap_component(Genode::Platform&)::startup+3352>, opcode=..., in=..., out=...)
    at .../repos/base/include/base/rpc_server.h:273
#21 0x0000000050053eaa in operator() (obj=0x50120c38 <Genode::bootstrap_component(Genode::Platform&)::startup+3352>, 
    __closure=<synthetic pointer>) at .../repos/base/src/lib/base/rpc_dispatch_loop.cc:71
#22 Genode::Object_pool<Genode::Rpc_object_base>::apply<Genode::Rpc_entrypoint::entry()::<lambda(Genode::Rpc_object_base*)> > (func=..., capid=<optimized out>, 
[... cut ...]
--Type <RET> for more, q to quit, c to continue without paging--
    at .../repos/base/include/base/object_pool.h:147
#23 Genode::Rpc_entrypoint::entry (this=0x501200c0 <Genode::bootstrap_component(Genode::Platform&)::startup+416>)
    at .../repos/base/src/lib/base/rpc_dispatch_loop.cc:68
#24 0x000000005008d66a in Genode::Thread::_thread_start ()
    at .../repos/base-linux/src/lib/base/thread_linux.cc:94
#25 0x00000000500a4bb7 in thread_start ()
    at .../repos/base-linux/src/lib/syscall/spec/x86_64/lx_clone.S:62
```

nfeske, 2023-09-12:
@ttcoder, may you give https://github.com/nfeske/genode/commit/ba6fdafe16b8ff857ac245bb34f666b3d5746cbb at try?

ttcoder, 2023-09-12:
Roxors !
(yea my parlance is stilll stuck in the 1990's 8-)

Tried without, then with, the patch applied, and that fixes it indeed. As a bonus (compared to relying on a work-around as I was doing until now), I no longer get warnings in LOG about leaked resources.

Closing (or does the procedure call for attaching a "Fixed" bubble label first ?). Guess this will be part of 22.11 release?

Next I'll try to asap file a ticket about the leaked pthread capability, as that is the last remaining tidbit making my 'distro' experience less than perfect. (Norman: you might be pleased to learn I already started getting in touch with stations, asking if they want to beta test my 'distro')

nfeske, 2023-09-12:
Thanks for the immediate feedback! Let's keep the issue open until the patch has entered the master branch. This will happen automatically because I have annotated the commit message with a `fixed` tag referring to this issue.

Stoked to know that your work is getting into the hands of real users. :)

commit fad9f4fd (2023-09-12): vfs/pipe: fix dangling read_ready_waiters element

This patch removes the obsolete 'io_progress_elem', which was wrongly
enqueued to the 'read_ready_waiters' fifo and not dequeued at
destruction time.

Fixes #4987