genodelabs/genode #4994: run/sdl2: exception during program load 'Genode::Region_map::Region_conflict'

mewmew, 2023-09-07:
I tried running the `run/sdl2` scenario on the *staging* branch of Genode, and a `Region_conflict` error.

Revisions: genodelabs/genode@e78a84196db646f65341459b83cd72abaf46c509 and genodelabs/genode-world@9e06bf73cc5313b0530e68568c8fd516817b0bcf.

```
Genode 23.08
751 MiB RAM and 63253 caps assigned to init
[init -> test-sdl2] Error: LD: exception during program load: 'Genode::Region_map::Region_conflict'
[init -> test-sdl2] Error: Uncaught exception of type 'Genode::Region_map::Region_conflict'
```

<details><summary>detailed output</summary>

```
  Program test/sdl2/test-sdl2
make[1]: Leaving directory '/home/u/Desktop/genode_fresh/genode/build/x86_64'
genode build completed
checking configuration syntax
  CHECK init
using 'core-nova.a' as 'core.a'
install bootloader
creating ISO image...
xorriso 1.5.6 : RockRidge filesystem manipulator, libburnia project.

spawn qemu-system-x86_64 -display sdl -serial mon:stdio -m 800 -cdrom var/run/sdl2.iso -machine q35 -cpu Nehalem-v2 -net nic,model=e1000,netdev=net0 -netdev user,id=net0
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EDX.spec-ctrl [bit 26]

Bender Version 0.9-beta7-51-g28ba2ad-dirty
Patching BDA with I/O port 0x3f8.
Bender: Hello World.
hwp config: eeo=na irq=na hwp=na epp=na epb=na

NOVA Microhypervisor v9-e78a841 (x86_64): [gcc 12.3.0] [MBI2]

[ 0] TSC:2923156 kHz BUS:1006601 kHz (measured)
[ 0] CORE:00:00:0 6:1a:3:0 [1] Intel Core i7 9xx (Nehalem Core i7, IBRS update)

Hypervisor NOVA (API v9)
core     image  [0000000000100000,0000000000c13000)
binaries region [0000000000275000,0000000000c13000) free for reuse
detected physical memory: 0x0000000000000000 - size: 0x000000000009fc00
use      physical memory: 0x0000000000000000 - size: 0x000000000009f000
detected physical memory: 0x0000000000100000 - size: 0x0000000031edf000
use      physical memory: 0x0000000000100000 - size: 0x0000000031edf000
reserved memory: 0x314ca000 - size: 0xb14338 type=-2
reserved memory: 0x9fc00 - size: 0x400 type=2
reserved memory: 0xf0000 - size: 0x10000 type=2
reserved memory: 0x31fdf000 - size: 0x21000 type=2
reserved memory: 0xb0000000 - size: 0x10000000 type=2
reserved memory: 0xfed1c000 - size: 0x4000 type=2
reserved memory: 0xfffc0000 - size: 0x40000 type=2
reserved memory: 0xfd000000 - size: 0x3e8000 type=-5
reserved memory: 0x200000 - size: 0x4000 type=-1
reserved memory: 0x1000 - size: 0x1000 type=-1
reserved memory: 0x1000000 - size: 0x800000 type=-1
reserved memory: 0x1800000 - size: 0x1c00000 type=-1
reserved memory: 0x31fe22fc - size: 0x0 type=-3
Hypervisor reports 1x1 CPU 
Warning: CPU has no invariant TSC.
mapping: affinity space -> kernel cpu id - package:core:thread
 remap (0x0) -> 0 - 0:0:0 boot cpu
ROM modules:
 ROM: [0000000031900000,0000000031928bc8) acpi_drv
 ROM: [0000000031b67000,0000000031b67981) config
 ROM: [0000000000024000,0000000000028000) core_log
 ROM: [0000000031929000,0000000031960d10) decorator
 ROM: [0000000031b66000,0000000031b66366) decorator_init.config
 ROM: [0000000031f0b000,0000000031f0c7b1) drivers.config
 ROM: [0000000031998000,000000003199a7df) en_us.chargen
 ROM: [0000000031b68000,0000000031ba1718) event_filter
 ROM: [0000000031e56000,0000000031e5631a) event_filter.config
 ROM: [0000000031c21000,0000000031c21032) fb_drv.config
 ROM: [0000000031ba2000,0000000031ba201d) focus
 ROM: [0000000031e79000,0000000031eea750) init
 ROM: [0000000031995000,0000000031996cd0) init.xsd
 ROM: [0000000031997000,00000000319975f4) layouter.config
 ROM: [0000000031ca0000,0000000031d8cb08) ld.lib.so
 ROM: [000000003199b000,0000000031b65958) libc.lib.so
 ROM: [0000000031772000,00000000317db0c8) nitpicker
 ROM: [00000000317dc000,00000000318ffba8) pc_usb_host_drv
 ROM: [0000000031eeb000,0000000031f0a800) pci_decode
 ROM: [0000000031c22000,0000000031c9f0a8) platform_drv
 ROM: [0000000000020000,0000000000023000) platform_info
 ROM: [0000000031964000,0000000031994ac0) pointer
 ROM: [000000003174b000,0000000031771ab8) ps2_drv
 ROM: [0000000031d8d000,0000000031daeb18) report_rom
 ROM: [0000000031e57000,0000000031e78c28) rom_filter
 ROM: [0000000031640000,000000003174ade0) sdl2.lib.so
 ROM: [0000000031e55000,0000000031e55542) special.chargen
 ROM: [0000000031961000,0000000031963490) test-sdl2
 ROM: [0000000031ba3000,0000000031bbe468) timer
 ROM: [0000000031bbf000,0000000031c20250) usb_hid_drv
 ROM: [0000000031f92000,0000000031fdd178) vesa_fb_drv
 ROM: [0000000031daf000,0000000031e529c0) vfs.lib.so
 ROM: [0000000031f59000,0000000031f91ea8) window_layouter
 ROM: [0000000031f0d000,0000000031f58f60) wm
 ROM: [0000000031e53000,0000000031e54055) wm.config

36884K kernel memory

Genode 23.08
751 MiB RAM and 63253 caps assigned to init
[init -> test-sdl2] Error: LD: exception during program load: 'Genode::Region_map::Region_conflict'
[init -> test-sdl2] Error: Uncaught exception of type 'Genode::Region_map::Region_conflict'
[init -> test-sdl2] Warning: abort called - thread: ep
[init -> drivers -> acpi_drv] Found MADT
[init -> drivers -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0
[init -> drivers -> acpi_drv] MADT IRQ 5 -> GSI 5 flags: 13
[init -> drivers -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13
[init -> drivers -> acpi_drv] MADT IRQ 10 -> GSI 10 flags: 13
[init -> drivers -> acpi_drv] MADT IRQ 11 -> GSI 11 flags: 13
[init -> drivers -> acpi_drv] Found MCFG
[init -> drivers -> acpi_drv] MCFG BASE 0xb0000000 seg 0x0 bus 0x0-0xff
[init -> drivers -> acpi_drv] RSDT OEM 'BOCHS ', table id 'BXPC    ', revision 1, creator 'BXPC' (1)
[init -> drivers -> acpi_drv] SMBIOS 3 table (entry point: 0x156d00 structures: 0xf6d20)
[init -> drivers -> usb_hid_drv] --- USB HID input driver ---
[init -> drivers -> usb_hid_drv] Warning: use compatibility mode: will claim all HID devices from USB report
[init -> drivers -> usb_hid_drv] Configured HID screen with 0x0 (multitouch=false)
[init -> drivers -> fb_drv] fb mapped to 0x6000
[init -> drivers -> ps2_drv] Warning: setting of mode indicators failed
[init -> drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[init -> drivers -> ps2_drv] detected ExPS/2 mouse - activating scroll-wheel and 5-button support
[init -> drivers -> fb_drv] Found: VESA BIOS version 3.0
[init -> drivers -> fb_drv] OEM: SeaBIOS VBE(C) 2011
[init -> drivers -> fb_drv] Found: physical frame buffer at 0xfd000000 size: 16777216
[init -> drivers -> fb_drv] fb mapped to 0x11000000
[init -> drivers -> fb_drv] using VESA mode 640x480@32
```

</details>

## Steps to reproduce

```bash
git clone https://github.com/genodelabs/genode.git
git clone https://github.com/genodelabs/genode-world.git genode/repos/world
cd genode
./tool/create_builddir x86_64
./tool/ports/prepare_port dde_linux jitterentropy linux nova x86emu
./tool/ports/prepare_port expat libc libdrm mesa sdl2 stdcxx zlib
./tool/ports/prepare_port grub2
# edit build/x86_64/etc/build.conf to use required REPOSITORIES for run/sdl2 demo.
make -C build/x86_64 run/sdl2 KERNEL=nova BOARD=pc DEPOT=omit
```

cnuke, 2023-09-08:
@mewmew thanks for reporting, I'll take a look.

cnuke, 2023-09-08:
Commit [3564aa9](https://github.com/cnuke/genode-world/commit/3564aa91956cc962c76a0ce89f6d17099e6d6412) prevents the linker from creating an incomplete binary (that results in the error you encountered) and [ee6a865](https://github.com/cnuke/genode-world/commit/ee6a865ce11ae7d3db2d1790e4b4b99ac5d62ed4) changes `test-sdl2` to be libc only (otherwise we would have to link against `base` as well).

mewmew, 2023-09-08:
Beautiful, now it's working!

Thanks a lot for the quick resolution @cnuke.

[image]

chelmuth, 2023-09-08:
I merged the commits to genode-world staging.

mewmew, 2023-09-22:
Closing this issue as it's been fixed in staging.

mewmew, 2023-09-22:
@jschlatow mentioned that you prefer to have the issues open until merged with the master branch. Thus re-opening this for now. Feel free to add the *fixed* label.

chelmuth, 2023-10-04:
Fixed in genode-world/master.

closed 2023-10-04 by chelmuth