genodelabs/genode #4995: run/avplay: Audio_out-session creation failed for both KERNEL=nova BOARD=pc and KERNEL=linux BOARD=linux

mewmew, 2023-09-11:
Having played around with SDL rendering (thanks @cnuke, @nfeske and @chelmuth for helping me get up to speed), I now wanted to take a spin at audio playback.

As a first step I wanted to get a minimal running example which plays a sound and then work backwards to try and understand all the components involved.

I tried running `run/avplay` with both `KERNEL=linux BOARD=linux` and `KERNEL=nova BOARD=pc` on *staging* (bad8d9d67e225411316251658fa3d04a34f55a99) and genode-world *staging* (1a415947b5086d5ddfa525b75613b2e953aade3f) but ran into the same issue, namely:

```
[init -> avplay] Error: Audio_out-session creation failed (label="", ram_quota=535560, cap_quota=4, channel="front left")
[init -> avplay] Error: could not connect to 'Audio_out' service
```

<details>
<summary>detailed output for KERNEL=nova BOARD=pc</summary>

```
spawn qemu-system-x86_64 -display sdl -serial mon:stdio -m 800 -cdrom var/run/avplay.iso -machine q35 -cpu Nehalem-v2 -net nic,model=e1000,netdev=net0 -netdev user,id=net0
qemu-system-x86_64: warning: TCG doesn't support requested feature: CPUID.07H:EDX.spec-ctrl [bit 26]

Bender Version 0.9-beta7-51-g28ba2ad-dirty
Patching BDA with I/O port 0x3f8.
Bender: Hello World.
hwp config: eeo=na irq=na hwp=na epp=na epb=na

NOVA Microhypervisor v9-bad8d9d (x86_64): [gcc 12.3.0] [MBI2]

[ 0] TSC:2932239 kHz BUS:1009718 kHz (measured)
[ 0] CORE:00:00:0 6:1a:3:0 [1] Intel Core i7 9xx (Nehalem Core i7, IBRS update)

Hypervisor NOVA (API v9)
core     image  [0000000000100000,000000000334e000)
binaries region [0000000000275000,000000000334e000) free for reuse
detected physical memory: 0x0000000000000000 - size: 0x000000000009fc00
use      physical memory: 0x0000000000000000 - size: 0x000000000009f000
detected physical memory: 0x0000000000100000 - size: 0x0000000031edf000
use      physical memory: 0x0000000000100000 - size: 0x0000000031edf000
reserved memory: 0x2ed8f000 - size: 0x324f338 type=-2
reserved memory: 0x9fc00 - size: 0x400 type=2
reserved memory: 0xf0000 - size: 0x10000 type=2
reserved memory: 0x31fdf000 - size: 0x21000 type=2
reserved memory: 0xb0000000 - size: 0x10000000 type=2
reserved memory: 0xfed1c000 - size: 0x4000 type=2
reserved memory: 0xfffc0000 - size: 0x40000 type=2
reserved memory: 0xfd000000 - size: 0x3e8000 type=-5
reserved memory: 0x200000 - size: 0x4000 type=-1
reserved memory: 0x1000 - size: 0x1000 type=-1
reserved memory: 0x1000000 - size: 0x800000 type=-1
reserved memory: 0x1800000 - size: 0x1c00000 type=-1
reserved memory: 0x31fe22fc - size: 0x0 type=-3
Hypervisor reports 1x1 CPU
Warning: CPU has no invariant TSC.
mapping: affinity space -> kernel cpu id - package:core:thread
 remap (0x0) -> 0 - 0:0:0 boot cpu
ROM modules:
 ROM: [000000002f240000,000000002f268bc8) acpi_drv
 ROM: [000000002f269000,000000002f8e2798) avcodec.lib.so
 ROM: [000000002ef05000,000000002ef08af8) avdevice.lib.so
 ROM: [000000002f934000,000000002f97bad8) avfilter.lib.so
 ROM: [000000002ef21000,000000002f0897f0) avformat.lib.so
 ROM: [000000002ef09000,000000002ef20b58) avplay
 ROM: [0000000031ed6000,0000000031eea4c0) avresample.lib.so
 ROM: [0000000031d54000,0000000031d7d118) avutil.lib.so
 ROM: [0000000031a0d000,0000000031a0da4a) config
 ROM: [0000000000030000,0000000000034000) core_log
 ROM: [000000002f8e3000,000000002f91ad10) decorator
 ROM: [0000000031a0c000,0000000031a0c366) decorator_init.config
 ROM: [0000000031f0b000,0000000031f0c7b1) drivers.config
 ROM: [000000003183e000,00000000318407df) en_us.chargen
 ROM: [0000000031a0e000,0000000031a47718) event_filter
 ROM: [0000000031df9000,0000000031df931a) event_filter.config
 ROM: [0000000031b22000,0000000031b22032) fb_drv.config
[... 97 lines cut ...]
git clone  --branch staging https://github.com/genodelabs/genode-world.git genode/repos/world
cd genode
./tool/create_builddir x86_64
# edit build/x86_64/etc/build.conf to use required REPOSITORIES for run/avplay demo (gems + world + libports).
./tool/ports/prepare_port jitterentropy linux nova x86emu
./tool/ports/prepare_port libav libc sdl zlib
./tool/ports/prepare_port mesa
./tool/ports/prepare_port grub2
./tool/depot/create FORCE=1 REBUILD= UPDATE_VERSIONS=1 genodelabs/pkg/x86_64/drivers_interactive-pc genodelabs/pkg/x86_64/motif_wm genodelabs/bin/x86_64/base-nova genodelabs/bin/x86_64/init genodelabs/bin/x86_64/nitpicker CROSS_DEV_PREFIX=/usr/local/genode/tool/23.05/bin/genode-x86-
./tool/depot/create FORCE=1 REBUILD= UPDATE_VERSIONS=1 genodelabs/pkg/x86_64/drivers_interactive-linux genodelabs/bin/x86_64/base-linux CROSS_DEV_PREFIX=/usr/local/genode/tool/23.05/bin/genode-x86-

# for KERNEL=nova BOARD=pc
make -C build/x86_64 run/avplay KERNEL=nova BOARD=pc DEPOT=omit

# for KERNEL=linux BOARD=linux
make -C build/x86_64 run/avplay KERNEL=linux BOARD=linux DEPOT=omit
```

## Cool demo

Exactly why there was an issue with the Audio-out session, I have no idea.

The graphics were way cool though!

[image]

mewmew, 2023-09-11:
On an (un)related note, building the `run/avplay` demo from scratch outlined all but one dependency required.

All of these commands were suggested automatically by the build system:
```
./tool/ports/prepare_port jitterentropy linux nova x86emu
./tool/ports/prepare_port libav libc sdl zlib
./tool/ports/prepare_port grub2
./tool/depot/create FORCE=1 REBUILD= UPDATE_VERSIONS=1 genodelabs/pkg/x86_64/drivers_interactive-pc genodelabs/pkg/x86_64/motif_wm genodelabs/bin/x86_64/base-nova genodelabs/bin/x86_64/init genodelabs/bin/x86_64/nitpicker CROSS_DEV_PREFIX=/usr/local/genode/tool/23.05/bin/genode-x86-
./tool/depot/create FORCE=1 REBUILD= UPDATE_VERSIONS=1 genodelabs/pkg/x86_64/drivers_interactive-linux genodelabs/bin/x86_64/base-linux CROSS_DEV_PREFIX=/usr/local/genode/tool/23.05/bin/genode-x86-
```

However, the `mesa` requirement was missing, which resulted in the following error when building the sdl library:
```
    COMPILE  events/SDL_events.o
In file included from /home/u/Desktop/genode_audio_play/genode/contrib/sdl-8543c5f3448ebea97afa84aee6691e38a78e756e/src/lib/sdl/src/events/../video/SDL_sysvideo.h:37,
                 from /home/u/Desktop/genode_audio_play/genode/contrib/sdl-8543c5f3448ebea97afa84aee6691e38a78e756e/src/lib/sdl/src/events/SDL_sysevents.h:24,
                 from /home/u/Desktop/genode_audio_play/genode/contrib/sdl-8543c5f3448ebea97afa84aee6691e38a78e756e/src/lib/sdl/src/events/SDL_events.c:28:
/home/u/Desktop/genode_audio_play/genode/contrib/sdl-8543c5f3448ebea97afa84aee6691e38a78e756e/include/SDL/SDL_opengl.h:44:10: fatal error: GL/gl.h: No such file or directory
   44 | #include <GL/gl.h>      /* Header File For The OpenGL Library */
      |          ^~~~~~~~~
compilation terminated.
make[3]: *** [/home/u/Desktop/genode_audio_play/genode/repos/base/mk/generic.mk:58: events/SDL_events.o] Error 1
make[2]: *** [var/libdeps:746: sdl.lib] Error 2
make[1]: *** [Makefile:336: gen_deps_and_build_targets] Error 2
make[1]: Leaving directory '/home/u/Desktop/genode_audio_play/genode/build/x86_64'
Error: Genode build failed
make: *** [Makefile:431: run/avplay] Error 252
make: Leaving directory '/home/u/Desktop/genode_audio_play/genode/build/x86_64'
```

In other words, `mesa` was required for `sdl` (specifically for the `GL/gl.h` header), but no suggestion was given to prepare the `mesa` port (i.e. `./tool/ports/prepare_port mesa`).

These suggestion are very helpful, so I wanted to outline this missing dependency so that it can be included in the build system (if possible).

An example of the "suggestions" I'm talking about was given for `grub2`, as it is required to prepare to boot media for running the demo in Qemu.

```
Port not prepared or outdated:
  grub2

You can prepare respectively update it as follows:
  /home/u/Desktop/genode_audio_play/genode/tool/ports/prepare_port grub2
```

nfeske, 2023-09-12:
Thanks @mewmew for reporting.

> These suggestion are very helpful, so I wanted to outline this missing dependency so that it can be included in the build system (if possible).

Commit https://github.com/genodelabs/genode-world/commit/831c5497161f674083a48bbfce4dbe971ebd00af addresses this shortcoming.

The SDL audio backend accepts the missing audio-out connection, which is quite unusual for a Genode component. But the error messages appear nevertheless. The avplay.run script is not supposed to provide any audio though. To silence those messages, we could in principle wire up the component to the black_hole server. 

Like the other run scripts based on gui_app.inc, avplay.run uses the bare-bones drivers_interactive pkgs for the most basic set of drivers needed for simple graphical scenarios. To enjoy audio, one is instead supposed to run the application directly on Sculpt. For avplay, there actually exists a pkg recipe at world/recipes/pkg/avplay.

Down the road, I think it would be best to support audio in Goa (using linux_audio_drv for testing on Linux), and successfully turn the gui_app.inc-based run scripts into Goa projects.

BTW, thank you for your note of appreciation about the little demo. It's this one: https://www.pouet.net/prod.php?which=16946 :-)