genodelabs/genode #4997: add missing libc support for ripgrep

atopia, 2023-09-13:
The `ripgrep` command line program calls or links to a couple of functions via the `libc` crate that so far have not been provided by our C library.

To enable basic support for ripgrep:

- Implement FreeBSD11 ABI versions of `lstat` and `readdir_r` and provide them as versioned symbols in our `compat_libc`
- Provide dummy implementations of the following functions:
  - `pthread_set_name_np()`
  - `posix_spawn_file_actions_addchdir_np()`
  - `cpuset_getaffinity()`
- Implement support for FreeBSD's `KERN_ARND` sysctl interface

atopia, 2023-09-13:
Cleaned up and rebased on current staging. @chelmuth please have a look and give it a spin.

chelmuth, 2023-09-14:
Merged to staging.

atopia, 2023-09-28:
@chelmuth please pick the attribution fixups.

chelmuth, 2023-09-28:
Merged the fixups to staging.

chelmuth, 2023-10-04:
Fixed in master.

closed 2023-10-04 by chelmuth