genodelabs/genode #5000: nova: propagate uncached policy to page attributes

jschlatow, 2023-09-19:
The last two days I have been debugging the IOMMU implementation for the platform driver, trying to solve sporadic DMA access faults. Everything looks as if changes to the IOMMU translation table do not propagate to memory in time. Reading the spec back and forth, trying different context-cache and IOTLB invalidation strategies, write-buffer flushing, and believing in ghosts, I looked into base-nova how my uncached ram dataspace holding the translations tables is handled. I am stunned seeing that the cached/uncached attribute is never propagated to nova. This perfectly explains my observations.

If I'm not mistaken, the nova kernel does not provide an interface for this (cf. #1566). I'm wondering if there is any subtle mechanism I'm overlooking, in particular because this would render all allocations of uncached DMA buffers being actually mapped as cached memory on base-nova. @alex-ab Do you have any insight?

alex-ab, 2023-09-20:
> If I'm not mistaken, the nova kernel does not provide an interface for this (cf. #1566). I'm wondering if there is any subtle mechanism I'm overlooking, in particular because this would render all [...]

Yes, the kernel provides no interface, beside the writecombining support, to specify caching attributes. The IOMMU kernel implementation explicitly flushes the IOMMU TLB when changes at the pagetable require it.

jschlatow, 2023-09-20:
> The IOMMU kernel implementation explicitly flushes the IOMMU TLB when changes at the pagetable require it.

That is correct, yet does not ensure that changes to the page table already propagated to memory. The nova kernel also executes a `clflush` on the modified page-table entries. I can do the same in the platform driver, which should solve the sporadic translation faults.

What I'm worried about, however, is that there are a few `Platform::Dma_buffer` instantiations in our code base using the `UNCACHED` attribute. Apparently, there is no issue with these being actually mapped as cached memory on base-nova. However, if we switched to base-hw we would probably see some performance degradation. Not to mention the discrepancy between what is expressed at API level and what we actually get by the kernel.

How can we fix this?
a) Should we add an interface to nova for specifying cache attributes?
b) Should we log a warning if uncached memory is tried to be attached by base-nova in order to find and eliminate the occurrences where we expect uncached memory but get cached memory?

chelmuth, 2023-09-20:
@cnuke @ssumpf How does the lacking support for `UNCACHED` affect *drivers/gpu/intel* which is a prominent user of this cache attribute?

jschlatow, 2023-09-20:
I just re-read the Snooping and Memory Type sections of the "Intel® Virtualization Technology for Directed I/O" spec. My current understanding is as follows:

- In legacy translation mode (used by nova), the memory type for paging structures and pages seems to be fixed to WB (write back).
- Unless a DMA read request has set the no-snoop bit, the access is snooped. (Snooping can also be enforced on page level even if the no-snoop bit is set, but this is not used by nova).
- I.e. accesses from both sides (processor and device) to a DMA buffer are coherent so there is no need for uncached memory.

ssumpf, 2023-09-20:
> @cnuke @ssumpf How does the lacking support for `UNCACHED` affect _drivers/gpu/intel_ which is a prominent user of this cache attribute?

In case `UNCACHED` is mapped as `CACHED` by Nova right now and because of @jschlatow observations, it should not affect the Intel GPU multiplexer. It might be that some older hardware is *sometimes* not so cache coherent, but this remains to be tested.

chelmuth, 2023-09-20:
> In case `UNCACHED` is mapped as `CACHED` by Nova right now and because of @jschlatow observations, it should not affect the Intel GPU multiplexer. It might be that some older hardware is _sometimes_ not so cache coherent, but this remains to be tested.

Is this an empirical statement? I'm asking as I could not generally map it to https://github.com/genodelabs/genode/blob/c2f3ef52a09423e29daeec5c59d2807d251231d5/repos/os/src/drivers/gpu/intel/context.h#L724-L729 and https://github.com/genodelabs/genode/blob/c2f3ef52a09423e29daeec5c59d2807d251231d5/repos/os/src/drivers/gpu/intel/ring_buffer.h#L185-L192. Does the Intel GPU disable *snooping*?

chelmuth, 2023-09-20:
Especially `sequence_number()` looks suspicious to me because `clflush` means *clean/write-back and invalidate* on x86.

ssumpf, 2023-09-20:
@chelmuth: This isn't originally my code. As far as I understand it, `cflush` only writes back modified cache lines. We never modify the sequence number in memory, but want to read the one written by the GPU from the hardware status page of the ring buffer of the current context. The `cflush` is thus, an invalidate operation that makes sure the old value is not in cache. When looking at newer kernel versions, there is no `cflush` anymore. Maybe it is from times when there were problems with the IOMMU and GPUs, I honestly don't know.

nfeske, 2023-09-20:
We introduced the uncached attribute for RAM dataspaces only for ARM platforms. To the best of my knowledge - thanks to the strong cache coherency of x86 - RAM dataspaces used as DMA buffers never need to be allocated as uncached on x86. So I think its fine that NOVA never made this distinction.

jschlatow, 2023-09-20:
> Does the Intel GPU disable _snooping_?

The Intel GPU is typically in the domain of a separate DMA remapping hardware unit. Looking at my X260, this unit has slightly different capabilities. For instance, it does not support to force snooping for no-snoop requests. I'm not sure when this would be needed, but I'm wondering why it does not support this mode. Most notably, however, this hardware unit supports memory-type support. The implication of this is that as soon as the hardware unit is not operated in legacy translation mode but the unit's MTRRs are not explicitly enabled, the effective memory type will be set to UC (for the GPU). According to my understanding, this would prevent snooping and therefore  require a clflush from the CPU side. Now my wild guess: maybe the clflush got eliminated in newer kernel versions by enabling the MTRRs appropriately.

jschlatow, 2023-09-20:
> We introduced the uncached attribute for RAM dataspaces only for ARM platforms. To the best of my knowledge - thanks to the strong cache coherency of x86 - RAM dataspaces used as DMA buffers never need to be allocated as uncached on x86. So I think its fine that NOVA never made this distinction.

I basically agree. Am I correct assuming that base-hw does the same?

Nevertheless, I'd vote for reflecting this by explicitly allocating DMA buffers as cached in the x86-specific driver components such as _drivers/gpu/intel_.

jschlatow, 2023-10-03:
Commit 5f65f41 applies my aforementioned suggestion of explicitly requesting CACHED dma buffers in _drivers/gpu/intel_.

chelmuth, 2023-10-04:
@jschlatow may it be worth to add a hook `_check_cacheability(Cache)` to `Core::Ram_dataspace_factory` that could be implemented per platform to deny RAM allocation were `cache != CACHED` on PC platforms?

jschlatow, 2023-10-04:
@chelmuth I had something similar in mind, yet this would complicate platform-independent code as e.g. _lx_kit/env.h_ that needs to use UNCACHED allocations on non-PC platforms.

chelmuth, 2023-10-04:
`Lx_kit::Env::uncached_memory` is IMHO a counter example as it must not be used on PC, or something is fairly broken. Did I get you wrong that you strove to remove the *UNCACHED means CACHED on PC* ambiguity?

jschlatow, 2023-10-04:
> `Lx_kit::Env::uncached_memory` is IMHO a counter example as it must not be used on PC, or something is fairly broken.

If I'm not mistaken, it is used via `dma_alloc_attrs` by _nic/pc_.

chelmuth, 2023-10-05:
So, our implementation of `dma_alloc_attrs()` should be fixed to take the platform requirements into account?

jschlatow, 2023-10-05:
I'm a bit indecisive whether there is a good fix for the _UNCACHED means CACHED on PC ambiguity_.

Denying UNCACHED RAM allocations on PC would imo void the platform-independency of the Platform_session API as clients would have to know whether UNCACHED RAM allocations are permitted on the target platform. Most clients are probably platform-specific anyway, yet there are instances (such as _test/platform_drv_) that would require a mechanism for detecting cacheability support.

Alternatively, I could (temporarily) add a warning in `Core::Ram_dataspace_factory` in order to identify all the instances where UNCACHED RAM is trying to be allocated on PC and think about how these can be fixed individually. This would still permit UNCACHED allocations on PC in the end but cleanup our code base. I think this will also be a first sensible step before introducing the aforementioned API change.

jschlatow, 2023-10-05:
I had a more detailed look and came up with rather straightforward fixes for dde_linux's `dma_alloc_attrs()` (cb99f50) and the dde_bsd audio driver (7cbb40b).

There are, however, a couple of instances where I'm not sure how to fix them:
* The `Expanding_pd_session_client` and `Io_mem_dataspace_component` in base use UNCACHED allocations on all platforms. My idea would be to introduce a platform-specific internal hook to decide whether to use UNCACHED or CACHED.
* The nvme driver also uses UNCACHED allocations. Since this is used by pc and imx, I also need to add a platform-specific component-internal hook. Another idea would be to add a method to the base API but since this seems to be the only affected component, I would like to avoid that.
* For completeness: The legacy lx_kit also uses UNCACHED allocations but I would rather not touch that since it is going to be removed.

chelmuth, 2023-10-06:
> I had a more detailed look and came up with rather straightforward fixes for dde_linux's `dma_alloc_attrs()` [...]

Great, both commits reflect what I had in mind.

>     * The `Expanding_pd_session_client` and `Io_mem_dataspace_component` in base use UNCACHED allocations on all platforms. My idea would be to introduce a platform-specific internal hook to decide whether to use UNCACHED or CACHED.

I'm not sure that `Expanding_pd_session_client::try_alloc()` states the correct default value for `cache` resp. if it should declare a default value at all. `try_alloc()` is an overridden virtual method of `virtual Alloc_result Ram_allocator::try_alloc(size_t size, Cache cache = CACHED) = 0`, which reflects my intention that RAM is cached by default. So, I suggest to change `Expanding_pd_session_client::try_alloc()` accordingly.

Io_mem dataspaces are either uncached or write-combined but as far as I can tell never cached. In any case, let's not mix RAM and memory-mapped I/O in this discussion.

>     * The nvme driver also uses UNCACHED allocations. Since this is used by pc and imx, I also need to add a platform-specific component-internal hook. Another idea would be to add a method to the base API but since this seems to be the only affected component, I would like to avoid that.

As the cacheability property may require precautions in the drivers itself this also holds for the NVMe PCI driver. I'd suggest to add a architecture-specific header to the driver that selects the correct cache attribute and possibly other precautions like cache-clean/invalidate operations. I understand that this driver does not need additional precautions on ARM (uncached) and x86 (coherent CPU and devices) DMA buffers but should IMO clearly declare this intention on allocation. The lx_kit on the other supports both cached and uncached DMA buffers as Linux drivers use these in a platform-aware fashion.

>     * For completeness: The legacy lx_kit also uses UNCACHED allocations but I would rather not touch that since it is going to be removed.

I totally agree.

jschlatow, 2023-10-09:
> I'm not sure that `Expanding_pd_session_client::try_alloc()` states the correct default value for `cache` resp. if it should declare a default value at all. `try_alloc()` is an overridden virtual [...]

I think it's best to remove the default argument for `Expanding_pd_session_client`. I have not found any use of it since try_alloc() appears to be used only via its base class so that the overridden default argument becomes ineffective. Removing the default argument will therefore not change the cacheability property or, if I missed any instance in which `Expanding_pd_session_client::try_alloc` is called, we will get a compile-time error. I've implemented this in [2565efa](https://github.com/jschlatow/genode/commit/2565efab69c1ac1802c9fbac1b96e3bfe4c04d96).

> Io_mem dataspaces are either uncached or write-combined but as far as I can tell never cached. In any case, let's not mix RAM and memory-mapped I/O in this discussion.

I agree.

> As the cacheability property may require precautions in the drivers itself this also holds for the NVMe PCI driver. I'd suggest to add a architecture-specific header to the driver that selects the [...]

I've implemented this in [e31db85](https://github.com/jschlatow/genode/commit/e31db85dd8c22de7d074cf63c7ccd5337d0db137). You can find all commits related to this issue on my [5000_uncached_ram_on_x86 branch](https://github.com/jschlatow/genode/commits/5000_uncached_ram_on_x86).

chelmuth, 2023-10-10:
I'm ready to merge the series after the Sculpt release. Thanks @jschlatow for your patience and effort ;-)

chelmuth, 2023-10-26:
Merged to staging just now.

chelmuth, 2023-11-07:
I discovered 2c44a07a2224b0c1a37fcf4de7f2b50eeb216d8a breaks the USB driver on our x86_32 test machine: The fb_bench test fails quite often since 2023-10-27 with *NMI* (foc, pistachio) or *Task gate invoked* (nova). The platform is quite dated (see below) but I want to make 101% sure recent platforms are not affected.

> [ 0] CORE:00:00:0 6:17:6:0 [60c] Intel(R) Core(TM)2 Duo CPU     E7300  @ 2.66GHz
> [ 1] CORE:00:01:0 6:17:6:0 [60c] Intel(R) Core(TM)2 Duo CPU     E7300  @ 2.66GHz

Any ideas?

jschlatow, 2023-11-07:
> Any ideas?

No explanation, however, since the cache attribute is ignored by nova (I'm not sure about foc and pistacio though), I'd suspect this might be connected to using a different Mem_allocator of lx_kit. I have two ideas on how to narrow this down:
1. Change the Cache attribute of `Lx_kit::Env::memory` to UNCACHED. Do the problems vanish?
2. Revert 2c44a07 and change the Cache attribute of `Lx_kit::Env::uncached_memory` to CACHED. Does this cause the same issues?

chelmuth, 2023-11-07:
>     1. Change the Cache attribute of `Lx_kit::Env::memory` to UNCACHED. Do the problems vanish?

The test still fails.

>     2. Revert [2c44a07](https://github.com/genodelabs/genode/commit/2c44a07a2224b0c1a37fcf4de7f2b50eeb216d8a) and change the Cache attribute of `Lx_kit::Env::uncached_memory` to CACHED. Does this cause the same issues?

Nope, test works with this change.

Now I'm even more confused.

jschlatow, 2023-11-07:
That is indeed puzzling but (imo) reinforces my suspicion that it is not related to the Cache attribute. I'm wondering whether there is a buffer flow somewhere that's causing these errors. I was thinking about cherry-picking 775bb32 but since `Mem_allocator` uses relatively large DMA buffers as backing store for its allocations, we cannot be sure to catch all overflows. Instead, we'd need to (temporarily) change `Mem_allocator`'s implementation to allocate separate DMA buffers for every allocation.

jschlatow, 2023-11-07:
By the way, while looking into the dde_linux code, I found another place where uncached DMA buffers are being allocated. It's in _lx_emul/shadow/mm/dmapool.c_. I'll provide a fix for this as well.

jschlatow, 2023-11-07:
> By the way, while looking into the dde_linux code, I found another place where uncached DMA buffers are being allocated. It's in _lx_emul/shadow/mm/dmapool.c_. I'll provide a fix for this as well.

This is addressed by 8a93ef5

chelmuth, 2023-11-08:
> By the way, while looking into the dde_linux code, I found another place where uncached DMA buffers are being allocated. It's in _lx_emul/shadow/mm/dmapool.c_. I'll provide a fix for this as well.

Correct, I also caught this while debugging.

chelmuth, 2023-11-08:
Heureka, I fixed the bug in ebd2071813fa52f3f3dac99b86a198102a8ed55c.

jschlatow, 2023-11-08:
> Heureka, I fixed the bug in [ebd2071](https://github.com/genodelabs/genode/commit/ebd2071813fa52f3f3dac99b86a198102a8ed55c).

Marvelous!

nfeske, 2023-11-30:
Fixed in master.