genodelabs/genode #5002: platform_drv: add IOMMU support

jschlatow, 2023-09-22:
I have added basic Intel-IOMMU support to the platform driver. The work (in progress) is found at my [iommu branch](https://github.com/jschlatow/genode/commits/iommu). Even though there are a couple of improvements on my list to be implemented (e.g. fault event interrupts, queued invalidation support, better translation-table allocation), it should be fully functional on Intel devices.

So far, I have tested the implementation on an x260 and a t490s. At this stage, I'd like to invite everyone to give it a spin on their devices and let me know if there are any issues. You'll find "enabled IOMMU..." messages (and potentially error messages) from the platform driver in the log and a dump of IOMMU registers at _/report/drivers/iommu_.

For your convenience, I published a [sculpt-pc-2023-09-22.img](https://depot.genode.org/jschlatow/image/sculpt-pc-2023-09-22.img.xz) on my depot (not added to my index though). Use the ram_fs to deploy mesa_gears. The ram_fs is also populated with typical depot archives for which I added launchers. If you encounter any issues, please provide a dump of _/report_.

nfeske, 2023-09-22:
The image boots successfully on the Framework gen12 laptop. I see two IOMMU messages in the log ("enabled IOMMU drhd1" and "enabled IOMMU drhd0").

After starting the scenario by "using" the RAM fs, the system becomes somewhat unstable, with the leitzentrale UI getting stuck at various points. I could not spot suspicious messages in the log though.

jschlatow, 2023-09-22:
@nfeske Thanks for testing the image on the Framework laptop. Would you mind emailing me the _iommu_ report? I'm interested in the capabilities of the gen12 IOMMU.

jschlatow, 2023-11-02:
I updated my [iommu branch](https://github.com/jschlatow/genode/commits/iommu) and published a corresponding [sculpt-pc-2023-11-02.img](https://depot.genode.org/jschlatow/image/sculpt-pc-2023-11-02.img.xz) along with a 23.11 depot index.

While the previous version used a fixed 2MB backing store for page tables for every session, the current version allocates these on demand so that the RAM quota increases only marginally for non-DMA driver components.

I haven't noticed any issues with this version on my T490s and X260. I'd therefore consider it ready for a review and intensive testing.

Note that I only implemented DMA remapping and omitted interrupt remapping, which I'd like to approach next. There are also a couple of obvious optimisations I'd like to implement, namely:

- use guard pages for DMA buffers
- omit repeated context-cache and IOTLB invalidations on session destruction
- consolidate page-table-specific header files with base-hw

jschlatow, 2023-11-02:
Commit 775bb32 and df44685 implement guard pages resp. delayed context-cache and IOTLB invalidations.

chelmuth, 2023-11-02:
> Commit [775bb32](https://github.com/genodelabs/genode/commit/775bb322cfb1b9107b2b821095d34bd5597c6f4a) [...] implement[s] guard pages

May I ask for the rationale (resp. workings) of the guard pages? Are these just another RAM page at the end of the DMA region?

jschlatow, 2023-11-03:
> > Commit [775bb32](https://github.com/genodelabs/genode/commit/775bb322cfb1b9107b2b821095d34bd5597c6f4a) [...] implement[s] guard pages
> 
> May I ask for the rationale (resp. workings) of the guard pages? Are these just another RAM page at the end of the DMA region?

It's not a page (i.e. frame) of RAM but a free page in the virtual IO memory space between DMA buffers. This makes sure that a simple DMA buffer overflow does not corrupt other DMA buffers. Since it is only a matter of virtual address allocation and thus does not cost any additional RAM, I think it's worth implementing.

chelmuth, 2023-11-03:
Thanks for the clarification. For the record, I was only asking for more insights and not questioning your intentions ;-)

nfeske, 2023-11-03:
@jschlatow I just gave your 2023-11-02 image a quick try (btw, you missed to list the image in your index - I patched the index file locally). Unfortunately, the image fails to boot up on the Framework. I only see the splash screen of the boot loader. Maybe it's just the display driver. I'll try adding the report dump for another boot.

jschlatow, 2023-11-03:
@nfeske Thanks for the feedback. I updated my image index.

Can you please retry downloading the img.xz and dd-ing onto your thumb drive? I tried the update from within Sculpt and got into the same issue as you whereas the dd method works. I suspect the update method does not update the grub.cfg. Since I removed the "iommu" from nova's boot arguments, the grub.cfg must be updated accordingly.

nfeske, 2023-11-03:
> Can you please retry downloading the img.xz and dd-ing onto your thumb drive?

I'm happy to report that this version works flawlessly on my Framework laptop. :-)

>  suspect the update method does not update the grub.cfg. Since I removed the "iommu" from nova's boot arguments, the grub.cfg must be updated accordingly.

Ouch! This is indeed not considered by the update mechanism yet. We'll need to ponder how to provide a safe update/downgrade path then.

m-stein, 2023-11-06:
@jschlatow I just played around a little with your image (network, video, audio, vm, gpu, ...) and it works like a charm! Nothing to report :)

alex-ab, 2023-11-06:
I made as short boot up test and succeed for T420, T470p, TFuji3, Fuji5b, test-t490 and Fuji6.

The X201 stops with a scrambled screen. The vanilla 23.10 image works. The serial output I got:

```
Genode sculpt-23.10-32-g08cd4bfdea <local changes>
3775 MiB RAM and 63249 caps assigned to init
[init] Warning: runtime: configured RAM exceeds available RAM, proceed with 3533861616
Error: illegal READ at address 0xfffffffffffbb6a8 by pager_object: pd='init -> drivers -> usb_drv' thread='ep' ip=0x10a2f00
```

My AMD machines are not protected anymore because of the missing IOMMU support of the new platform driver, but this is expected.

chelmuth, 2023-11-06:
Just tested your image incl. WiFi on Framework Gen13 (i7-1360P) without flaws.

cnuke, 2023-11-06:
I booted the image on Fuji4 and tested `wired` and `wifi` and also did a `ls -lR` in the inspect window of the NVMe's file system. Everything worked fine AFAICT.

jschlatow, 2023-11-06:
> Ouch! This is indeed not considered by the update mechanism yet. We'll need to ponder how to provide a safe update/downgrade path then.

I think we need to forcibly disable the iommu support in NOVA so that any "iommu" boot option is ignored. I'll work on a corresponding patch.

jschlatow, 2023-11-06:
**Thank you all for testing the image on various devices!**

> The X201 stops with a scrambled screen. The vanilla 23.10 image works. 

That is strange because the illegal read is from the CPU and the address does not look like a DMA-buffer address. I need to ponder on how to debug this.

> My AMD machines are not protected anymore because of the missing IOMMU support of the new platform driver, but this is expected.

If you have time next week, we could have a look together at adding AMD support.

jschlatow, 2023-11-06:
@nfeske Commit e0df47e disables the "iommu" command line option for nova in order to provide a safe update path.

jschlatow, 2023-11-06:
@alex-ab The ip looks like the illegal READ occurs during backtrace. I've rebuild the pc_usb_host_drv with `-fno-omit-frame-pointer` and published a new image (2023-11-06). Could you give this a try?

The image also contains the latest optimisations (guard pages, delayed IOTLB invalidations) and can be updated from within Sculpt.

alex-ab, 2023-11-06:
> @alex-ab The ip looks like the illegal READ occurs during backtrace. I've rebuild the pc_usb_host_drv with `-fno-omit-frame-pointer` and published a new image (2023-11-06). Could you give this a [...]

With the new image the pagefault is gone, but X201 has the same scrambled output with block (top/down) of colorful lines and dark sections. On the USB stick are also no logs stored from report_dump. The same image works on Fuji5b. Maybe you may rebuild the image with LOG=core, so that you get potential more output, or ?

atopia, 2023-11-07:
Your image runs my work setup on the Thinkpad X1 Nano Gen2 (i7-1280P) flawlessly.

jschlatow, 2023-11-07:
@alex-ab I just published a new image (2023-11-07) which is built with LOG=core.

nfeske, 2023-11-07:
> Commit https://github.com/genodelabs/genode/commit/e0df47eb505fa8a61d2f276c7c6d015fdcaecb2d disables the "iommu" command line option for nova in order to provide a safe update path.

That's a sensible idea. Thanks for removing this obstacle for now. (I guess it will hit us again once we switch out the microkernel)

alex-ab, 2023-11-07:
> @alex-ab I just published a new image (2023-11-07) which is built with LOG=core.

Thanks, the log output is:
```
Genode sculpt-23.10-32-gc38d00b39a <local changes>[0m
3775 MiB RAM and 63249 caps assigned to init[0m
[init] [34mWarning: runtime: configured RAM exceeds available RAM, proceed with 3533869808[0m[0m
[init -> runtime] parent provides[0m[0m
...
[init -> runtime] monitor ready[0m[0m
[init -> drivers] [34mWarning: dynamic: configured RAM exceeds available RAM, proceed with 155493588[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> acpi_drv] Found MADT[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13[0m[0m
[init -> drivers -> acpi_drv] Found MCFG[0m[0m
[init -> drivers -> acpi_drv] MCFG BASE 0xe0000000 seg 0x0 bus 0x0-0xff[0m[0m
[init -> drivers -> acpi_drv] Found DMAR[0m[0m
[init -> drivers -> acpi_drv] 36 bit DMA physical addressable[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbhid[0m[0m
[init -> drivers -> usb_hid_drv] usbhid-hid-core: USB HID core driver[0m[0m
[init -> drivers -> usb_hid_drv] sched_clock: Marking stable (0, 5640000)->(6275000, -635000)[0m[0m
[init -> drivers -> usb_hid_drv] [34mWarning: use compatibility mode: will claim all HID devices from USB report[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] XSDT OEM 'LENOVO', table id 'TP-6Q   ', revision 5120, creator ' LTP' (0)[0m[0m
[init -> drivers -> acpi_drv] SMBIOS table (entry point: 0x1c6850 structures: 0xe0010)[0m[0m
[init -> drivers -> platform_drv] [31mError: IOMMU does not support 4-level page tables[0m[0m
[init -> drivers -> platform_drv] [31mError: IOMMU does not support 4-level page tables[0m[0m
[init -> drivers -> platform_drv] [31mError: IOMMU does not support 4-level page tables[0m[0m
[init -> drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)[0m[0m
[init -> drivers -> usb_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256[0m[0m
[init -> drivers -> usb_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> usb_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver genode[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd2[0m[0m
[init -> drivers -> platform_drv] [34mWarning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: EHCI Host Controller[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: new USB bus registered, assigned bus number 1[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd1[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: irq 19, io mem 0xf2728400[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: USB 2.0 started, EHCI 1.00[0m[0m
[init -> drivers -> usb_drv] hub 1-0:1.0: USB hub found[0m[0m
[init -> drivers -> usb_drv] hub 1-0:1.0: 3 ports detected[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv] MGGC_0_2_0_PCI[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Graphics_mode_select:               0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Gtt_graphics_memory_size:           0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Versatile_acceleration_mode_enable: 0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Igd_vga_disable:                    0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Ggc_lock:                           0x0[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: EHCI Host Controller[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: new USB bus registered, assigned bus number 2[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: irq 23, io mem 0xf2728000[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: USB 2.0 started, EHCI 1.00[0m[0m
[init -> drivers -> usb_drv] hub 2-0:1.0: USB hub found[0m[0m
[init -> drivers -> usb_drv] hub 2-0:1.0: 3 ports detected[0m[0m
[... cut ...]
[init -> runtime -> ram_fs] [31mError: 	name="depot.tar"[0m[0m
[init -> runtime] child "ram_fs" announces service "File_system"[0m[0m
[init -> drivers -> dynamic -> intel_fb_drv]     LVDS-1:  enable name=' 1280x800' id=1  mode=1280x 800@60  fb=1280x 800[0m[0m
[init -> drivers -> dynamic -> intel_fb_drv] framebuffer reconstructed - virtual=1280x800 physical=1280x800[0m[0m
[init -> drivers -> dynamic -> intel_fb_drv]     LVDS-1:  enable name=' 1280x800' id=1  mode=1280x 800@60  fb=1280x 800[0m[0m
```

jschlatow, 2023-11-07:
> [init -> drivers -> platform_drv] �[31mError: IOMMU does not support 4-level page tables�[0m�[0m

There it is. I was waiting for a device not supporting 4-level page tables.

jschlatow, 2023-11-07:
@alex-ab I published sculpt-pc-2023-11-07-a, which adds support for 3-level page tables.

alex-ab, 2023-11-08:
> @alex-ab I published sculpt-pc-2023-11-07-a, which adds support for 3-level page tables.

Thanks @jschlatow. X201 still shows scrambled output. The new log  ouput is as follows:
```
Genode sculpt-23.10-32-gc38d00b39a <local changes>[0m
3775 MiB RAM and 63249 caps assigned to init[0m
...
[init -> runtime] monitor ready[0m[0m
[init -> drivers] [34mWarning: dynamic: configured RAM exceeds available RAM, proceed with 155493588[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> acpi_drv] Found MADT[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13[0m[0m
[init -> drivers -> acpi_drv] Found MCFG[0m[0m
[init -> drivers -> acpi_drv] MCFG BASE 0xe0000000 seg 0x0 bus 0x0-0xff[0m[0m
[init -> drivers -> acpi_drv] Found DMAR[0m[0m
[init -> drivers -> acpi_drv] 36 bit DMA physical addressable[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbhid[0m[0m
[init -> drivers -> usb_hid_drv] usbhid-hid-core: USB HID core driver[0m[0m
[init -> drivers -> usb_hid_drv] sched_clock: Marking stable (0, 5706000)->(6412000, -706000)[0m[0m
[init -> drivers -> usb_hid_drv] [34mWarning: use compatibility mode: will claim all HID devices from USB report[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] XSDT OEM 'LENOVO', table id 'TP-6Q   ', revision 5120, creator ' LTP' (0)[0m[0m
[init -> drivers -> acpi_drv] SMBIOS table (entry point: 0x1c6850 structures: 0xe0010)[0m[0m
[init -> drivers -> platform_drv] [34mWarning: Requires explicit write-buffer flushing (not implemented)[0m[0m
[init -> drivers -> platform_drv] drhd0: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd0: Capability=0xc9008020e30272[0m[0m
[init -> drivers -> platform_drv] drhd0: Extended Capability=0x1000[0m[0m
[init -> drivers -> platform_drv] [34mWarning: Requires explicit write-buffer flushing (not implemented)[0m[0m
[init -> drivers -> platform_drv] drhd1: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd1: Capability=0xc0000020230272[0m[0m
[init -> drivers -> platform_drv] drhd1: Extended Capability=0x1000[0m[0m
[init -> drivers -> platform_drv] [34mWarning: Requires explicit write-buffer flushing (not implemented)[0m[0m
[init -> drivers -> platform_drv] drhd2: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd2: Capability=0xc9008020630272[0m[0m
[init -> drivers -> platform_drv] drhd2: Extended Capability=0x1000[0m[0m
[init -> drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)[0m[0m
[init -> drivers -> usb_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256[0m[0m
[init -> drivers -> usb_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> usb_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver genode[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd1[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault recording overflow[0m[0m
[init -> drivers -> platform_drv] [31mError: Faults records for drhd1[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault: hi=0x8000000300000010, reason=0x3, type=0x0, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0x10, info=0xbe000[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd2[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv] MGGC_0_2_0_PCI[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: EHCI Host Controller[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: new USB bus registered, assigned bus number 1[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: irq 19, io mem 0xf2728400[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Graphics_mode_select:               0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Gtt_graphics_memory_size:           0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Versatile_acceleration_mode_enable: 0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Igd_vga_disable:                    0x0[0m[0m
[... cut ...]
[init -> leitzentrale -> backdrop] [34mWarning: attempt to handle the same signal context twice (nested)[0m[0m
[init -> leitzentrale -> decorator] [34mWarning: attempt to handle the same signal context twice (nested)[0m[0m
[init -> drivers -> dynamic -> ahci_drv] 		#0: off (ATA)[0m[0m
[init -> drivers -> dynamic -> ahci_drv] 		#1:  off (unknown device signature)[0m[0m
[init -> drivers -> dynamic -> ahci_drv] 		#4:  off (unknown device signature)[0m[0m
[init -> drivers -> dynamic -> ahci_drv] 		#5:  off (unknown device signature)[0m[0m
```

jschlatow, 2023-11-08:
> [init -> drivers -> platform_drv] �[34mWarning: Requires explicit write-buffer flushing (not implemented)�[0m�[0m

@alex-ab Thanks for testing. This line is the culprit. I'll implement write-buffer flushing and provide a new image.

jschlatow, 2023-11-08:
@alex-ab I just published sculpt-pc-2023-11-08, which adds write-buffer-flushing support.

alex-ab, 2023-11-08:
@jschlatow: Thanks, tested, similar result:

```
Genode sculpt-23.10-32-gc38d00b39a <local changes>
...
[init -> drivers -> usb_hid_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> acpi_drv] Found MADT[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13[0m[0m
[init -> drivers -> acpi_drv] Found MCFG[0m[0m
[init -> drivers -> acpi_drv] MCFG BASE 0xe0000000 seg 0x0 bus 0x0-0xff[0m[0m
[init -> drivers -> acpi_drv] Found DMAR[0m[0m
[init -> drivers -> acpi_drv] 36 bit DMA physical addressable[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbhid[0m[0m
[init -> drivers -> usb_hid_drv] usbhid-hid-core: USB HID core driver[0m[0m
[init -> drivers -> usb_hid_drv] sched_clock: Marking stable (0, 5779000)->(6484000, -705000)[0m[0m
[init -> drivers -> usb_hid_drv] [34mWarning: use compatibility mode: will claim all HID devices from USB report[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] XSDT OEM 'LENOVO', table id 'TP-6Q   ', revision 5120, creator ' LTP' (0)[0m[0m
[init -> drivers -> acpi_drv] SMBIOS table (entry point: 0x1c6850 structures: 0xe0010)[0m[0m
[init -> drivers -> platform_drv] drhd0: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd0: Capability=0xc9008020e30272[0m[0m
[init -> drivers -> platform_drv] drhd0: Extended Capability=0x1000[0m[0m
[init -> drivers -> platform_drv] drhd1: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd1: Capability=0xc0000020230272[0m[0m
[init -> drivers -> platform_drv] drhd1: Extended Capability=0x1000[0m[0m
[init -> drivers -> platform_drv] drhd2: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd2: Capability=0xc9008020630272[0m[0m
[init -> drivers -> platform_drv] drhd2: Extended Capability=0x1000[0m[0m
[init -> drivers -> usb_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256[0m[0m
[init -> drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)[0m[0m
[init -> drivers -> usb_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> usb_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver genode[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd2[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd1[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault recording overflow[0m[0m
[init -> drivers -> platform_drv] [31mError: Faults records for drhd1[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault: hi=0x8000000300000010, reason=0x3, type=0x0, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0x10, info=0xbe000[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: EHCI Host Controller[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: new USB bus registered, assigned bus number 1[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: irq 19, io mem 0xf2728400[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv] MGGC_0_2_0_PCI[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Graphics_mode_select:               0x0[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: USB 2.0 started, EHCI 1.00[0m[0m
[init -> drivers -> usb_drv] hub 1-0:1.0: USB hub found[0m[0m
[init -> drivers -> usb_drv] hub 1-0:1.0: 3 ports detected[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Gtt_graphics_memory_size:           0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Versatile_acceleration_mode_enable: 0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Igd_vga_disable:                    0x0[0m[0m
[init -> drivers -> dynamic -> intel_gpu_drv]   Ggc_lock:                           0x0[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault recording overflow[0m[0m
[init -> drivers -> platform_drv] [31mError: Faults records for drhd1[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault: hi=0x8000000300000010, reason=0x3, type=0x0, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0x10, info=0xc0000[0m[0m
[... cut ...]
[init -> drivers -> dynamic -> intel_fb_drv]     LVDS-1:  enable name=' 1280x800' id=1  mode=1280x 800@60  fb=1280x 800[0m[0m
[init -> drivers -> dynamic -> intel_fb_drv] framebuffer reconstructed - virtual=1280x800 physical=1280x800[0m[0m
[init -> drivers -> dynamic -> intel_fb_drv]     LVDS-1:  enable name=' 1280x800' id=1  mode=1280x 800@60  fb=1280x 800[0m[0m
[init -> leitzentrale -> backdrop] [34mWarning: attempt to handle the same signal context twice (nested)[0m[0m
[init -> leitzentrale -> decorator] [34mWarning: attempt to handle the same signal context twice (nested)[0m[0m
```

jschlatow, 2023-11-08:
Thanks for testing. The fault records are a good indicator. The fault type/reason 0x3 indicates invalid programming of context-table entries. I'll look into that.

jschlatow, 2023-11-08:
@alex-ab I missed adapting the context-table entries appropriately when adding support for 3-level page tables. I just published sculpt-pc-2023-11-08-a, which contains the fix.

alex-ab, 2023-11-09:
> @alex-ab I missed adapting the context-table entries appropriately when adding support for 3-level page tables. I just published sculpt-pc-2023-11-08-a, which contains the fix.

There is no new image in the index file. On the depot.genode.org is also no new image file. Probably you forgot to upload it ?

jschlatow, 2023-11-09:
> There is no new image in the index file. On the depot.genode.org is also no new image file. Probably you forgot to upload it ?

Indeed. I uploaded it just now.

alex-ab, 2023-11-09:
@jschlatow: Optical it became a bit better, the upper part of the display is scrambled, the lower part shows the Leitzentrale and the Genode logo. The log output is:

```
Genode sculpt-23.10-32-gc38d00b39a <local changes>[0m
3775 MiB RAM and 63249 caps assigned to init[0m
[init] [34mWarning: runtime: configured RAM exceeds available RAM, proceed with 3533865712[0m[0m
[init -> runtime] parent provides[0m[0m
[init -> runtime]   service "ROM"[0m[0m
[init -> runtime]   service "PD"[0m[0m
[init -> runtime]   service "RM"[0m[0m
[init -> runtime]   service "CPU"[0m[0m
[init -> runtime]   service "LOG"[0m[0m
[init -> runtime]   service "IO_MEM"[0m[0m
[init -> runtime]   service "IO_PORT"[0m[0m
[init -> runtime]   service "IRQ"[0m[0m
[init -> runtime]   service "Gui"[0m[0m
[init -> runtime]   service "Timer"[0m[0m
[init -> runtime]   service "Block"[0m[0m
[init -> runtime]   service "Report"[0m[0m
[init -> runtime]   service "Platform"[0m[0m
[init -> runtime] monitor ready[0m[0m
[init -> drivers] [34mWarning: dynamic: configured RAM exceeds available RAM, proceed with 155493588[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> acpi_drv] Found MADT[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 0 -> GSI 2 flags: 0[0m[0m
[init -> drivers -> acpi_drv] MADT IRQ 9 -> GSI 9 flags: 13[0m[0m
[init -> drivers -> acpi_drv] Found MCFG[0m[0m
[init -> drivers -> acpi_drv] MCFG BASE 0xe0000000 seg 0x0 bus 0x0-0xff[0m[0m
[init -> drivers -> acpi_drv] Found DMAR[0m[0m
[init -> drivers -> acpi_drv] 36 bit DMA physical addressable[0m[0m
[init -> drivers -> usb_hid_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> usb_hid_drv] usbcore: registered new interface driver usbhid[0m[0m
[init -> drivers -> usb_hid_drv] usbhid-hid-core: USB HID core driver[0m[0m
[init -> drivers -> usb_hid_drv] sched_clock: Marking stable (0, 5813000)->(6519000, -706000)[0m[0m
[init -> drivers -> usb_hid_drv] [34mWarning: use compatibility mode: will claim all HID devices from USB report[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=0[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] DMA remapping structure type=1[0m[0m
[init -> drivers -> acpi_drv] XSDT OEM 'LENOVO', table id 'TP-6Q   ', revision 5120, creator ' LTP' (0)[0m[0m
[init -> drivers -> acpi_drv] SMBIOS table (entry point: 0x1c6850 structures: 0xe0010)[0m[0m
[init -> drivers -> platform_drv] drhd0: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd0: Capability=0xc9008020e30272[0m[0m
[init -> drivers -> platform_drv] drhd0: Extended Capability=0x1000[0m[0m
[init -> drivers -> platform_drv] drhd1: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd1: Capability=0xc0000020230272[0m[0m
[init -> drivers -> platform_drv] drhd1: Extended Capability=0x1000[0m[0m
[init -> drivers -> platform_drv] drhd2: Version=1.0[0m[0m
[init -> drivers -> platform_drv] drhd2: Capability=0xc9008020630272[0m[0m
[init -> drivers -> platform_drv] drhd2: Extended Capability=0x1000[0m[0m
[init -> drivers -> usb_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256[0m[0m
[init -> drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)[0m[0m
[init -> drivers -> usb_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns[0m[0m
[init -> drivers -> usb_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver usbfs[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver hub[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new device driver usb[0m[0m
[init -> drivers -> usb_drv] time-clocksource: Switched to clocksource dde_counter[0m[0m
[init -> drivers -> usb_drv] usbcore: registered new interface driver genode[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd2[0m[0m
[init -> drivers -> platform_drv] enabled IOMMU drhd1[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault recording overflow[0m[0m
[init -> drivers -> platform_drv] [31mError: Faults records for drhd1[0m[0m
[init -> drivers -> platform_drv] [31mError: Fault: hi=0x8000000700000010, reason=0x7, type=0x0, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0x10, info=0xbe000[0m[0m
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: EHCI Host Controller[0m[0m
[... 201 lines cut ...]
[init -> runtime]   ELF binary: chroot[0m[0m
[init -> runtime]   priority:   3[0m[0m
[init -> runtime]   provides service File_system[0m[0m
[init -> runtime] child "report_dump"[0m[0m
[init -> runtime]   RAM quota:  81672K[0m[0m
[init -> runtime]   cap quota:  966[0m[0m
[... cut ...]
[init -> runtime] [34mWarning: depot: could not revert session RAM quota (service=File_system cid=<unconstructed> args='root="/", writeable=0, tx_buf_size=4194304, cap_quota=12, label="depot_rom -> ", ram_quota=4259021, diag=0' state=CLOSE_REQUESTE[0m
[init -> runtime] D ram_quota=4259021, cap_quota=12)[0m[0m
[init -> runtime] [34mWarning: usb-1-3.drv: could not revert session RAM quota (service=Block cid=<unconstructed> args='tx_buf_size=4194304, cap_quota=5, label="usb-1-3.part_block -> ", ram_quota=4207821, diag=0' state=CLOSE_REQUESTED ram_quota=420[0m
[init -> runtime] 7821, cap_quota=5)[0m[0m
```

jschlatow, 2023-11-09:
According to the values of the capability registers, the device does not support large pages (2MB and 1GB). This perfectly explains the various IOMMU faults of types 0x7 (failure to resolve the address to a page table in a preceding paging entry) and 0xC (non-zero reserved fields in a paging entry).

jschlatow, 2023-11-09:
@alex-ab Yet another image: sculpt-pc-2023-11-09

alex-ab, 2023-11-10:
@jschlatow: great! The graphical output is now working. Still some IOMMU faults in the log. The USB driver dies and is not working. I tested additional network and wifi, works :) and used the ahci disk to download packages, also works fine.

```
[init -> drivers -> platform_drv] drhd0: Version=1.0
[init -> drivers -> platform_drv] drhd0: Capability=0xc9008020e30272
[init -> drivers -> platform_drv] drhd0: Extended Capability=0x1000
[init -> drivers -> platform_drv] drhd1: Version=1.0
[init -> drivers -> platform_drv] drhd1: Capability=0xc0000020230272
[init -> drivers -> platform_drv] drhd1: Extended Capability=0x1000
[init -> drivers -> platform_drv] drhd2: Version=1.0
[init -> drivers -> platform_drv] drhd2: Capability=0xc9008020630272
[init -> drivers -> platform_drv] drhd2: Extended Capability=0x1000
[init -> drivers -> usb_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256
[init -> drivers -> ps2_drv] Using keyboard with scan code set 1 (xlate)
[init -> drivers -> usb_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[init -> drivers -> usb_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[init -> drivers -> usb_drv] usbcore: registered new interface driver usbfs
[init -> drivers -> usb_drv] usbcore: registered new interface driver hub
[init -> drivers -> usb_drv] usbcore: registered new device driver usb
[init -> drivers -> usb_drv] time-clocksource: Switched to clocksource dde_counter
[init -> drivers -> usb_drv] usbcore: registered new interface driver genode
[init -> drivers -> platform_drv] enabled IOMMU drhd2
[init -> drivers -> platform_drv] Warning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)
[init -> drivers -> platform_drv] Warning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)
[init -> drivers -> platform_drv] Warning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: EHCI Host Controller
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: new USB bus registered, assigned bus number 1
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: irq 19, io mem 0xf2728400
[init -> drivers -> usb_drv] ehci-pci 00:1d.0: USB 2.0 started, EHCI 1.00
[init -> drivers -> usb_drv] hub 1-0:1.0: USB hub found
[init -> drivers -> usb_drv] hub 1-0:1.0: 3 ports detected
[init -> drivers -> platform_drv] Warning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)
[init -> drivers -> platform_drv] Warning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: EHCI Host Controller
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: new USB bus registered, assigned bus number 2
[init -> drivers -> platform_drv] Warning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)
[init -> drivers -> platform_drv] enabled IOMMU drhd1
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: irq 23, io mem 0xf2728000
[init -> drivers -> usb_drv] ehci-pci 00:1a.0: USB 2.0 started, EHCI 1.00
[init -> drivers -> platform_drv] Error: Fault recording overflow
[init -> drivers -> platform_drv] Error: Faults records for drhd1
[init -> drivers -> platform_drv] Error: Fault: hi=0x8000000500000010, reason=0x5, type=0x0, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0x10, info=0xffffff
[init -> drivers -> usb_drv] hub 2-0:1.0: USB hub found
[init -> drivers -> usb_drv] hub 2-0:1.0: 3 ports detected
[init -> drivers -> usb_drv] sched_clock: Marking stable (1984546000, 46739000)->(2073173000, -41888000)
[init -> drivers -> dynamic -> intel_gpu_drv] MGGC_0_2_0_PCI
[init -> drivers -> dynamic -> intel_gpu_drv]   Graphics_mode_select:               0x0
[init -> drivers -> dynamic -> intel_gpu_drv]   Gtt_graphics_memory_size:           0x0
[init -> drivers -> dynamic -> intel_gpu_drv]   Versatile_acceleration_mode_enable: 0x0
[init -> drivers -> dynamic -> intel_gpu_drv]   Igd_vga_disable:                    0x0
[init -> drivers -> dynamic -> intel_gpu_drv]   Ggc_lock:                           0x0
[init -> drivers -> usb_drv] usb 1-1: new high-speed USB device number 2 using ehci-pci
[init -> drivers -> platform_drv] Warning: Performed IOTLB invalidation with different granularity (requested=2, actual=1)
[init -> drivers -> dynamic -> intel_gpu_drv] Warning: No supported Intel GPU detected - no GPU service
[init -> drivers -> platform_drv] Error: Fault recording overflow
[init -> drivers -> platform_drv] Error: Faults records for drhd1
[init -> drivers -> platform_drv] Error: Fault: hi=0x8000000500000010, reason=0x5, type=0x0, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0x10, info=0xffffff
[init -> drivers -> dynamic -> intel_fb_drv] --- Intel framebuffer driver started ---
[init -> drivers -> dynamic -> intel_fb_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256
[init -> drivers -> dynamic -> intel_fb_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[init -> drivers -> dynamic -> intel_fb_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[init -> drivers -> dynamic -> intel_fb_drv] time-clocksource: Switched to clocksource dde_counter
[init -> drivers -> dynamic -> intel_fb_drv] disabling PPGTT to avoid GPU code paths
[... cut ...]
[init -> drivers -> platform_drv] Error: Fault: hi=0xc0000006000000e8, reason=0x6, type=0x1, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0xe8, info=0x0
[init -> drivers -> usb_drv] usb 1-1: USB disconnect, device number 2
[init -> drivers -> usb_drv] usb 1-1.2: USB disconnect, device number 3
[init -> runtime -> usb-1-3.drv] Device unplugged
[init -> runtime -> usb-1-3.drv] Error: complete error: packet not succeded
[init -> runtime -> usb-1-3.drv] Error: request pending: tag: 282 read: 1 buffer: 0x807200 lba: 16802088 size: 1048576
[init -> leitzentrale -> manager] Warning: sculpt partition unexpectedly vanished
```

jschlatow, 2023-11-10:
Now its getting tricky. Apparently, the GPU tries writing to a page at `0xffffff000` and the usb driver tries reading from `0x0`. I think I need to have a look at the device next week when I'm in the office.

chelmuth, 2023-11-10:
> ```
> [init -> drivers -> usb_drv] ehci-pci 00:1d.0: fatal error
> [init -> drivers -> usb_drv] ehci-pci 00:1d.0: HC died; cleaning up
> [init -> drivers -> platform_drv] Error: Faults records for drhd2 [...]

I can't interpret the `Fault` but the host controller also died without ebd2071813fa52f3f3dac99b86a198102a8ed55c if the DMA buffer crossed a 4K boundary without triggering *NMI/Task Gate* in many cases.

jschlatow, 2023-11-10:
> I can't interpret the `Fault` but the host controller also died without [ebd2071](https://github.com/genodelabs/genode/commit/ebd2071813fa52f3f3dac99b86a198102a8ed55c) if the DMA buffer crossed a 4K boundary without triggering _NMI/Task Gate_ in many cases.

That's a good hint and worth a try. I cherry-picked the commit and published a new image as sculpt-pc-2023-11-10.

alex-ab, 2023-11-10:
> That's a good hint and worth a try. I cherry-picked the commit and published a new image as sculpt-pc-2023-11-10.

With the new image USB is working. In the log are now just the same drhd1 fault entries.

cnuke, 2023-11-14:
I tested the image on another machine, DS57U with Celeron 3205U (Broadwell), where the IOMMU worked fine :+1:.
```
[init -> drivers -> platform_drv] drhd0: Version=1.0
[init -> drivers -> platform_drv] drhd0: Capability=0x1c0000c40660462
[init -> drivers -> platform_drv] drhd0: Extended Capability=0x7e1ff0505e
[init -> drivers -> platform_drv] drhd1: Version=1.0
[init -> drivers -> platform_drv] drhd1: Capability=0xd2008c20660462
[init -> drivers -> platform_drv] drhd1: Extended Capability=0xf010da
[…]
[init -> drivers -> platform_drv] enabled IOMMU drhd0
[init -> drivers -> platform_drv] enabled IOMMU drhd1
```

I noticed a fault when re-enabling the Atheros AR9462 wireless card but that has happend before as well and the card worked fine nonetheless.

```
[init -> drivers -> platform_drv] Error: Fault recording overflow
[init -> drivers -> platform_drv] Error: Faults records for drhd1
[init -> drivers -> platform_drv] Error: Fault: hi=0x8000000500000400, reason=0x5, type=0x0, AT=0x0, EXE=0x0, PRIV=0x0, PP=0x0, Source=0x400, info=0x4d4
```

(I also stumpled upon what I assume is the issue @chelmuth descripted yesterday in the `usb_host` driver

```
[init -> drivers -> usb_drv] Error: lx_emul_mem_free called with invalid ptr 0x9e0
```

The message was printed after expanding the _GENODE_ partition on the USB stick.)

jschlatow, 2023-11-22:
I just squashed the recent fixup commits, implemented default mappings to allow access to reserved memory while keeping the IOMMU disabled at all times, rebased my iommu branch on current staging, and published the corresponding sculpt image as sculpt-pc-2023-11-22.

The added functionality should not change much, yet, I will be grateful if you could give the most interesting devices (very old and very new ones) a quick test run.

chelmuth, 2023-11-23:
[sculpt-pc-2023-11-22](https://depot.genode.org/jschlatow/image/sculpt-pc-2023-11-22.img.xz) works on the following devices.

- FUJITSU LIFEBOOK U7413 (i7-1355U)
- Framework Gen13 FRANPECP06  (i7-1360P)

atopia, 2023-11-23:
Works on the Lenovo Thinkpad X1 Nano Gen2 (i7-1280P).

nfeske, 2023-11-23:
Works without a hitch on my Gen12 Framework laptop.

alex-ab, 2023-11-24:
The X201 works fine.

jschlatow, 2023-11-24:
With the recent cosmetic fixups, I would like to close up this issue as the Intel-specific DMA remapping is fully functional for the entire range of devices we have at hand.

I would like to address interrupt remapping in a separate issue with a different time frame. Although I have a work-in-progress implementation for MSI/MSI-X, this still needs pass-through of compatibility format interrupts to be enabled. The reason for this is that the global interrupt controller is managed by the kernel so that we cannot program its redirection table entries properly. Legacy interrupts are thus still in compatibility format. I don't see any benefit in applying interrupt remapping solely for MSI/MSI-X as preventing arbitrary compatibility format interrupts is the main motivation for interrupt remapping.

chelmuth, 2023-11-24:
Thanks for the clean series. I fully agree with your rationale regarding interrupt remapping. Therefore, I completed a first review of the commits and merged the series to staging. I'm going to have a closer look at 3348bacb277dc963dd135c74a51008c239cf4940 on Monday.

@skalk Please review the platform_drv adaptions.
@alex-ab Do you agree with the base-nova changes?

chelmuth, 2023-11-27:
I finished the review and have no further remarks beside the comment about `assert()` and some cosmetic fixes in add80355b3b560ef2aa1ebc7eb2ea0bc9e5bb575.

alex-ab, 2023-11-27:
> @skalk Please review the platform_drv adaptions. @alex-ab Do you agree with the base-nova changes?

No, I can not agree to some of the changes of commit "base-nova: disable iommu" and "platform_drv: remove device_pd".
- The commits are too early in the sense, that they disable a well working feature and replaces it with an (great!) intermediate state.
- The commits are developer commits and should stay on developer branches. As soon as the feature is complete, the right time will come to apply them to Genode master.
- Nevertheless applying them today means that users and customers get less protection than before, which is hardly understandable which will use this Genode release as an update.
- Regarding "base-nova: disable iommu", the patch at the kernel is not necessary to disable the usage by the kernel, the command-line change is sufficient. The shortcoming of the user-level Sculpt update mechanism does not justify a kernel patch per se.

chelmuth, 2023-11-27:
> No, I can not agree to some of the changes of commit "base-nova: disable iommu" and "platform_drv: remove device_pd".

Please give some more details about your concerns below (and keep in mind the nature of a non-Sculpt release) to help us understand the severity of your statements. I must admit your reaction caught me somewhat flat-footed because of its vehemence. I still think we can merge the series but are open for discussion.

> * The commits are too early in the sense, that they disable a well working feature and replace it with a (great!) intermediate state.

What concrete functionality is missing to render the current state *intermediate* beside the documented exclusion of interrupt remapping? 

> * The commits are developer commits and should stay on developer branches. As soon as the feature is complete, the right time will come to apply them to Genode master.

I can't follow your statement of *developer commits*. @jschlatow wrote: *With the recent cosmetic fixups, I would like to close up this issue [...]* which was for me a clear sign to review and merge the commits.

> * Nevertheless applying them today means that users and customers get less protection than before, which is hardly understandable by whom who will use this Genode release as an update.

Please clearly state *less protection* excl. interrupt remapping or if this is the actual point, correct @jschlatow's statements.

> * Regarding "base-nova: disable iommu", the patch at the kernel is not necessary to disable the usage by the kernel, the command-line change is sufficient. The shortcoming of the user-level Sculpt update mechanism does not justify a kernel patch per se.

Do you have any alternative approach in mind?

alex-ab, 2023-11-27:
> What concrete functionality is missing to render the current state _intermediate_ beside the documented exclusion of interrupt remapping?

- as stated by @jschlatow the interrupt remapping is missing and thereby feature wise not equivalent to the former Genode release

> Please clearly state _less protection_ excl. interrupt remapping or if this is the actual point, correct @jschlatow's statements.

- I am referring to the interrupt remapping.

> Do you have any alternative approach in mind?

- Obviously, keep the full feature version intact and add the new (again great!) version of @jschlatow as an option until it is feature complete.

skalk, 2023-11-27:
I've finished my review. Thank you @jschlatow for your impressive commit series, it is a pleasure to read it. I do not have any objections.

jschlatow, 2023-11-27:
@alex-ab I've reverted the questionable commits and pushed a few fixup commits. Now, NOVA's device PD is used by default. By removing the _iommu_ boot option and changing the acpi_drv's config to `ignore_drhd="no"` one can enable the platform driver's DMA remapping support.

@chelmuth Thanks for your review. I've addressed your comment about assert in 921a8cd. Would you rather merge d0bce51 as a revert commit or should I split this up into fixup commits so they can be cleanly squashed?

@skalk Thanks for your review.

nfeske, 2023-11-27:
@jschlatow I merged the commits w/o review from your iommu branch to staging to let you automated test chew on them.

jschlatow, 2023-11-27:
@chelmuth I pushed a small fixup that corrects a minor flaw that came with the default mappings support: There was no log message anymore that indicated the enablement of the IOMMU.

jschlatow, 2023-11-27:
@alex-ab At dinner, I had a flash of inspiration w.r.t. how we can address the concerns you raised earlier while making the platform-driver managed IOMMU the default. By allowing NOVA to boot with IOMMU enabled but letting the platform driver take ownership, we can actually provide a smooth upgrade path, keep the AMD IOMMU support in NOVA active, and let NOVA still manage interrupt remapping while the platform driver takes care of DMA remapping.

I couldn't wait and implemented this on my iommu_boot_enabled branch. Works like a charm on my devices. I also uploaded a Sculpt image (2023-11-27) to my depot for a quick test run.

@alex-ab Would you be okay with this solution? Do you have any objections against 4f241cc?

@chelmuth The main commits are 28d4a8c and 4f241cc. The remaining commits are fixups and reverts of the commits I added earlier.

jschlatow, 2023-11-28:
> The remaining commits are fixups and reverts of the commits I added earlier.

I have to correct myself: In order to use NOVA's AMD IOMMU support, we must, of course, keep the device_pd related code in the platform driver. I am going to do a test run without e55019c and da5819d, and provide a corresponding sculpt image.

jschlatow, 2023-11-28:
I force pushed my _iommu_boot_enabled_ branch and published the corresponding sculpt-pc-2023-11-28.img.

I'm pleased with the result since I haven't thought about keeping both implementations (NOVA and ours) active before. As it turned out, they combine very well and provide (imo) a solid interim solution until we achieve kernel-agnostic interrupt remapping. 

The way this works out in the platform driver is as follows: The platform driver uses the device PD as a default if the platform info indicates IOMMU support and if there is no `<io_mmu/>` node for a particular device. As soon as the devices ROM mentions the drhd devices found by the acpi_drv, the platform driver will control the IOMMU directly and not use the device PD mechanism. The interrupt remapping table is maintained by NOVA irrespective of the device PD. The Intel IOMMU implementation in NOVA solely modifies the remapping table, i.e. it does not issue any IOMMU commands (e.g. invalidations). This allows the platform driver to take over control over the IOMMU.

[17 more replies]

closed 2023-11-30 by nfeske