genodelabs/genode #5003: sntp_client: support domain name resolution

a-dmg, 2023-09-25:
Dear Genodians,

I would like to propose an enhancement for the 'sntp_client', which consists of supporting domain name resolution instead of providing a fixed IP address for the destination SNTP server. The following changes are:

 - New class 'Dns_request' in os/include/net to handle DNS requests as UDP datagram and appropriate testing
 - New configuration attribute 'dst_addr', which accepts either an IP address or a domain name
 - New xsd type 'Net_address', which allows either an IP address or a domain name
 - Minor updates of the sntp_client/README

I hope my approach is reasonable and satisfying. Thanks for your time and feedback.

Cheers,
Alice

skalk, 2023-09-26:
Hi Alice,
thank you for your contribution! It is very much appreciated. When reading through the list, I was wondering why did you replace the `dst_ip` configuration parameter by the more versatile `dst_addr` parameter. In my eyes it is better to keep the precise meaning of an IP address, and just add a parameter like `dst_url` in addition. Thereby you leave it backward compatible in addition, and there if no need to adapt 3rd party packages using the sntp client.
Best regards,
Stefan

rite, 2023-09-26:
> I was wondering why did you replace the dst_ip configuration parameter by the more versatile dst_addr parameter.

Actually it was me who suggested this change in order to avoid ambiguities when both attributes are configured. However, I'm absolutely fine if you prefer to keep the existing `dst_ip` and add an alternative `dst_url`-attribute. In fact my argument regarding ambiguities would become obsolete anyways in case the `sntp_client` will support the configuration of multiple SNTP servers sometime (either by IPs, hostnames or mixed).

chelmuth, 2023-09-27:
Thanks for the contribution. I'd just like to note that I'd refrain from using `dst_url` as attribute name. A URL always contains a *scheme/protocol* in addition to the *network location* and, thus, requires attribute values like `ntp://pool.ntp.org`. As long as we do not want to go down that rabbit hole, I vote for `dst_addr` that supports DNS host names and direct IP equally.

skalk, 2023-09-27:
@chelmuth I see you point. @a-dmg @rite So please, ignore my objection.

a-dmg, 2023-09-28:
Thank you all for your feedback! @m-stein @skalk @chelmuth 
First, here are the changes I have committed:
 - The Domain_name class now use a Genode::String<> as member
 - I have adjusted the modified source files to be aligned with Genode's coding style. Hopefully, I have not introduced more problems...
 - I have withdrawn from using C++ bitsets, and now using Genode::Resiter<>

Also, please note that I have squashed the following commit:
 - [xsd/net_types.xsd: add Net_address type](https://github.com/a-dmg/genode/commit/95dea7b75a034cb10cda5d7917ba8e8400d675f9)
 - [sntp_dummy_rtc: adjust configuration to use pool.ntp.org](https://github.com/a-dmg/genode/commit/f5e2b626b3e88c6451cb68be8aee772584cd3d4a)

 Into:
 - [sntp_client: add DNS lookup option](https://github.com/a-dmg/genode/commit/5261e5347cc5bbafbe95f87f61ea70c6a4d1ad15)

chelmuth, 2023-10-02:
Thanks for the update @a-dmg. I plan the merge not until after the upcoming Sculpt release.

chelmuth, 2023-10-26:
Merged as 22c77607db3a1674b8ca032f3bd2094d6923b3d2 and b203f20871c9b2fe3fed0925110f331c2f804c7c (all sntp_client commits squashed).