genodelabs/genode #5009: core: extend system management functionality to be available per CPU

alex-ab, 2023-09-28:
On x86 machine specific registers (MSR) and their privileged instructions (rdmsr, wrmsr) are used to steer various system functionality which is just available to the kernel.

Some MSRs provide functionality to monitor system state (e.g. temperature, frequency) and adjust the power consumption (e.g Intel HWP or AMD pstate), which one may want to adjust/read out dynamically, e.g. on Sculpt OS. Depending on the MSR, see Intel/AMD specification, the instruction must be called per SMT thread (hyper_thread), or just per CPU core, or per CPU packet to be effective.

On Genode, by now, we have the Pd_session::managing_system() as a guarded privileged call on imx8 by the platform driver and on x86 for ACPI suspend functionality. This functionality is solely available on the boot CPU by now.

By extending the PD managing interface per Genode Affinity::Location, we will be able to leverage the MSR functionality per CPU core resp. per SMT thread when supported by the kernel.

alex-ab, 2023-09-28:
@nfeske: my issue_5009 contains the commits. I added the discussed change of this morning as fixup.

nfeske, 2023-09-29:
While testing, I came up with fixup https://github.com/genodelabs/genode/commit/1f2776ba9a882730040634f4386c89fa9dbf287c.

closed 2023-10-04 by chelmuth

alex-ab, 2023-10-04:
@chelmuth: on issue_5009 the both top most commits add the msr access for core on nova

commit 856c4ac6 (2023-09-28): base: extend PD session by system_control_cap

Per Affinity::Location a system control cap can be requested. The capability
provides an RPC interface to request and set Cpu_state, as provided by the
former Pd::managing_system(Cpu_state) method. Invocation of those system
control capabilities then *can* (see below) be executed on the desired CPU
as described by Affinity::Location.

The system control cap will be invalid for kernels that don't support
system_control/managing_system functionality at all.

The system control cap will be ever by the same, e.g. ignoring the
Affinity::Location parame [... cut]