genodelabs/genode #5012: sculpt: use debug monitor as runtime init

cproc, 2023-09-29:
Using the debug monitor as runtime init makes it possible to debug Sculpt components.

cproc, 2023-09-29:
Commit 9b5b20d replaces the Sculpt runtime init with the debug monitor in combination with the `terminal_crosslink` component and adds a menu with debug options ("monitor this component", "wait for GDB" and "map executable segments writeable") to the component launch dialog.

nfeske, 2023-09-29:
Very cool!

closed 2023-10-04 by chelmuth

nfeske, 2023-10-19:
Commit https://github.com/genodelabs/genode/commit/a96841dd4efe8850cbcec2ab5e538c21525ae5fb implements several tweaks of the debug dialog. @cproc could you please give the new version a test?

Besides textual changes, I moved the `monitor` checkbox right at the top level (a suggestion by @chelmuth). I also removed the implicit enabling of `wx` when clicking on `wait`. Instead, `wait` is only presented when `wx` is selected, like it was already done for the `monitor` selection.

While working on the commit, I wondered why the `wait` option depends on `wx` in the first place? Isn't `wait` simply deferring the thread startup (interplay with the CPU session)? Why are the executable text segments needed for this? Or is this dependency only there to avoid user mistakes?

cproc, 2023-10-19:
The new version seems to work as well. The `wait` option is implemented with a software breakpoint at the first instruction, which needs the write permission.

nfeske, 2023-10-19:
Thanks for testing and the clarification.

commit 3179445a (2023-09-29): sculpt: use debug monitor as runtime init

Fixes #5012