genodelabs/genode #5015: depot: provide debug symbols

cproc, 2023-09-29:
Currently, depot packages usually contain stripped binaries, but additional debug symbols would be helpful when debugging Sculpt components with GDB.

cproc, 2023-09-29:
Commit 3d4cfed implements a simple solution which creates `dbg` packages with the binaries from the `debug` build directories in addition to the `bin` packages.

It might be possible to extract the debug symbols from these binaries to reduce the size of the packages, but I don't know yet if that would make it more difficult for GDB to find the symbols. When I build the `sculpt_distribution` and `sculpt_distribution-pc` packages, the `bin` packages are about 150M in size and the `dbg` packages about 1G (the top 3 in size are qt5_base, vbox6 and mesa), so extracting the debug symbols might save about 15% of the size.

chelmuth, 2023-09-29:
The commit looks quite convenient but the doubled archiving of the binary data seems worth some more effort. How about the approach described at https://github.com/GabrielMajeri/separate-symbols?

cproc, 2023-10-13:
Commit 64c79fa uses the suggested debug link method with separate debug info files. To be able to find these files, GDB either needs to be told the corresponding paths with the `set debug-file-directory` command or there could be a temporary directory with symlinks to all involved binaries and debug info files in one place and the GDB user can select this directory with the `cd` command and then everything would be available in the current directory, which is also the default `solib-search-path`.

Unfortunately, GDB currently has a bug, though ,which prevents it from automatically finding the application binary's debug info file in the `dbg` depot package with a symlink to the application binary in the `bin` depot package in the temporary directory, because in this case GDB resolves the binary's symlink first and then looks for the debug info file in the `bin` depot package [1]. I did not see this problem with shared libraries so far.

[1] https://sourceware.org/bugzilla/show_bug.cgi?id=24143

cproc, 2023-10-18:
Commit 02bb382 creates separate debug info files in the Genode build system with symlinks for both the stripped binary and the corresponding `.debug` file in the `debug/` directory. I kept the `.stripped` suffix for stripped binaries in the build directory to keep them distinguishable from the original unstripped binaries in `make` rules. I also added more `ifdef SHARED_LIB` tests in `base/mk/lib.mk` because `make` complained about a circular `.debug : .debug` rule, for example, when a static library was built and both `LIB_SO` and `DEBUG_SO` were empty strings.

chelmuth, 2023-10-26:
I tested the patch on current staging but always get warnings like follows.

```
> make -C build/x86_64 test/log
...
checking library dependencies...
  Library ldso_so_support
    COMPILE  so_support.o
    MERGE    ldso_so_support.lib.a
.../repos/base/mk/lib.mk:265: warning: overriding recipe for target '.debug'
.../repos/base/mk/lib.mk:252: warning: ignoring old recipe for target '.debug'
make[2]: Circular .debug <- .debug dependency dropped.
```

Any chance to prvent them?

cproc, 2023-10-26:
Commit a275580 should fix the circular dependency problem.

cproc, 2023-11-01:
Fixup commit 5465750 removes the debug info files on `make clean`.

chelmuth, 2023-11-01:
What is the background of extract_api_artifacts (resp. why does `-delete` not work)?

After a look into *build.mk* I wonder how the all the clean rules concur in case of multiple make jobs. For example, shouldn't `clean_debug_dir` depend on `clean_targets` and `clean_libcache`?

cproc, 2023-11-01:
Commit fd9301f removes the actual debug info files and not only the symlinks in the `debug` directory and commit 822e783 fixes the clean rule dependencies in `build.mk`.

chelmuth, 2023-11-01:
Merged a9a460508100aa62c423f887148ca8056081d970 and 3ebaffb5be57ebd30f7111e1f30436c9f6179f57.

cproc, 2023-11-09:
Should the `dbg` packages always be built together with the `bin` packages or should it be optional, for example with an argument CREATE_DBG=1 or similar?

cproc, 2023-11-09:
Or maybe more like `DBG=1`, so the same argument name could also be used with `tool/depot/download`.

nfeske, 2023-11-10:
I like the `DBG=1` argument.

Initially I thought that it would be nice to specify the debug archives analogously to bin archives (e.g. `./depot/create nfeske/dbg/x86_64/report_rom`), but this would not work for building pkgs with debug info. Your suggestion does not have this problem.

cproc, 2023-11-23:
Commit c8833b4 implements the `DBG=1` argument for creating, publishing and downloading of `dbg` archives in addition to `bin` archives.

To avoid multiple calls of `build_bin_archive` for each `bin/dbg` archive pair, pattern rules as suggested in [1] are used in `depot/var/build.mk`. They can be replaced by "grouped targets" if Make 4.3 or above becomes a minimum requirement in the future.

[1] https://stackoverflow.com/questions/2973445/gnu-makefile-rule-generating-a-few-targets-from-a-single-source-file

nfeske, 2023-11-24:
This looks excellent to me!

commit 4c3677fd (2023-11-24): depot: add support for debug info files

Fixes #5015