genodelabs/genode #5020: libc: handle monitor execution by a POSIX signal handler

cproc, 2023-10-09:
When GDB is running on Genode (for example with the `gdb_interactive` run script and the Ctrl-C fix from issue #5019 on hardware), pressing Ctrl-C triggers an unresolved exception in GDB:

```
[init -> gdb] Warning: attempt to nested execution of signal handlers
[init -> gdb] Warning: attempt to nested execution of signal handlers
[init -> gdb] Error: Uncaught exception of type 'Genode::Reconstructible<Libc::Main_job>::Deref_unconstructed_object'
[init -> gdb] Warning: abort called - thread: ep
```

In this case, the SIGINT handler is executed during a suspended `monitor()` call from `select()` and it calls `write()`, which calls `monitor()` again and reconstructs the already in use `Main_job` object and then destructs it.

cproc, 2023-10-09:
Fixed by 937d4e6.

chelmuth, 2023-10-09:
The issue reads like what I apprehended [here](https://github.com/Time0o/genode/commit/416b2c74eeebd979059389d6730736cd8629c0a6#r126994144).

chelmuth, 2023-10-09:
> Fixed by [937d4e6](https://github.com/genodelabs/genode/commit/937d4e6336061853c1ddabdb07e5c6a14317d829).

Do I understand correctly that this implements a nested job? Isn't the context running on kernel stack already and should log "called from non-user context"?

cproc, 2023-10-09:
> The issue reads like what I apprehended [here](https://github.com/Time0o/genode/commit/416b2c74eeebd979059389d6730736cd8629c0a6#r126994144).

When I replace `_resume_main()` by `trigger_monitor_examination()` (without commit 937d4e6), the same exception occurs.

cproc, 2023-10-09:
> Do I understand correctly that this implements a nested job? Isn't the context running on kernel stack already and should log "called from non-user context"?

The POSIX signal handlers appear to be executed from user context if I understand this correctly:

https://github.com/genodelabs/genode/blob/a8feb41758b983621e6c41eb1cecf23d6e5f224a/repos/libports/src/lib/libc/internal/kernel.h#L372-L378

cproc, 2023-10-09:
There might be a risk that `_main_timeout` gets overwritten by the nested monitor job if both use a timeout:

https://github.com/genodelabs/genode/blob/a8feb41758b983621e6c41eb1cecf23d6e5f224a/repos/libports/src/lib/libc/internal/kernel.h#L369-L370

cproc, 2023-10-09:
Maybe the outer monitor job should be aborted and cause an EINTR error when a POSIX signal occurred? Then it would not matter if its timeout got overwritten.

cproc, 2024-03-22:
> I'm mostly bothered by the call to `_setjmp()` in the nested execution of POSIX signal handlers. This overwrites any previous state stored in `_user_context`. Arbitrarily flipping back and forth between `_user_context` and `_kernel_context` is not intended.

I don't have a better idea at the moment, unfortunately. But in the GDB case it is also possible to execute the `continue` command in the background (`continue &` or `c &`) and to interrupt the execution with the `interrupt` command instead of pressing Ctrl-C to avoid the error.

chelmuth, 2024-07-24:
#5305 addresses the jmpbuf issue.