genodelabs/genode #5021: usb session renewal: fully asynchronous & interface level access

skalk, 2023-10-11:
Within the past years we've observed certain limitations of the adhoc developed original USB session RPC API. First and foremost there were still several synchronous RPC calls in place to e.g. ask for USB descriptors. The problem here is the server side - an USB multiplexer, which interacts with multiple clients and serves different requests at once should not block during individual RPC calls. The other limitation regards USB devices with multiple interfaces of different classes, e.g. an USB headset with audio interfaces and HID interface to realize volume, play/stop buttons. Here, we would like to give access to the different interfaces to different USB clients resp. drivers.

The new design is inspired by the platform session API. An USB session now only provides a ROM dataspace, which reports all devices this USB session has access to. The USB device information includes its interfaces and endpoint descriptions. Therefore, an USB client does not necessarily need to access different USB descriptors during initialiyation, but can extract most information by the XML representation of its devices ROM. Morover, the USB session itself only provides two RPC calls to `acquire` and `release` an USB device. The `acquire` call returns a capability to a new USB device session.

The device session includes RPC calls to register a signal context and request a dataspace. Both together form a packet stream, wereby packets here are always interpreted as USB control transfers. If a client has full control over the device, it can send any form of control transfer here. If the client has only limited access to a certain interface, it can only send control transfers to get information over the device, and to set an alternate interface regarding the interface it has access to. Moreover, the device session includes two RPC call to `acquire` or `release` an interface of the device. The `acquire` call returns a capability to a new USB interface session.

The interface session again includes RPC calls to register signal context and retrieve a dataspace for its packet stream access. The packets of this stream represent either interrupt, bulk, or isochronous transfers to an endpoint of this interface, or a flush command to cancel all ongoing transfers regarding this interface.

Beside the pure USB session renewal, a client API for device and interface access is envisioned that loosely corresponds to the existent block job API. Thereby, we hide the ancient packet stream API from developers, and enforce that signaling is done on a less eagerly base to increase throughput.

skalk, 2023-10-11:
Commit 7b05ce4 shows the current state of the new USB session & client API. It moreover adapts the Genode C-API for USB server side, and the implementation of the DDE Linux USB host controller driver.
Commit 8278075 adapts the USB block driver to use te new USB client API. The whole initialization is now done asynchronously in form of a state-machine. I've marked both commits as work-in-progress, because not all files that got deprecated were removed, and several other USB clients are not adapted yet. Also some places might need some explanations. Anyway, I've pused the status quo, so that you can get a first impression and leave comments before all other clients get adapted.

chelmuth, 2023-10-24:
Today I took the time to dive into 7b05ce4 and very much enjoyed the ride. I'm impressed by the clearness of the (hierarchical) session interfaces for USB bus, devices, and interfaces as well as the C API. It's has become quite tangible how the implementation will evolve and solves the issues stated above.

Nevertheless, I have a small collection of questions and remarks.

**GENERAL** Can *usb_session/urb_handler.h* be removed? The file does not contain any code.
**GENERAL** Do you plan to deprecate *usb/usb.h* and *usb/packet_handler.h* entirely or may we lose essential functionality?
**GENERAL** I expect *genode_c_api/usb_client.cc* will be added next?
**usb/types.h** I suggest to move the file to *usb_session/* or do you have other plans?
**usb/types.h** I suggest to replace `genode_usb_isoc_transfer_header::data_offset` by a calculation.
**usb_session/usb_session.h** The recipients `PORT` and `RPIPE` should be removed as Wireless USB is no more. Promise! It was removed from the Linux kernel before 6.0.
**usb_session/device.h** The use of *read* and *write* in `Update_urbs_policy` feels strange. How about renaming the methods to `produce_out_content` and `consume_in_content` analogous to the USB specification for IN and OUT endpoints?
**usb_session/device.h** `Usb::Interface` has virtual destructor but is not used as a base class anywhere. If this is not planned, I suggest to remove `virtual`.
**usb_session/device.h** `_try_process_ack` and `_try_submit_pending_urb` mention *split read/write operations*. What are these operations and why do they complicate the packet-stream handling?
**genode_c_api/usb.h,usb.cc** What's the rationale behind the removal of `struct genode_usb_buffer payload`? I changed this with `Genode::Byte_range_ptr` in the back of my head.
**genode_c_api/usb.h,usb.cc** I suggest to unify `void * data`, `void *callback_data, `void *opaque_data`, and `void *opaque_callback_data`.
**genode_c_api/usb.cc,lx_emul/usb.c** You removed *class*-based access policies incl. interface-to-device class quirk. How is the plan to address this use case in the future?

skalk, 2023-10-25:
Thanks @chelmuth for your valueable review!

> 
> Nevertheless, I have a small collection of questions and remarks.
> 
> **GENERAL** Can _usb_session/urb_handler.h_ be removed? The file does not contain any code. 

Sure, it is a development artefact.

> **GENERAL** Do you plan to deprecate _usb/usb.h_ and _usb/packet_handler.h_ entirely or may we lose essential functionality?

Yes

> **GENERAL** I expect _genode_c_api/usb_client.cc_ will be added next?

Yes

> **usb/types.h** I suggest to move the file to _usb_session/_ or do you have other plans?

I agree.

> **usb/types.h** I suggest to replace `genode_usb_isoc_transfer_header::data_offset` by a calculation.

Thanks for the pointer, I'll have a look into this.

> **usb_session/usb_session.h** The recipients `PORT` and `RPIPE` should be removed as Wireless USB is no more. Promise! It was removed from the Linux kernel before 6.0.

Okay.

> **usb_session/device.h** The use of _read_ and _write_ in `Update_urbs_policy` feels strange. How about renaming the methods to `produce_out_content` and `consume_in_content` analogous to the USB specification for IN and OUT endpoints?

Yes, this was probably too much influenced by the Block Job API.

> **usb_session/device.h** `Usb::Interface` has virtual destructor but is not used as a base class anywhere. If this is not planned, I suggest to remove `virtual`.

I have to proof this.

> **usb_session/device.h** `_try_process_ack` and `_try_submit_pending_urb` mention _split read/write operations_. What are these operations and why do they complicate the packet-stream handling?

Good point! Yet another artefact from the Job API. In the first place, I tried to keep it as similar as possible to potentially extract a common pattern. But during development it was no realistic option anymore.

> **genode_c_api/usb.h,usb.cc** What's the rationale behind the removal of `struct genode_usb_buffer payload`? I changed this with `Genode::Byte_range_ptr` in the back of my head.

There is no good rationale behind it. I was trying to simplify the callbacks, and during different iterations, I assume` it got removed without need. I can re-insert it.

> **genode_c_api/usb.h,usb.cc** I suggest to unify `void * data`, `void *callback_data`, `void *opaque_data`, and `void *opaque_callback_data`.

Okay.

> **genode_c_api/usb.cc,lx_emul/usb.c** You removed _class_-based access policies incl. interface-to-device class quirk. How is the plan to address this use case in the future?

We had an offline discussion about this in preparation of this issue's re-design ambition, and my understanding of the outcome was that we only support per-device (named), and at maximum per-vendor-product (for development run-scripts) assigments, because the current approach is too inflexible.
If in contrast to the discussion result, as I've understood it, a class-based approach is still desired, we have to distinguish in between different interfaces now anyway. Overwriting the class of a device as the current quirk, would not be an option anymore.

chelmuth, 2023-10-25:
> > **genode_c_api/usb.cc,lx_emul/usb.c** You removed _class_-based access policies incl. interface-to-device class quirk. How is the plan to address this use case in the future?
> 
> We had an offline discussion about this in preparation of this issue's re-design ambition, and my understanding of the outcome was that we only support per-device (named), and at maximum per-vendor-product (for development run-scripts) assignments, because the current approach is too inflexible. If in contrast to the discussion result, as I've understood it, a class-based approach is still desired, we have to distinguish in between different interfaces now anyway. Overwriting the class of a device as the current quirk, would not be an option anymore.

Don't get me wrong, I could remember our discussion but also that we had no concrete solution for Sculpt (or a USB HID test that supports not yet known mice). Thus, I added this open question to the list as we have to find a solution before a final merge.

skalk, 2023-10-25:
@chelmuth I've realized all your comments within two additional fixup commits, and rebased it to staging. It is still work-in-progress, but so you can have a separate look at the changes, if you like to.
The only open discussion point is the last one. For now, I would proceed with the work on the Genode C API USB client part.

skalk, 2024-02-26:
@chelmuth I've pushed the latest state of this work on my branch issue#5021. It contains seven commits starting from commit id 8ecda87 that are now ready for staging in my eyes.

chelmuth, 2024-02-27:
@skalk Is it correct that usb_hid_drv no longer supports the config attribute *use_report*? If yes, we should adapt *repos/dde_linux/src/drivers/usb_hid/README* as well as the run scripts/drivers.config files.

skalk, 2024-02-27:
Well, not by intention, maybe by accident, I put it on the list. Could you push fixup commit cd30df5a6958a1794c220db8a9d4ed7437e86905 to staging? It solves the wrong copying of isochronous transfers in libusb.

skalk, 2024-02-27:
And fixup commit 51646a8dcee85cba52821793878a0d6e3055d60d as well, as it re-enables package build for the black_hole depot autopilot test.

skalk, 2024-02-27:
@chelmuth I've rebased and add another fixup commit for the usb_hid_drv `use_report` attribute that vanished. Please have a look at my issue#5021 branch.

chelmuth, 2024-02-27:
Merged the commits.

chelmuth, 2024-02-28:
You have to add a policy to /config/usb for vbox - usb_devices_rom will be removed. From my experience there are more Sculpt integration tasks to tackle after the release. Yesterday I made some steps with driver_manager to detect USB storage again but did not finish until a working state.

skalk, 2024-03-13:
@chelmuth commit 6c44bb5 shall accomplish this line of work.

chelmuth, 2024-03-20:
Commit 451a7268fbc601c177306100a515d3de08125090 implements reporting of device manufacturer and product strings as well as interface description string. Output like follows.

```xml
<device name="usb-1-2" class="0x0" manufacturer="Logitech" product="USB Laser Mouse" vendor_id="0x46d" product_id="0xc062" speed="low" acquired="true">
  ...
</device>
<device name="usb-1-4" class="0xef" manufacturer="Chicony" product="FJ Camera" vendor_id="0x4f2" product_id="0xb704" speed="high">
  <config active="true" value="0x1">
    <interface active="true" number="0x0" alt_setting="0x0" class="0xe" subclass="0x1" protocol="0x0" name="FJ Camera">
      <endpoint address="0x87" attributes="0x3" max_packet_size="0x10"/>
    </interface>
    ...
  </config>
</device>
```

skalk, 2024-03-20:
@chelmuth a feature, which is very welcome!

closed 2024-04-12 by chelmuth