genodelabs/genode #5024: lx_kit / wifi: decoupled execution causes wifi regression

atopia, 2023-10-16:
Since commit caac994da848b my home wifi is unusable on the Lenovo Thinkpad X1 Nano G2. On the Lenovo Thinkpad X260 the issue is also present but less severe. The issue appears to be some sort of liveliness problem that is triggered / aggravated by my specific wifi setup:

- From this commit, fetchurl invocations reliably fall into the startup timeout and are respawned by the depot download manager
- If I start pinging the Genode system from another machine on the network, fetchurl reliably succeeds (even when it is respawned after failing before)
- Encryption type (WPA2 vs WPA3), the wpa_supplicant update (even when reverted), frequency (2.4 GHz vs. 5 GHz) or the signal strength (even in sight of the AP) don't have have any significant effect on the issue
- On the other hand, the same setup will reliably work with my phone Hotspot AP
- Verbose wifi logging on a failing setup did not provide any immediate clues (no wifi disconnects or similar; in fact no logging output on failure at all although the system remains reachable from the outside)

atopia, 2023-10-16:
Having set up a more complete environment I can confirm that wifi works as long there is load on the network.

| condition | `ping -c 100 -i 2 <AP IP>` |
| --- | --- |
| with an incoming ping from another machine in the wifi (also -i 0.2) | rtt min/avg/max/mdev = 3.295/113.508/400.807/84.027 ms |
| while concurrently downloading an ISO image | rtt min/avg/max/mdev = 2.288/43.491/458.502/55.525 ms |
| standalone | rtt min/avg/max/mdev = 4.783/3433.563/10163.764/2887.774 ms |

Note that no packets are lost under "standalone" conditions.

chelmuth, 2023-10-17:
Do you have already more info on this issue? Is is related to missing interrupts or timeouts?

chelmuth, 2023-10-19:
@atopia commit d0a18461532cd7a5c1dadcc798bbea7da38d63f7 addresses this issue.

atopia, 2023-10-19:
@chelmuth I can confirm that your fix solves the issue for me too. Thanks a lot!

commit d0a18461 (2023-10-19): genode_c_api: remove global-static constructors

The execution of global-static constructors in Genode is optional for
native components or determined in the libc for libc components. By
convention, we avoid static constructors wherever possible but in rare
cases the component can tweak the point when constructors are executed
and initialize an environment beforehand. The wifi_drv applies this
pattern for the uplink initialization, but is a libc component on the
other hand, and, thus, statics in the Genode C API for uplink are
constructed later by the libc. So, the prior initialization is reve [... cut]
closed 2023-10-25 by chelmuth