genodelabs/genode #5026: system_shell: preload posix.lib.so

cproc, 2023-10-17:
Make sure that the essential POSIX libraries are always loaded first and in the same order to avoid fork/execve problems if the 'DT_NEEDED' order in binaries differs.

cproc, 2023-10-17:
Fixed by 01c4e1d.

nfeske, 2023-10-17:
I think that adding this preload quirk is not a good way forward, even as an interim solution. I originally added the preload feature merely as a tool during the transition from noux to the libc-based fork/execve and actually planned to ultimately remove it. I'm afraid that the addition of the preload config may (1) confuse users because it is impossible to explain without going deep into the fork/execve implementation, and (2) be taken as an invitation to (mis)use the preload feature for other purposes.

Let's better address the problem at a point that is not visible at the configuration/integration level. The fork/execve mechanism relies on the invariant that the four libraries marked as "keep" in https://github.com/genodelabs/genode/blob/master/repos/libports/src/lib/libc/execve.cc#L328 must be loaded in each posix program in the exact same order. Prior the tool chain update, this invariant was satisfied statically be linking the .lib.so files in the same order everywhere. Now, after the tool chain update, this linking order seems to be more lax, thereby violation the assumption of the fork/execve mechanism.

@cproc could you try enforcing the loading order in the dynamic linker instead? Like this: if a libc.lib.so dependency is present, load it (this should implicitly load vfs.lib.so). If a libm.lib.so dependency is present, load it. If a posix.lib.so dependency is present load it, load all other dependencies. This way, we codify the invariant in a single place (the dynamic linker) instead of spilling the preload fix everywhere where fork/execve is used.

nfeske, 2023-10-17:
It should actually suffice to only cover the posix.lib.so because this library depends on the libc and libm anyway. So it comes down to adding the rule to the dynamic linker: if posix.lib.so is present as a dependency, load it (which deterministically loads libc, libm, and vfs). Then load all other .so files.

chelmuth, 2023-10-17:
I like this approach that requires the following addition to *lib/posix*.

```diff
+++ b/repos/libports/lib/mk/posix.mk
@@ -1,6 +1,6 @@
 SRC_CC = construct.cc
 
-LIBS += libc
+LIBS += libc libm
 
 SHARED_LIB = yes
```

chelmuth, 2023-10-17:
With #5027 and #5028 merged, this issue can be closed.

closed 2023-10-17 by chelmuth