genodelabs/genode #5030: depot: add dummy recipe for compat-libc

jschlatow, 2023-10-18:
Since Goa's rust support depends on _api/compat-libc_, we need to make sure that we publish the corresponding api archive for every Sculpt release. Currently, Goa's dependencies are aggregated by _pkg/goa_. However, we can only make sure that _api/compat-libc_ is published along with _pkg/goa_ by adding a corresponding src archive to the latter. As there is no _src/compat-libc_ (static library only), we need to add a dummy recipe that we can reference in _pkg/goa_.

jschlatow, 2023-10-18:
Fixed by dfb5589

chelmuth, 2023-10-18:
Would you mind if I rename src/compat-libc-dummy to src/compat-libc? IMO it does not matter that there are no additional sources in the package.

jschlatow, 2023-10-18:
> Would you mind if I rename src/compat-libc-dummy to src/compat-libc? IMO it does not matter that there are no additional sources in the package.

My intention was to indicate that there is no library, yet, on second thought, sticking to src/compat-libc spares us any renaming in case we actually build a dynamic library.

closed 2023-10-25 by chelmuth