genodelabs/genode #5043: libc: implement sysconf(_SC_GETPW_R_SIZE_MAX)

atopia, 2023-10-26:
Currently, running ripgrep on Genode emits
 "Warning: sysconf(71) not implemented".

 Implement sysconf(_SC_GETPW_R_SIZE_MAX) and silence the warning.

chelmuth, 2023-10-27:
What happens if we just *returns -1 without changing errno* accoding to https://pubs.opengroup.org/onlinepubs/9699919799/? I'm afraid returning `sizeof(struct passwd)` is misleading.

atopia, 2023-11-01:
I wasn't aware that -1 was a valid non-error return value, but I agree that's a more generic solution since the goal is mainly silencing the warning anyway.

chelmuth, 2023-11-01:
Merged the commit to staging and chuckled to myself while reading `man getpwnam`.

> The getpwnam() and getpwuid() functions return a pointer to a passwd structure, or NULL if the matching entry is not found or an error  occurs. If an error occurs, errno is set appropriately. [GREAT!] If one wants to check errno after the call, it should be set to zero before the call. [UH?]

commit 1690a609 (2023-11-01): libc: silence sysconf(_SC_GETPW_R_SIZE_MAX) warning

Currently, running ripgrep on Genode emits
"Warning: sysconf(71) not implemented".

Return -1 without setting errno, which is a valid return value according
to getpwnam(3) and silence the warning.

Fixes #5043
commit e3a51478 (2023-11-02): libc: silence sysconf(_SC_GETPW_R_SIZE_MAX) warning

Currently, running ripgrep on Genode emits
"Warning: sysconf(71) not implemented".

Return -1 without setting errno, which is a valid return value according
to getpwnam(3) and silence the warning.

Fixes #5043