genodelabs/genode #5046: monitor: memory map and vanished mappings

cproc, 2023-11-02:
The debug monitor currently maintains a memory map for each inferior, which can be requested by GDB and which is currently updated on `attach()` and `detach()` calls. Unfortunately it can happen that memory mappings vanish without the knowledge of the monitor, for example when an attached dataspace gets destroyed. This can lead to inconsistencies in the memory map when parts of the freed regions are reused later, and keeping outdated information in the map is also not good for the memory use of the monitor and for the GDB user.

One way to solve the problem could be to check the validity of the memory map entries during `attach()` and `detach()` and when the map is requested by GDB, for example by trying to obtain the dataspace size of each entry. This would detect the case that a dataspace has been destroyed in the meantime, but might slow down the monitor when many regions exist. An alternative solution could be to obtain the memory map from the parent/core, but that would probably require an API change. Opinions?

cproc, 2023-11-02:
GitHub e-mail notification check, please ignore.

nfeske, 2023-11-02:
> [...] for example by trying to obtain the dataspace size of each entry. This would detect the case that a dataspace has been destroyed in the meantime

The monitor must not invoke dataspace capabilities that are passes as `attach` argument because the capability cannot be trusted by the monitor. By calling `dataspace.size()`, the monitor would risk to get stuck.

In my understanding, the problem of a mapping that disappears without `detach` can happen, e.g., when closing a nitpicker session, which destroys the virtual framebuffer shared between the client and the nitpicker. When nitpicker destroys the dataspace, it gets removed from all address spaces where this dataspace was attached. The monitor has no chance to get hold of such a side effect because the active participants (nitpicker and core) are outside the monitor's view.

One intuitive solution would be adding some kind of introspection interface to core's PD service. E.g., adding an RPC call for probing a virtual-address range. However, I'm very hesitant to extend core in this way because such an interface would inherently be prone to race conditions. Think of the situation where a mapping vanishes right after a probe call. The information can be outdated right after having acquiring it. It will work on 99.9% of cases. But the 0.1% would be a hell to debug.

I think a better way to handle this situation would be to keep virtual address ranges marked as being in use as long as not explicitly released by a `detach` call - preventing the reuse of such orphaned ranges for new attachments. This would ultimately solve possible ambiguities. Note that this change in core's region-map behavior may introduce possible virtual-memory leaks in cases where detach is currently not called by the client. But most shared-memory clients use a form of `Attached_dataspace` which already calls `detach`. Any remaining cases can be fixed.

chelmuth, 2023-11-03:
> I think a better way to handle this situation would be to keep virtual address ranges marked as being in use as long as not explicitly released by a `detach` call - preventing the reuse of such [...]

Do I overlook something or could this behavior be implemented in the monitored PD session for prototyping (resp. temporary quickfix) without touching core?

cproc, 2023-11-03:
> The monitor must not invoke dataspace capabilities that are passes as `attach` argument because the capability cannot be trusted by the monitor. By calling `dataspace.size()`, the monitor would risk to get stuck.

The monitor currently calls `dataspace.size()` to calculate the size of the region, because the parent `attach()` only returns the start address. Could `attach()` be enhanced to return the size of the attached region as well?

nfeske, 2023-11-03:
> Do I overlook something or could this behavior be implemented in the monitored PD session for prototyping (resp. temporary quickfix) without touching core?

That's a cool idea but it would require the monitor to implement a range allocator locally and only using `attach_at` at core. I think that the change of core would be simpler and the possible fallout (unexpected leaks of virtual memory) is unlikely or manageable.

> Could attach() be enhanced to return the size of the attached region as well?

That's a sensible idea. You are interested in the size of the mapping (which may just be a part of the dataspace), not the size of the dataspace after all.

nfeske, 2023-11-03:
Btw, to directly address the symptom of this issue (memory map containing overlapping regions), we could also consider adding a plausibility check to the monitor's attach implementation. Whenever the attached region conflicts with an existing entry of the memory map, we could simply remove the existing entry because it is obviously orphaned at that point.

This would spare us the trouble of changing core short term. In the longer term, I think we should aspire keeping orphaned regions marked a reserved until explicitly detached.

cproc, 2023-11-03:
Commit 43f7f53 adds the plausibility check to the monitor.

nfeske, 2023-11-03:
@cproc nice that you have picked up my suggestion!

While looking the patch, I wondered if you have considered keeping the regions in a `Registry` instead of using an old-school `List`?

cproc, 2023-11-06:
Commit 2aef455 replaces the region list with a registry (depends on 43f7f53).

cproc, 2023-11-06:
Commit 4bf1381 marks the regions which get detached by `Dataspace_component::detach_from_rm_sessions()` as reserved until `detach()` is called from somewhere else.

When I add a log message in `Dataspace_component::detach_from_rm_sessions()` for each affected region and run Sculpt, some of these messages are seen which could be related to component destruction and are probably to be expected, but also one message appears repeatedly when interacting with the menus.

cproc, 2023-11-06:
Commit 77bcf3b detaches the framebuffer dataspace in the GUI terminal before creating a new one. This fixes some of the "dataspace still attached" debug messages I saw when interacting with Sculpt menus.

cproc, 2023-11-07:
One place which appears to need a `detach()` call somewhere with the new "keep the region reserved" mechanism is in `acpi_drv` when used with Sculpt:

```
[init -> drivers -> acpi_drv] Error: Unknown exception occured - failure
[init -> drivers -> acpi_drv] Error: Uncaught exception of type 'Genode::Region_map::Region_conflict'
[init -> drivers -> acpi_drv] Warning: abort called - thread: ep
```

The error occurs here:

https://github.com/genodelabs/genode/blob/4550c28ff059cb97d8499d0b62c98989b6d88bd6/repos/os/src/drivers/acpi/memory.h#L186-L188

cproc, 2023-11-07:
@nfeske: I adapted the registry commit based on your latest suggestions in c28333d.

nfeske, 2023-11-07:
> I adapted the registry commit based on your latest suggestions

Thank you very much for your patience. :-)

chelmuth, 2023-11-08:
I merged b42d76dc4c1be2008e4c8aacbe976fc0b39dd67f and acac51990936c19b85a164382fcded978bfd7e7f but left out the core commit 290bd7dcaa0dac333553aee600ab6d71d7bdee1f for now as I'm not sure about the state of discussion in this regard.

cproc, 2023-11-09:
Commit 5deb183 fixes the region conflict in `acpi_drv`.

nfeske, 2023-11-30:
The immediate fix (in the form if the sanity check) for the problem at hand entered the master branch.

@cproc shall we close this issue? If pursuing the reserved-region approach further, we should probably better take this topic to another issue.