genodelabs/genode #5047: gdb: internal error triggered by pagination

cproc, 2023-11-02:
Depending on the terminal size, an internal GDB error can occur during the `monitor_gdb.run` test:

```
Thread 1.1 "test-monitor_gdb" stopped.
_start ()
--Type <RET> for more, q to quit, c to continue without paging--/.../contrib/gdb-e79d176438a5519950459e7b9b0752ed5993de83/src/noux-pkg/gdb/gdb/remote.c:8361: internal-error: wait: Assertion `!async_event_handler_marked (rs->remote_async_inferior_event_token)' failed.
A problem internal to GDB has been detected,
further debugging may prove unreliable.
```

cproc, 2023-11-02:
Commit 5bf0141 disables pagination before connecting to the debug monitor in `monitor_gdb.run`. I'm planning to investigate the GDB error as well.

chelmuth, 2023-11-02:
Good catch. I wonder why GDB never changed this option's default to *off*. The setting is part of my *.gdbinit* file for 20 years and I inherited it from someone working with the tool much longer.

cproc, 2023-11-06:
Commit 7e208c2 fixes the failed assertion in GDB.

chelmuth, 2023-11-06:
> Commit [7e208c2](https://github.com/genodelabs/genode/commit/7e208c2a0cb30e1d836fcec94712d288fab510e5) fixes the failed assertion in GDB.

As the patch changes GDB: Do I have to rebuilt `tool_chain ENABLE_FEATURES=gdb`?

cproc, 2023-11-06:
> As the patch changes GDB: Do I have to rebuilt `tool_chain ENABLE_FEATURES=gdb`?

Yes

commit e2cc21f9 (2023-11-06): gdb: fix failed assertion triggered by pagination

Fixes #5047
closed 2023-11-30 by chelmuth