genodelabs/genode #5055: ports: tool_chain_auto.run often times out

cproc, 2023-11-15:
Currently, the `tool_chain_auto.run` test often times out on NOVA and hw.

On NOVA, I noticed that the retry loop in `Parent_service::initiate_request()` does't seem to actually retry the request on `Out_of_ram` and `Out_of_caps` exceptions, because the session phase got changed to `CLOSED` in `Try_parent_service::initiate_request()` when the exception occurred there.

cproc, 2023-11-15:
Commit e859303 restores the original session phase in `Parent_service::initiate_request()`, which seems to fix the `tool_chain_auto.run` test on NOVA.

nfeske, 2023-11-15:
You made my day! Thank you very much for pinning down this issue that was nagging me in the back of my head for quite some time now. I merged your fix to staging right away.

cproc, 2023-11-16:
I also noticed that the initial ROM session request for `/bin/bash` is repeated 10 times:

```
[init] Local_connection_base(): 0: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 1: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 2: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 3: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 4: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 5: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 6: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 7: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 8: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
[init] Local_connection_base(): 9: service=ROM cid=4 args='label="/bin/bash", ram_quota=6144, cap_quota=3, diag=0' state=CREATE_REQUESTED ram_quota=6K, cap_quota=3
[init] Async_service::initiate_request(): ROM, ram_quota=6144, cap_quota=3, diag=0, label="/bin/bash"
```

cproc, 2023-11-16:
On hw (x86_64) the problem ist that `Allocator::Alloc_error::OUT_OF_CAPS` is translated to an `Out_of_ram` exception. Throwing `Out_of_caps` in this case solves the problem, but for completeness, how should `Allocator::Alloc_error::DENIED` be handled?

https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base-hw/src/core/platform_pd.cc#L140-L151

nfeske, 2023-11-16:
> how should Allocator::Alloc_error::DENIED be handled?

Under which condition can this error occur? From the tip of my head I cannot think of any.

cproc, 2023-11-16:
> > how should Allocator::Alloc_error::DENIED be handled?
> 
> Under which condition can this error occur? From the tip of my head I cannot think of any.

In this specific case where `Cap_space::upgrade_slab()` is used with a `Sliced_heap`, `DENIED` is returned on `Region_map::Invalid_dataspace` and `Region_map::Region_conflict` exceptions:

https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base/src/lib/base/sliced_heap.cc#L75-L81

cproc, 2023-11-16:
`Allocator::throw_alloc_error()` throws a `Denied` exception. Maybe this function could be used in `Cap_space::upgrade_slab()` in the error case?

https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base/include/base/allocator.h#L98-L109

nfeske, 2023-11-16:
I agree. Both error cases can never occur in practice unless there is a bug in core. One would mean that the virtual address space of core is exhausted, which is impossible because it is so sparsely populated. The other would mean that there is a core-internal inconsistency (if the dataspace allocated by core's sliced heap is invalid - how can this possibly happen at this point - after the successful allocation?).

In both cases, it would be OK to let core die (by not catching the `Denied` exception).

nfeske, 2023-11-30:
Fixed in master.