genodelabs/genode #5057: hw: distinguish `Alloc_error` conditions

cproc, 2023-11-16:
There are some places in the hw code where `Alloc_error` conditions are not distinguished and a mismatched exception could be thrown:

https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base-hw/src/core/platform_pd.cc#L48
https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base-hw/src/core/platform_pd.cc#L148
https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base-hw/src/core/platform_thread.cc#L82
https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base-hw/src/core/rpc_cap_factory.h#L106
https://github.com/genodelabs/genode/blob/dc39a8db622fbf6c24d6c26f6ef5e510cbf58a3d/repos/base-hw/src/core/spec/arm/virtualization/vm_session_component.cc#L83
https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base-hw/src/core/spec/x86_64/virtualization/vm_session_component.cc#L81
https://github.com/genodelabs/genode/blob/7cb466f6eb9b2f02ea78b938542f503e82b23aa6/repos/base-hw/src/core/spec/x86_64/virtualization/vm_session_component.cc#L111

cproc, 2023-11-16:
Commit 135341a covers `Cap_space::upgrade_slab()` as discussed in #5055.

nfeske, 2023-11-16:
Thanks @cproc, I merged https://github.com/genodelabs/genode/commit/135341a724488fa0dc2245ed22b86c52801c906b to staging.