genodelabs/genode #5058: dialog: add text-area widget

nfeske, 2023-11-16:
By converting the editing part of app/text_area into a dialog widget, the text editor becomes re-usable.

m-stein, 2023-11-27:
@nfeske I found that one of a3f8827f6d3280348781600795522c4fae3c6462, 8f1e753955b0fb213c3132262b32759dd1bb4a93 breaks my Sculpt (with 8f1e753955, I run into ``dialog.h:27:10: fatal error: dynamic_array.h: No such file or directory``). The problem is that when I try to edit a file in the Leitzentrale file tab, I get:
```
[init -> runtime] Error: editor: environment ROM session denied (label="editor", ram_quota=6144, cap_quota=3, diag=0)
```
When looking into ``config/managed/runtime`` I see ``<start name=editor>`` but correctly equipped with ``<binary name="text_area">``. When I adapt the Sculpt Manager to produce ``<start name="text_area">`` instead, I get:
```
[init -> runtime] Error: text_area: environment ROM session denied (label="text_area", ram_quota=6144, cap_quota=3, diag=0)
```
I will further investigate.

m-stein, 2023-11-27:
@nfeske This 6557513947 fixes the problem for me. I refrained from introducing ``recipe/src/dialog`` and ``recipe/api/dialog`` given the fact that it would currently be used by only one archive. I'm still a bit confused by the fact that the program requested a ROM named "editor" instead of "text_area" without the fixup.

m-stein, 2023-11-27:
@nfeske This 6557513947 fixes the problem for me. I refrained from introducing recipe/src/dialog and recipe/api/dialog given the fact that it would currently be used by only one archive. I'm still a bit confused by the fact that the program requested a ROM named "editor" instead of "text_area" without the fixup.

nfeske, 2023-11-28:
Thanks @m-stein for solving this issue, which slipped through my fingers because I always had `build: app/text_area` present in my .sculpt file during testing and development. So I did not notice the packaging issue. Your fixup is perfectly fine. I merged it to staging just now.

Regarding your question about the binary name: the binary name of the component differs from the start name (using the `<binary>` node. When init prints the error message, it has not yet applied the label rewriting.

closed 2023-11-30 by chelmuth