genodelabs/genode #5061: Do not implicitly build any shared libraries

nfeske, 2023-11-20:
The build system automatically executes the build all dependencies of each target, which is convenient when using many small inter-dependent libraries. But with complex libraries (like Qt), one always needs to locally build those complex libraries even though the library internals are rarely touched. As another example, the Sculpt modding procedure described in https://genodians.org/nfeske/2023-11-10-modding-sculpt requires the local build of the C runtime when merely tweaking the menu_view component that happens to depend on the C runtime. But the local build of the C runtime is actually never used (the downloaded binary archive suffices as long as the C runtime's code remains untouched).

In the context of issue #4860 I have now revisited all run scripts regarding the build of shared libraries to be integrated into the boot image. The `[build_artifacts]` fosters now the consistency between build arguments and build_boot_image arguments.

With issue #4860 settled, we can finally cut the implicit build dependencies between a target and the shared libraries the target depends on. For building a target, merely the ABIs of the used shared libraries are needed as build dependencies.

For building a shared library, one needs to explicitly specify `lib/<libname>` as build argument now.

nfeske, 2023-11-24:
Does it work when replacing `append_if` by `lappend_if`?

m-stein, 2023-11-24:
@nfeske Yep, that works as well. So, I rather adapt the script?

nfeske, 2023-11-24:
I'd very much appreciate a fixup to my original commit (I guess it's b54ca6f76665d97dd403075f0d26700d5b6a4278).

nfeske, 2023-11-24:
> I'd very much appreciate a fixup to my original commit

Nevermind, I already added a fixup to staging.

m-stein, 2023-11-26:
@nfeske Thanks! The script works again.

commit 85681bf8 (2023-11-20): mk: remove implicit build of shared libraries

This patch removes the implicit build of all shared libraries a target
depends on. Targets only depend on the respective ABIs instead. This
alleviates the need to locally build complex shared libraries (think of
Qt) when developing applications. Instead, application developers can
use binary depot archives.

The implementation splits the mk/lib.mk file into three files:
- mk/a.mk   for building one static library (.lib.a)
- mk/so.mk  for building one shared object  (.lib.so)
- mk/abi.mk for building one ABI stub       (.abi.so)

Fixes #5061