genodelabs/genode #5072: dde_rump: Github disables SVN support

trimpim, 2023-12-05:
A coworker tried to prepare the `dde_rump` port. This resulted in the following error:

```
svn: E170013: Unable to connect to a repository at URL 'https://github.com/justincormack/netbsd-src/trunk/src/sys/arch'
svn: E900002: Subversion support temporarily disabled.

GitHub has temporarily disabled Subversion support as part of a planned brownout
in order to give you advance notice of the upcoming permanent removal of
Subversion support on 2024-01-08.

Please see https://github.blog/2023-01-20-sunsetting-subversion-support/ for details.
```

Today seems to be one of the brownout events announced in the post.

a-dmg, 2023-12-05:
Dear Genodians,

As @trimpim mentioned, I encountered this issue after setting up a new development machine.
I have proposed one commit that introduces new install rules for port '.git-sparse'. That takes advantage of `git sparse-checkout` [as described here](https://github.blog/2020-01-17-bring-your-monorepo-down-to-size-with-sparse-checkout/) to only checkout the required source subdirectory for a port.  

I am still somewhat satisfied with this hotfix. Any other port using SVN to perform a partial clone might need to be changed. Also, I am sure that this solution is proeficient. However, it shows how this feature can achieve a similar result without SVN.

a-dmg, 2023-12-07:
@nfeske thank you for pointing out an existing fix! I have adjusted my commit to use the one in Goa. However, it requires minor adjustments that I provide in a fixup commit for you to review easily.

Here are details about the two problems I encountered, first the path in `contrib/<port>/...` did not exist:
```
➜  contrib git:(master) /home/alice/Repos/genode/tool/ports/prepare_port dde_rump
dde_rump  download https://github.com/ssumpf/rump.git
dde_rump  git Cloning into 'src/lib/dde_rump'...
dde_rump  update src/lib/dde_rump
dde_rump  download https://github.com/justincormack/netbsd-src
dde_rump  git Cloning into '/tmp/tmp.Hh6hZE7aAl'...
remote: Enumerating objects: 1698, done.
remote: Counting objects: 100% (1698/1698), done.
remote: Compressing objects: 100% (1203/1203), done.
remote: Total 1698 (delta 527), reused 1557 (delta 434), pack-reused 0
Receiving objects: 100% (1698/1698), 2.60 MiB | 1.57 MiB/s, done.
Resolving deltas: 100% (527/527), done.
Updating files: 100% (1767/1767), done.
dde_rump  download https://github.com/justincormack/netbsd-src
dde_rump  git Cloning into '/tmp/tmp.SvSUWT1rMV'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (39/39), done.
remote: Total 46 (delta 36), reused 13 (delta 5), pack-reused 0
Receiving objects: 100% (46/46), 34.48 KiB | 4.31 MiB/s, done.
Resolving deltas: 100% (36/36), done.
mv: cannot move '/tmp/tmp.SvSUWT1rMV/src/common/lib/libc/arch/aarch64/atomic' to 'src/lib/libc/aarch64/atomic': No such file or directory
...
```

After fixing it, I encountered another error because the paths in patches did not match what is in `contrib/<port>/...`:
```
➜  contrib git:(master) /home/alice/Repos/genode/tool/ports/prepare_port dde_rump
dde_rump  download https://github.com/ssumpf/rump.git
dde_rump  git Cloning into 'src/lib/dde_rump'...
dde_rump  update src/lib/dde_rump
dde_rump  download https://github.com/justincormack/netbsd-src
dde_rump  git Cloning into '/tmp/tmp.nrXry3fMnl'...
remote: Enumerating objects: 1698, done.
remote: Counting objects: 100% (1698/1698), done.
remote: Compressing objects: 100% (1203/1203), done.
remote: Total 1698 (delta 527), reused 1557 (delta 434), pack-reused 0
Receiving objects: 100% (1698/1698), 2.60 MiB | 12.85 MiB/s, done.
Resolving deltas: 100% (527/527), done.
dde_rump  download https://github.com/justincormack/netbsd-src
dde_rump  git Cloning into '/tmp/tmp.E8uUVqyvnl'...
remote: Enumerating objects: 46, done.
remote: Counting objects: 100% (46/46), done.
remote: Compressing objects: 100% (39/39), done.
remote: Total 46 (delta 36), reused 13 (delta 5), pack-reused 0
Receiving objects: 100% (46/46), 34.48 KiB | 4.93 MiB/s, done.
Resolving deltas: 100% (36/36), done.
dde_rump  download https://github.com/justincormack/netbsd-src
dde_rump  git Cloning into '/tmp/tmp.l1XQwpXqPQ'...
remote: Enumerating objects: 26, done.
remote: Counting objects: 100% (26/26), done.
remote: Compressing objects: 100% (17/17), done.
remote: Total 26 (delta 17), reused 13 (delta 7), pack-reused 0
Receiving objects: 100% (26/26), 31.75 KiB | 3.53 MiB/s, done.
Resolving deltas: 100% (17/17), done.
dde_rump  apply patches/rump.patch
dde_rump  apply patches/aarch64.patch
The text leading up to this was:
--------------------------
|diff --git a/arm/arm32/kobj_machdep.c b/arm/arm32/kobj_machdep.c
|index 6fc772f..146f580 100644
|--- src/lib/dde_rump_backport/arm/arm32/kobj_machdep.c
|+++ src/lib/dde_rump_backport/arm/arm32/kobj_machdep.c
--------------------------
File to patch: 
...
```

nfeske, 2023-12-13:
The fix entered the master branch today.

closed 2023-12-13 by nfeske