genodelabs/genode #5073: file vault: investigate incompatibility with < 23.05 containers

m-stein, 2023-12-06:
The current File Vault version usually crashes with a rump kernel panic when trying to unlock a container created with a File Vault older than 23.05. . Although Sculpt 23.05 introduced a migration approach for affected containers, It should be investigated what is the cause of the error. One would at least expect the File Vault to detect a seemingly compromised container and warn about bad hashes on a way lower level. More general, the issue may indicate a bug in the latest File Vault and should therefore not be left unharvested.

m-stein, 2023-12-06:
Among small improvements in the file_vault_client.run script, I found that the cause for the incompatibility was caused by a bug in the former VFS CBE library that was used by File Vault versions < 23.05.

A name clash that the compiler did not complain about caused a data member to be initialized with itself rather than the method argument that was meant to be used (519c0ce6e5). This caused the crypto  back-end to use an IV of 0 instead of the physical block address and therefore produced garbage client-FS data. The data passed the hash checks anyway because hashes are only checked against encrypted (on-disc) data. 

With 96f93d3719, I can demonstrate that file_vault_client.run succeeds on a container of an unfixed 23.04 File Vault. However, I am not sure whether such a feature is desired for master.

Any opinions?

ssumpf, 2023-12-06:
`A name clash that the compiler did not complain about caused a data member to be initialized with itself rather than the method argument that was meant to be used (https://github.com/genodelabs/genode/commit/519c0ce6e5f6183a09e7144359d1c4b105613996).`

This is indeed a very mean and hard to find bug. So, congratulations from my side!

m-stein, 2023-12-07:
@ssumpf Thanks! :) I'm happy that another waft of mist around that project dissolved.

tomga, 2023-12-07:
> A name clash that the compiler did not complain about caused a data member to be initialized with itself rather than the method argument that was meant to be used

@m-stein, I'd rather expect that local method argument was overwritten with its own value but effect for struct member is the same - not initialized.

But anyway it is not the main thing I wanted to raise. You have another argument `data_index` that is "initialized" in exactly the same way. Shouldn't the fix also cover it?

m-stein, 2023-12-08:
@tomga Thanks for your addition! You're right with both.

However, the data_index bug didn't affect my work at this issue. The CBE VFS-plugin that contains the bug was removed from the Genode repo for good and I revived it only temporarily for the purpose of clarifying the backwards incompatibility.

m-stein, 2023-12-08:
In an offline discussion, we decided to not merge https://github.com/genodelabs/genode/commit/96f93d37190f4548a9ea1a7c2da5e775d4ede8e9 . The found bug in the < 23.05 File Vault versions renders their block encryption insecure and users should not be motivated to continue using affected containers but rather migrate their data to a container created with a recent File Vault version. Therefore, I'll close the issue.

closed 2023-12-08 by m-stein