genodelabs/genode #5076: Debugging with GDB on Sculpt

cproc, 2023-12-08:
Currently, we already have the Sculpt runtime debug monitor and a `gdb_x86` depot package which provides a GDB client on Sculpt. What's still missing is an automated way to provide GDB with the needed debug info files, which involves two steps:

- downloading the `dbg` depot archives containing the debug info files of the monitored components
- creating a temporary directory per monitored component in a RAM file system with symbolic links to all involved binaries and debug info files

I've built a new `gdb_support` subsystem now which does both of these steps using the information from `/config/managed/deploy` (to obtain the `pkg` archive paths of the monitored components for downloading the dependent `dbg` archives with a dedicated `depot_download` subsystem instance) and from `/config/managed/runtime` (to obtain the `bin` depot paths of the involved binaries and shared libraries for the creation of the symbolic links into the corresponding `bin` and `dbg` depot directories).

But now there is the question how to download and find `ld.lib.so.debug`. This file is part of the `base-<kernel>` `dbg` archive, but the corresponding `bin` archive is not a dependency of the `pkg` archives of the (kernel-agnostic) monitored components, so there is no `pkg` archive path which can be given to the `depot_download` subsystem to automatically download the `base-<kernel>` `dbg` archive as a dependency and there is also no information about the version of the `base-<kernel>` `dbg` archive available which would be needed for the symbolic link into the depot (the kernel name could probably be derived from the `platform_info` ROM). Also, the `ld.lib.so.debug` file must match the used `ld.lib.so` file, which is part of the Sculpt base system and might slightly differ from a version built and provided by the depot owner of a monitored component.

One way to solve the problem could be to provide `ld.lib.so.debug` as a ROM module like `ld.lib.so`, but this would increase the size of the base system. Any other ideas?

nfeske, 2023-12-08:
Thanks @cproc for sharing your line of thoughts. I'm excited about how far you have already taken this topic.

> One way to solve the problem could be to provide ld.lib.so.debug as a ROM module like ld.lib.so, but this would increase the size of the base system. 

The increase would be quite substantial. In the case of `ld-nova.lib.so` it is 7.5 MIB (still 2.5 MiB gzip'ed). I would prefer to avoid that.

> Any other ideas?

I think it would be best to supplement the information about the concrete base binary archive used for the image into the image as part of the `boot_info` ROM, which is already present in Sculpt. So we could add another attribute like `base="cproc/bin/x86_64/base-nova/2023-10-25` to it. With this in place, the gdb_support subsystem can get hold of the depot content to download.

> Also, the ld.lib.so.debug file must match the used ld.lib.so file, which is part of the Sculpt base system and might slightly differ from a version built and provided by the depot owner of a monitored component.

That is an interesting point. In cases where a customized lib/ld can be integrated into a modular sculpt as depot.tar (to be mounted in the ram_fs via gems/sculpt/ram_fs/depot config snippet), we could add kernel-specific base binary archive into `depot.tar`. So the gdb_support subsystem would already see the matching base archive present when looking into the depot.

In the special case where sculpt is instructed to `build: lib/ld`, it would of course be nice to supplement the current build-directory's version of the debug info into the Sculpt image. Maybe, we could introduce a convention for this special case by defining the version as `current_build`. So the boot-info ROM would say: `base="cproc/bin/x86_64/base-nova/current_build` and the depot rom could feature the debug info at `depot/cproc/dbg/x86_64/base-nova/current_build/...`. That's just an idea though. I think we don't need to address the support for `build: lib/ld` immediately.

cproc, 2023-12-08:
> I think it would be best to supplement the information about the concrete base binary archive used for the image into the image as part of the `boot_info` ROM, which is already present in Sculpt. So [...]

So far I used the changes in commits a061b63 and a307d1c to download the `dbg` archives as optional dependencies of the `pkg` archives of the monitored components. The `gdb_support` subsystem just puts the `pkg` paths with a `debug="yes"` attribute into an `installation` report-ROM for the `depot_download` subsystem (a sub-init using the existing `gems/recipes/raw/depot_download/depot_download.config`). So, this mechanism currently only works with `pkg` archives. Should there be a `base-nova` `pkg` archive in addition to the `bin` archive or should the download mechanism somehow get changed to support the direct download of the `base-nova` `dbg` archive as well?

nfeske, 2023-12-08:
I think the depot_download subsystem should better support the direct downloading of dbg archives. It would be very natural to support the following:

```
<installation>
  <archive path="genodelabs/dbg/x86_64/base-nova/2023-10-25"/>
</installation>
```

cproc, 2024-01-29:
> I think the depot_download subsystem should better support the direct downloading of dbg archives. It would be very natural to support the following:
> 
> ```
> <installation>
>   <archive path="genodelabs/dbg/x86_64/base-nova/2023-10-25"/>
> </installation>
> ```

I added support for that in `depot_query` so far by just adding the given `bin` and `dbg` archive paths to the binary dependencies, which worked fine in the Sculpt debugging scenario with `source="no"`, but when I added a `bin` archive path for automated testing in `depot_download.run`, which downloads the corresponding `src` archive as well, the problem occurred that the download of the `src` archive failed because `Depot::Archive::name()` returned the architecture instead of the name when I used it for deriving the `src` archive path from the `bin` archive path. For `pkg` archives the architecture is currently obtained from the `arch` attribute of the `installation` XML node, but when I tried to use that for `bin` and `dbg` archives as well and omitted the architecure in the `installation` archive path, the path reported in the `missing` XML node by `depot_query` (with architecture) did not match the `depot_download` job path (without architecture) and the `depot_download` state report did not contain the `bin` archive path.

So, should I enhance the `Depot::Archive` class with special cases for `bin` and `dbg` paths (with architecture) or solve the problem in a different way?

nfeske, 2024-01-29:
> should I enhance the Depot::Archive class with special cases for bin and dbg paths (with architecture) [...]?

Intuitively, this makes much sense. `Archive::name()` should of course not return the architecture. ;-)

cproc, 2024-03-22:
Commit 6eacbc1 adds `terminal` to the Sculpt index and commit 44a1285 adds `gdb_x86` to the Sculpt distribution and index.

nfeske, 2024-03-22:
Thank you. I merged both commits to staging.

chelmuth, 2024-04-19:
The work was merged to master and I successfully debugged nano3d and qt5_textedit simultaneously on my Sculpt installation.

closed 2024-04-19 by chelmuth