genodelabs/genode #5077: tresor: test and fix corner-case tree dimensions

m-stein, 2023-12-11:
There are several issues when trying to drive the Tresor library on a container with minimum dimensions (1 level, degree of 1, 1 leaf) or maximum dimensions (6 levels, degree of 64, 64*6 leaves). Either we fix those issues or we restrict the configuration space accordingly. In any case we should test the corner cases on a regular basis.

m-stein, 2024-03-21:
Based on the commits for issue #5148 I've created a proposal for review that fixes this issue and some other problems I found on the way:
06cb20d734 tresor: fix and test handling of minimal trees
748ad3fa7b tresor: fix faults on failed free-tree requests
bdf68d79fa tresor_tester: check uninitialized vba data
5ef64e35cc tresor: discard snapshots only when writing the sb
eb629dc97c tresor: don't decode supeblock before hash check
6b4025d1bf tresor: check only the last secured superblock
af96ad07e4 tresor: don't halt on extension beyond limits

chelmuth, 2024-04-12:
What's the state of this issue?

chelmuth, 2024-04-19:
> What's the state of this issue?

Commits are merged. Can we close this issue?

m-stein, 2024-04-19:
Yes.

closed 2024-04-19 by m-stein