genodelabs/genode #5079: [pc_nic_drv] on ThinkCentre M73 : "Unable to load firmware ... rtl8168g-2.fw"

ttcoder, 2023-12-16:
On 23.05 networking does not work on my M73 so I upgraded to Genode 23.11 hoping the new dde_linux based code would support it.

Result : no change, in that my thinkpads still work great as ever, but the thinkcentre still does not work. It looks like this might be feasible to "fix" it though (?), can't say as I'm not very familiar with firmwares and such:

Didn't take a photo of the screen yet, but it looks somewhat like this:

```
[nic_drv] .... e1000
[nic_drv] .... realtek 8169 8168g/8111g
[nic_drv] realtek r8169 main 02:00:0 : Unable to load firmware rtl_nic/rtl8168g-2.fw (-1)
```

Not sure what is involved, but looking around I see that #4921 tracks the creation of the new driver and that bd553c81ad5 mentions something about firmware(s).

(the M73 is a tentative "reference platform" for our stations as it has the necessary ports and is ubiquitous and cheap in refurbished condition).

cnuke, 2023-12-18:
@ttcoder could you please provide the complete `[nic_drv] …` log snippet (redacting information like MAC addresses is completely fine).

During my testing with the realtek devices at hand usage of a an external firmware blob was optional - I am curious what kind of device you have where it is mandatory. As we already have support for loading firmware in the `wifi` driver adding it to the `pc_nic_drv` - or rather make it available for other drivers as well - is of course possible.

ttcoder, 2023-12-18:
[image]

ttcoder, 2023-12-18:
The previous run's photo is less blurry:

[image]

The log is partially cropped from the right ; tell me if you're missing info and I'll try to setup a higher screen resolution and wider window.

cnuke, 2023-12-19:
At least the first photo shows the device working up to the point where link up is detected. Although we cannot rule out problems resulting from being unable to load the firmware, for now I would consider this less likely.

I assume you are unable to obtain a IP address on this device. So as a first try, in case you acquire the IP information via DHCP and have some control over the server side, you could check if you spot DHCP requests coming from and offers being send to the device. If there are indeed requests coming through we know sending packets works and there is something wrong when receiving them.

If sending is working and receiving is not it could be similar to a problem we see with Intel-based NICs where the symptoms are the same when the cable is unplugged for a longer period of time. On the other hand we also encountered other devices were the use of MSIs is not supported and we forcefully have to use legacy IRQs. You can look into the `pci_decode` component and try to apply the same MSI treatment like was done for the other devices and see if that makes a difference.

If both sending and receiving are not working then we have to dig deeper.

ttcoder, 2023-12-21:
[image]

ttcoder, 2023-12-21:
Summary first:

1) outward ping (the one built with make app/ping): repeatedly says "snd ETH", but it should also say "rcv" maybe ?
2) inbound ping, from another computer on the LAN : "no route to host"
3) are there .run files that show how to tweak pci_decode for MSI ? That sounds promising

Details:
My router is an Orange lifebox so I have no control on how it does DHCP service. So I thought maybe it would help if I set a _static_ IP to go around the DHCP problems. I modified nic_router's config, domain "uplink" now has an address (192.168.1.50, outside of the DHCP range), and gateway (but no DNS) so that it won't send DHCP requests to my router.

Tested on my T410, seems ok. Well outbound "ping" stops after just one request (probably by design ?), but I can access a port on another computer (which runs "nc -l -p 80"), and other computers can 'ping' Genode.

So I used the same USB thumbdrive on the ThinkCentre M73, with the above results : outbound ping seems unconclusive?, inbound ping (from another host running "ping 192.168.1.50") is no-go.

Yesterday I also ran "grep -r msi", plus "msix", "legacy", and so on, to find .run files in genode and genode-world that would tweak legacy interrupts but came up empty. What would be a good term to use for grepping ?

cnuke, 2023-12-21:
>     1. outward ping (the one built with make app/ping): repeatedly says "snd ETH", but it should also say "rcv" maybe ? The 192.168.1.1 address is my router.
> 
>     2. inbound ping, from another computer on the LAN : "no route to host"
> 
>     3. are there .run files that show how to tweak pci_decode for MSI ? That sounds promising
> 
> […]

If you use a static configuration you could directly connect a second system and use wireshark or something similar on this system to capture the traffic and look for packets from the M73 (or put the system in a mode where it captures all packets flowing in your local network if setting up a direct connection is too cumbersome). You can also make the `nic_router` more verbose by setting `verbose_packets` in its '<config>' node to `yes` and look for packets handled by the `uplink` domain.

> Yesterday I also ran "grep -r msi", plus "msix", "legacy", and so on, to find .run files in genode and genode-world that would tweak legacy interrupts but came up empty. What would be a good term to use for grepping ?

I am afraid there is currently no way to apply such a quirk via config change but only by adding it to the component directly, see commit 0e82322.

ttcoder, 2023-12-27:
No luck when tweaking MSI/MSI-X : I added a commit similar to 0e82322 but targetting all devices of class "ethernet" (and with tracing to be sure it gets activated as it should) but no visible change.

I think there is a Be port of wireshark, I should probably look into that, to (IIUC) examine net traffic of my T410 from my T430 on the same LAN.

I've enabled `<domain name="uplink" verbose_packets="yes">` in the nic_router config, but no tracing appears on the M73 (by contrast, I get lots of tracing on my T410, for the initial DHCP negociation and all further traffic from applications). Does that mean pc_nic_drv refuses to talk to nic_router ?
See screenshot on M73:
[image]

cnuke, 2024-01-02:
> No luck when tweaking MSI/MSI-X : I added a commit similar to [0e82322](https://github.com/genodelabs/genode/commit/0e82322605664ab513a09f4dd07e0fb7ddb24f82) but targetting all devices of class "ethernet" (and with tracing to be sure it gets activated as it should) but no visible change.

Well, it was worth a try but a long shot anyway to be honest.

> I've enabled `<domain name="uplink" verbose_packets="yes">` in the nic_router config, but no tracing appears on the M73 (by contrast, I get lots of tracing on my T410, for the initial DHCP [...]

Not necessarily. You can also enable _verbose_domain_state_ in the `<config>` node that will show you the session state of each domain. There should be some message along the line `[nic_router] [uplink] NIC session: 1` shortly after `[nic_drv] create uplink for net device eth0`.

I prepared a debug commit [c70a3eb](https://github.com/cnuke/genode/commit/c70a3ebfd04058578a31996b65993b61818b45d6) that adds a `printk` call to the TX/RX/IRQ handlers (note that from the uplink's point of view TX/RX are swapped) of the driver.

ttcoder, 2024-01-03:
`verbose_domain_state="yes"` is done, or to be more precise, it's was already enabled for all the screenshots I posted above, as it's been part of this config since the beginning : https://chiselapp.com/user/ttcoder/repository/genode-haiku/file?udc=1&ln=on&ci=b2d4d377673a52189e58a801200984515be10c0e&name=genode-haiku%2Fjam%2Fjamrules-qemu-hog

So that's why I'm puzzled, I, too, though there would be a `[uplink] NIC session: 1` LOG message, like occurs on other computers I tested (ThinkPads)...

I'd like to try pc_nic_drv with the "printk" patch, but (assuming it builds similarly to usb_host) I'm unable to build dde_linux drivers due to #4648 :-/

I'm guessing the next step is to try out wireshark then.

chelmuth, 2024-01-04:
I suggest to switch to plain Sculpt for NIC debugging. That's a familiar scenario and gives @cnuke the chance to provide you with system images including tweaked components, i.e., pc_nic_drv incl. debugging output.

What puzzles me most is that nic_router does not respond to "create uplink..." with some state change. @m-stein do you know of possible reasons?

chelmuth, 2024-01-04:
Under Sculpt, you may have a look into */report/runtime/state* in Inspect or system-shell, and find `caps="300"` configured for nic_router.

ttcoder, 2024-01-04:
No change with 390 caps and 20M on nic_router.

In a few days I'll try again after setting 2300 caps and 100 M and reviewing the rest of the net chain.

A possibly important detail: in this run I had "verbose dropped packets" enabled, and it shows this:
`.. [default] dropped packet.. (gratuitous ARP request)`
Didn't take a screenshot as the rest is identical, esp. the last line.

That's consistent with my known-to-work T410, my router is known for generating that kind of requests.

So I suppose that means external traffic works, but it's internal RPC that causes trouble, which is unexpected.

m-stein, 2024-01-04:
@chelmuth If the NIC router is not run with ``<config verbose="yes">``, session requests may fail in ``Net::Uplink_session_root::_create_session`` without any error or state change. Possible reasons here:
* missing or malformed 'mac_address' argument in request args
* an exception while attaching the accounted RAM dataspace the new session will live in
* insufficient RAM/CAP quota donations for creating a session

These merely throw Service_denied or Insufficient_x_quota, which might be handled silently by the generic session request code as well.

Furthermore, if the session label doesn't resolve to any existing domain in the router, the new session might remain unattached without further notice, waiting for a suitable domain to attach to. The message ``[domain] NIC sessions: x`` is only printed when attaching to a domain (given ``verbose_domain_state="yes"``).

ttcoder, 2024-01-11:
On a hunch I also tried enabling tracing of acpi_drv and pci-decode, but nothing special in logging (that was with 100M and 2300 caps nic_router BTW).

I'm going to buy a PCI-express Nic card (intel E1000 type probably, seems most compatible with Genode).
Everything works so well on my T410, it seems like it all ought to work on the M73 as well... even if there are hints that the problem lies elsewhere than in hardware support.

I could also add tracing/instrumentation in nic_router at the point where it is supposed to log "uplink: nic sessions:".
Re-reading the above, sounds like I should add the "generally verbose" switch also, not just feature-specific verbosity.

ttcoder, 2024-01-13:
[image]

I'm back from the shop with a PCI-e Nic. Tested it to work on MS-Win, but it's a no-go on my distro (didn't try SculptOS yet), and this time there is no `create uplink` message at all.

The log now shows _two_ Realtek Nic's, one at 02:00.0 and one at 03:00.0 ; but then I'm realizing, perhaps Genode insists on using the first one, and ignores the fact that I plugged the Ethernet cable in the second Nic ?

I'll have to study the SculptOS deployment scripts, maybe I forgot something when upgrading to 23.08 and 23.11, like some of the iommu-related changes or such.

chelmuth, 2024-01-15:
> I'm back from the shop with a PCI-e Nic. Tested it to work on MS-Win, but it's a no-go on my distro (didn't try SculptOS yet), and this time there is no `create uplink` message at all.
> 
> Anyway, the log now shows _two_ Realtek Nic's, one at 02:00.0 and one at 03:00.0 ; but then I'm realizing, perhaps Genode insists on using the first one, and ignores the fact that I plugged the Ethernet cable in the second Nic ?

So you didn't go for a Intel NIC while shopping? Anyway, your scenario of two Ethernet NICs in pc_nic_drv is untested a may have surprising effects. I'd suggest to disable/remove the old card for testing.

> (edit) it's also curious that it says "drop gratuitous ARP request" even though it did _not_ claim to be online yet...

The dropped gratuitous ARP request was received in the *default* domain not on the uplink. I'd expect this message is part of all your logs.

ttcoder, 2024-01-16:
Went back to the shop and insisted on the chipset this time, they said they'd get back to me if they can source one with an Intel chipset and back-order it.

Removed the extra Nic and went back to the on-board one. Gotta be careful with this "family computer", but I did run an extra test and made a interesting observation : nothing happens on plug/unplug cycles. The LOG windows do not output anything, it's as if the ethernet cable didn't exist. That could be because the concerned components are "stuck" in some way.

Anyway I'm going to focus on the "what differs between SculptOS and my system ?" aspect, in the coming weeks.

ttcoder, 2024-02-08:
Finally cracked that nut : `pc_nic_drv` was being started too soon in the boot process.

If I move it into a `sequence` (adjusting the "policy" permission nodes accordingly in platform_drv and nic_router) with a /bin/sleep that delays it by a few seconds (similar to how SculptOS does not start networking immediately, it waits for the user to click the config buttons) then it starts up nicely. It takes a dozen seconds to get a DHCP lease, and then I'm able to do outbound connnections for FTP and Genode Depot packages, inbounds connections as FTPd and SFTPd, and so on. (thus I was *also* able to transfer mp3 files and confirm that the ThinkCentre M73 also has Audio-out support in Genode). Life's good. Will send a new test image to my tester soon.

Closing.

closed 2024-02-08 by ttcoder