genodelabs/genode #5081: gpu/intel: support to release/acquire pci device before/after ACPI suspend/resume

alex-ab, 2023-12-20:
Some of our test notebooks, e.g. T470p and T490, do successfully suspend and resume, but without working display driver afterwards. It turns out, restarting also the GPU driver after resume, lead to better results, e.g. T470p with working external HDMI output and on T490 with working internal and external output. However, restarting the GPU driver is no real option, e.g. for Sculpt OS or other solutions, because by restarting the GPU driver we lose all GPU clients with restarts, e.g. vbox which may use the GPU, some native browsers and ported games.

Instead the GPU driver must be kept alive and instead the required platform resources must be shut down gracefully and resumed after ACPI resume.

alex-ab, 2023-12-20:
The issue_5081 branch contains the current working state. The both first intel/gpu commits are intended as pure refactor work, without removing or adding any functionality, in order to be able in the last commit actually to release/acquire the device. @ssumpf , @cnuke  may you please have a look, especially on the refactor commits. I hope I did not missed any substantial.

cnuke, 2023-12-21:
@alex-ab thanks for the commits, they look very good to me and at first glance I did not spot anything concerning. The `if (with([]{})) …` pattern appeared somewhat alien initially as I am not used to that.

ssumpf, 2023-12-21:
@alex-ab: Looks good to me also. Please test the Tigerlake (GEN12) GPU also. Doom3 is additionally a good test, for example, @cnuke  can show you how to set it up.

alex-ab, 2024-01-10:
I tested now the refactor commits successfully also on Tigerlake test machine and got the game also running on several machines. Also forcefully restarting the gpu client in middle of operation let continue the GPU driver so far.

If possible, I would appreciate when the refactor commits (the ones without WiP) may be considered for staging to avoid rebasing hassle/headache since they are quite extensive, esp. in event of upcoming #4081.

nfeske, 2024-01-11:
Thanks for the notice, @alex-ab. I merged the three commits 16d38b539d8eef40ef475f11aed48afd7ad0c904, e04bf91412374bb65f99adf507d2a17589899959, and e779c4fc1074f8570a86453431dd23503b13ae64 to staging now.

BTW, I like the gpu/intel patches quite a lot. Thanks for paying so much attention to the details also (like tightening const-ness or improving the file headers). It is good to see the much of the internal state gone.

alex-ab, 2024-01-25:
I updated to current staging and finalized the GPU driver and the test case to working on X201 and T490.

alex-ab, 2024-02-20:
I rebased the remaining gpu commit to staging and added another one to replace the bool with_(fn) -> in with_(fn, fn_errro)

commit 20e29aee (2024-02-21): gpu/intel: release device before suspend

- monitor system ROM changes
- stop processing of new Jobs before suspend
- destruct platform device before suspend, but keep platform DMA buffers
- re-construct platform device and reinit resources (mmio, irq) on resume
- re-start GPU job scheduling on resume

Fixes #5081
commit 0684101f (2024-02-28): gpu/intel: release device before suspend

- monitor system ROM changes
- stop processing of new Jobs before suspend
- destruct platform device before suspend, but keep platform DMA buffers
- re-construct platform device and reinit resources (mmio, irq) on resume
- re-start GPU job scheduling on resume

Fixes #5081