genodelabs/genode #5084: tool/dde_linux/create_dummies: broken environment

skalk, 2024-01-04:
Recently, the tool `create_dummies` got altered by @ssumpf to make it useable for GNU Make 4.4. The environment gets completly reset before calling a sub-make. Unfortunately this breaks the tool for me (GNU Make 4.3), because the sub-make does not find the given target anymore.

skalk, 2024-01-04:
@ssumpf could you give commit 692005d a try, whether it still works in your environment?

chelmuth, 2024-01-04:
I picked the commit for staging too to test it in our development environments.

ssumpf, 2024-01-04:
> @ssumpf could you give commit [692005d](https://github.com/genodelabs/genode/commit/692005de5d6b1469221d69ddada14bd4cd6da7b6) a try, whether it still works in your environment?

@skalk: Will do. What is a little strange is that I have tested with both Make versions.

ssumpf, 2024-01-04:
> @ssumpf could you give commit [692005d](https://github.com/genodelabs/genode/commit/692005de5d6b1469221d69ddada14bd4cd6da7b6) a try, whether it still works in your environment?

@skalk: Works with 692005d and Make 4.3/4.4 here. Also works without 692005d and Make 4.3/4.4 here. So if 692005d fixes something on your system please go ahead.

closed 2024-02-29 by chelmuth