genodelabs/genode #5087: add FindXxx.cmake files to api archives

jschlatow, 2024-01-05:
As discussed in genodelabs/goa#52, the cmake-support for Goa requires certain FindXxx.cmake files. Instead of hosting these files statically in the Goa tree (thereby making all libraries available by default), I'd rather like to add these to the related api archives. This way, the Goa user gets correct hints from cmake w.r.t. what archives need to be added to the _used_apis_ file.

jschlatow, 2024-01-05:
Commit 32975a4 adds those FindXxx.cmake files that have been so far identified by @cnuke, @ssumpf and myself. I will create a separate issue for api/sdl2 in genode-world.

@cnuke, @ssumpf: May you try the modified api archives (available in my depot as version 2024-01-05) with your yquake2 resp. dhewm3 port and the current goa staging branch? The corresponding sdl2 archives are available on my depot as version 2023-12-15-b.

ssumpf, 2024-01-05:
@jschlatow: I will need to move the dhewm3 project to a current version of Goa in the first place, which with the new SDL2 improvements should be possible. Also dhewm3 requires OpenAL. So I still have some work to do before I can test this, but I put it on my TODO list.

cnuke, 2024-01-08:
@jschlatow thanks, works fine with my projects :+1:.

nfeske, 2024-02-29:
Merged in master.

closed 2024-02-29 by nfeske