genodelabs/genode #5089: nic_drv: 'kvfree_call_rcu' not implemented

cproc, 2024-01-08:
When I tried to run Sculpt on a PC with an ASUS Q170T main board, which has both an Intel and a Realtek NIC, the following error occured:

```
[init -> runtime -> nic_drv] NR_IRQS: 4352, nr_irqs: 4352, preallocated irqs: 256
[init -> runtime -> nic_drv] time-clocksource: dde_counter: mask: 0xffffffffffffff max_cycles: 0x1d854df40, max_idle_ns: 3526361616960 ns
[init -> runtime -> nic_drv] time-clocksource: jiffies: mask: 0xffffffff max_cycles: 0xffffffff, max_idle_ns: 7645041785100000 ns
[init -> runtime -> nic_drv] time-clocksource: Switched to clocksource dde_counter
[init -> runtime -> nic_drv] e1000-e1000_main: Intel(R) PRO/1000 Network Driver
[init -> runtime -> nic_drv] e1000-e1000_main: Copyright (c) 1999-2006 Intel Corporation.
[init -> runtime -> nic_drv] e1000e-netdev: Intel(R) PRO/1000 Network Driver
[init -> runtime -> nic_drv] e1000e-netdev: Copyright(c) 1999 - 2015 Intel Corporation.
[init -> runtime -> nic_drv] e1000e 00:1f.6: Interrupt Throttling Rate (ints/sec) set to dynamic conservative mode
[init -> runtime -> nic_drv] e1000e 00:1f.6 00:1f.6 (uninitialized): Failed to initialize MSI interrupts.  Falling back to legacy interrupts.
[init -> runtime -> nic_drv] e1000e 00:1f.6 00:1f.6 (uninitialized): Reset blocked by ME
[init -> runtime -> nic_drv] e1000e 00:1f.6: PHY reset is blocked due to SOL/IDER session.
[init -> runtime -> nic_drv] e1000e 00:1f.6 eth0: (PCI Express:2.5GT/s:Width x1) 34:97:f6:db:97:5f
[init -> runtime -> nic_drv] e1000e 00:1f.6 eth0: Intel(R) PRO/1000 Network Connection
[init -> runtime -> nic_drv] e1000e 00:1f.6 eth0: MAC: 12, PHY: 12, PBA No: FFFFFF-0FF
[init -> runtime -> nic_drv] realtek-r8169_main 03:00.0: unknown chip XID d8f, contact r8169 maintainers (see MAINTAINERS file)
[init -> runtime -> nic_drv] Error: Function kvfree_call_rcu not implemented yet!
[init -> runtime -> nic_drv] Backtrace follows:
[init -> runtime -> nic_drv] backtrace "ep"
[init -> runtime -> nic_drv] Will sleep forever...
```

cproc, 2024-01-08:
Fixed by a608b1f (implementation taken from a630563).

chelmuth, 2024-01-09:
@cproc would you mind to consolidate this implementation in *dde_linux/src/lib/lx_emul/shadow/kernel/rcu/tree.c* and adapt the drivers?

It may be worthwhile to comply to the conditional calculation of `ptr` in https://elixir.bootlin.com/linux/v6.1.71/source/kernel/rcu/tree.c#L3344

cproc, 2024-01-10:
Commit 86cbaa2 adds `kvfree_call_rcu()` to the lx_emul lib.

chelmuth, 2024-01-10:
Great! I merged all three commits.

commit 86cbaa20 (2024-01-10): lx_emul: add kvfree_call_rcu

Fixes #5089
commit 3062b7d7 (2024-02-26): lx_emul: add kvfree_call_rcu

Fixes #5089
closed 2024-02-29 by chelmuth