genodelabs/genode #5090: Draft: I2C: allow selection of communication speed per device on the bus

trimpim, 2024-01-08:
I stumbled upon this when implementing a VFS-Plugin to read the content of I2C at23xxxx EEPROMS.

 - Currently only the driver for the i.MX8mm plugin implements the actual setting this.
   - we currently host this in a separate git repository
   - I'm thinking to give this upstream as well once I had the time to ensure it works on other platforms
 - The API change shouldn't impact other drivers/run scripts

Suggestions for improvements are welcome.

chelmuth, 2024-01-23:
After looking at your patch I got the feeling *bus speed* is a property of the bus driver not each transmission. Is this correct? And if yes, wouldn't it be better to instantiate the driver with the requested speed than parameterizing each transmission?

trimpim, 2024-01-25:
The !2C driver for the iMX devices sets the clock divisor which determines the speed on each start of transmission. This is the reason for my current implementation. 

A quick look through the different drivers in the Linux sources revealed to me, that some (maybe most) drivers set the bus speed at probe time.

I will change the patch to represent this new insight.

chelmuth, 2024-01-25:
Many thanks for reconsidering the bus configuration mechanics. May it make sense to provide `bus_speed` as config attribute/node of the driver component if it is not changeable by sessions?

trimpim, 2024-01-25:
Yes, this is my plan for the change. The attribute will move to the `<config>` node.

chelmuth, 2024-01-26:
I merged 41adae3636a428ddbd04f46bd12fcbdb2b991241 to staging. Do I understand correctly that *i2c_mcp9808.run* still uses default bus speed (and therefore needs no config attribute)?

@skalk please have look

trimpim, 2024-01-26:
If no bus speed is configured, the `imx8q_evk_i2c_drv` uses the default value of 400kHZ as it did before.

skalk, 2024-01-29:
I'm fine with the commit. Thanks for the contribution.

closed 2024-02-29 by chelmuth