genodelabs/genode #5092: sculpt: preset for turning Sculpt into a Goa test target

nfeske, 2024-01-15:
To streamline Goa's now `--target sculpt` features (https://github.com/genodelabs/goa/issues/77), Sculpt should offer a ready-to-use preset that hosts the goa testbed out of the box.

nfeske, 2024-01-15:
The commits above provide the goa_testbed preset for the PC and PinePhone versions of Sculpt OS.

cproc, 2024-01-23:
I get the following error with `sculpt_test.run` for PC on the current staging branch, which seems to be related to commit 0c75f77:

```
can't read "api_name": no such variable
    while executing
"_depot_contains_archive $user api $api_name $api_version"
    (procedure "_collect_src_archive_from_depot" line 28)
    invoked from within
"_collect_src_archive_from_depot $user $name $version"
    (procedure "_collect_from_depot" line 50)
    invoked from within
"_collect_from_depot $archive"
    (procedure "_collect_pkg_archive_from_depot" line 21)
    invoked from within
"_collect_pkg_archive_from_depot $user $name $version"
    (procedure "_collect_from_depot" line 49)
    invoked from within
"_collect_from_depot [pkg_archive_paths [referenced_pkg_values]]"
    (file "/.../repos/gems/run/sculpt.run" line 931)
    invoked from within
"source ${genode_dir}/repos/gems/run/sculpt.run"
    (file "/.../repos/gems/run/sculpt_test.run" line 3)
    invoked from within
"source $include_name"
    ("foreach" body line 6)
    invoked from within
"foreach include_name [get_cmd_arg --include ""] {
        # first check if the include name is absolute
        if {[string first "/" $include_name] == 0} {
                puts ..."
    (file "/.../tool/run/run" line 1201)
Makefile:445: recipe for target 'run/sculpt_test' failed
make: *** [run/sculpt_test] Error 1
```

nfeske, 2024-01-23:
I can only guess that the effect is indirectly caused by the addition of the concrete "jschlatow/pkg/system_info/2024-01-04-a" pkg in the preset. It probably does not occur when specifying `DEPOT=omit`, or does it? Anyway, I'd appreciate you looking into the error, e.g., by instrumenting depot.inc, eventually leading us the assumption that makes the run tool stumble?

jschlatow, 2024-01-23:
This is probably related to genodelabs/goa#72.

cproc, 2024-01-23:
Yes, it was caused by the full api paths. Commit 9cb7d66 fixed the problem.