genodelabs/genode #5093: lighttpd: enable mod_webdav

jschlatow, 2024-01-15:
Thanks to @cnuke's preliminary work on enabling mod_webdav (and thereby HTTP PUT) for lighttpd, I was able to create the goa_testbed runtime and eliminate the remaining issues w.r.t. uploading large files.

jschlatow, 2024-01-15:
I pushed the commit series to my 'issue5093_mod_webdav' branch.

I am also using this issue to piggyback the tcp_terminal recipe, which is required by goa_testbed.

gstrauss, 2024-02-03:
I am a lighttpd developer and am curious about a few things in the patches.

Does Genode OS return EAGAIN or EWOULDBLOCK if a file is opened O_NONBLOCK?  Why do the patches comment out O_NONBLOCK in lighttpd `src/stat_cache.c`?  (or is that patch now removed?)  lighttpd opens files O_NONBLOCK to avoid blocking on POSIX mandatory file locks or FIFOs, neither of which are expected in normal operation.

Does Genode OS not support FD_CLOEXEC?  Why do the patches `#if 0` that in lighttpd `src/fdevent.c`

Does Genode OS not support POSIX.1-2008 AT extensions?  Why do the patches `#undef AT_FDCWD` and `#undef _ATFILE_SOURCE` in lighttpd `src/mod_webdav.c`?

lighttpd build supports autoconf, CMake, SCONS, and meson.  Genode OS files might leverage those build scripts for static builds rather than manually duplicating flags.  https://wiki.lighttpd.net/InstallFromSource

jschlatow, 2024-02-05:
> Regarding your first question w.r.t. `O_NONBLOCK`. We actually return `EWOULDBLOCK` (I suspect because we first implemented this for sockets). I just learned that `EAGAIN` would be correct for regular files. Thanks for your hint!

On a second thought, our libc (based on FreeBSD libc) actually defines `EWOULDBLOCK=EAGAIN` so this shouldn't make a difference. I removed `O_NONBLOCK` because the write operation to the temporary file (during upload) stalled under certain circumstances and this caused a non-recoverable error in lighttpd.

cnuke, 2024-02-05:
> > Does Genode OS not support FD_CLOEXEC? Why do the patches #if 0 that in lighttpd src/fdevent.c

> > Does Genode OS not support POSIX.1-2008 AT extensions? Why do the patches #undef AT_FDCWD and #undef _ATFILE_SOURCE in lighttpd src/mod_webdav.c?

> Unfortunately, I have no proper answer for your other patch related questions since these have been added by @cnuke. @cnuke would you share your insights?

As you already mentioned our POSIX support is limited and not all features have been implemented. `FD_CLOEXEC` and the *AT* extensions are such cases were the support is currently missing, hence the patches that omit their usage.

gstrauss, 2024-02-05:
> I removed `O_NONBLOCK` because the write operation to the temporary file (during upload) stalled under certain circumstances and this caused a non-recoverable error in lighttpd.

On many *nix filesystems, O_NONBLOCK affects open() blocking on FIFOs and mandatory file locks, but filesystem operations such as `write()` may still block.  lighttpd does not expect to handle writing to files resulting in EAGAIN or EWOULDBLOCK, and looping there would result in spinning on CPU until the `write()` progresses.  lighttpd handles deferred writes to most sockets used by lighttpd, but lighttpd does not handle deferred writes to files.

Does the compiler on Genode OS have any preprocessor defines that lighttpd could use -- e.g. `#ifdef __Genode__` or something similar -- to identify Genode OS?  If so, I could add exclusions in lighttpd to either avoid O_NONBLOCK for files opened O_RDWR or O_WRONLY, or to unset O_NONBLOCK after open() for O_RDWR or O_WRONLY, to preserve the open() with O_NONBLOCK protections against malicious local users.

Might Genode OS return EAGAIN or EWOULDBLOCK for reads from files opened O_NONBLOCK?  If so, then after open(), lighttpd should remove O_NONBLOCK from files opened O_RDONLY, too.

You might consider repeating
`+#undef AT_FDCWD`
`+#undef _ATFILE_SOURCE`
in mod_dirlisting.c, similar to your patch for mod_webdav.c
(It would be better if Genode OS simply did not define AT_FDCWD or _ATFILE_SOURCE if not implemented.)

Also, FD_CLOEXEC is used in server.c and configfile.c, in addition to fdevent.c
(It would be better if Genode OS simply did not define FD_CLOEXEC if not implemented.)

FYI: building lighttpd with mod_mbedtls or mod_wolfssl would lead to smaller execuables than building with mod_openssl, though mod_openssl and mod_gnutls are faster

repos/ports/src/app/lighttpd/target.inc can remove from FILTER_OUT files that are not longer part of lighttpd src/
              mod_cml.c
              mod_cml_funcs.c
              mod_cml_lua.c
              mod_compress.c
              mod_evasive.c
              mod_flv_streaming.c
              mod_geoip.c
              mod_mysql_vhost.c
              mod_secdownload.c
              mod_ssi_expr.c
              mod_ssi_exprparser. 
              mod_trigger_b4_dl.c
              mod_uploadprogress.c
              mod_usertrack.c

repos/ports/src/app/lighttpd/target.inc can remove the following from FILTER_OUT and add to repos/ports/src/app/lighttpd/plugin-static.h to be included in lighttpd without adding additional dependencies:
              mod_alias.c
              mod_evhost.c
              mod_expire.c
              mod_fastcgi.c
              mod_redirect.c
              mod_rewrite.c
              mod_scgi.c
              mod_setenv.c
              mod_simple_vhost.c
(though if you do not build with pcre2 or with pcre, then mod_rewrite and mod_redirect will have basic functionality, but without regexes)

gstrauss, 2024-02-05:
If FD_CLOEXEC is not support in Genode OS, then instead of all the `#if 0` patches, why not `#undef FD_CLOEXEC` and `#undef F_DUPFD_CLOEXEC` in fdevent.c, configfile.c, and server.c after `#include <fcntl.h>` in those files?

nfeske, 2024-02-29:
All commits are part of Genode 24.02.