genodelabs/genode #5095: virt_linux: add support for the arm_v6 architecture

ssumpf, 2024-01-19:
* enable arm_v6 (used on the RPI1) in virt_linux
* split spec `arm` into `arm_v6` and `arm_v7` where necessary
* add `arm_v6` to USB HID and NIC drivers
* rename spec `arm_64`  to `arm_v8` where appliable

ssumpf, 2024-01-22:
The commits above (branch issue_5095_5096_initcalls_arm_v6 in my genode, genode-imx, genode-allwinner, genode-world repos) adjust the specs of _dde_linux/lib/mk_ from _arm_ and _arm_64_ to _arm_v6_, _arm_v7_, _arm_v8_. I did hopefully identify all occurrences where changes where necessary. For `virt_linux` the _arm_v6_ architecture is added with adjustments required in the `usb_net` and `usb_hid` drivers. Additionally, for the Raspberry PI1 the SMSC95xx drivers has been added to `usb_net`.

[2218bb1](https://github.com/ssumpf/genode-world/commit/2218bb121a0b59caa1fccd6258ec74f64d3d2a2c) and  [afd4a49](https://github.com/ssumpf/genode/commit/afd4a4903a9795e1e24fa3cc370964ffe7daef80) remove files I identified on the way as unused artifacts.

skalk, 2024-01-23:
@ssumpf thank you for doing this (kind of annoying) cleanup work, I'm fine with the commit series.

chelmuth, 2024-01-23:
@ssumpf please rebase on current staging - I get conflicts.

ssumpf, 2024-01-23:
> @ssumpf please rebase on current staging - I get conflicts.

Will do.

ssumpf, 2024-01-23:
@chelmuth: I have rebased my _issue_5095_5096_initcalls_arm_v6_ branch to staging.

chelmuth, 2024-01-24:
Thanks, merged to staging.

Regarding *sorting* please see dfa9c985a8fcbf4eeae2153478be5b2d333b0208 and fdc8f111eaf618095273855f159cdd42cbbb8eee. I triple checked and are sure my changes are correct. Strong indicator: resorting changed *repos/dde_linux/src/drivers/usb_hid/spec/arm_v6/source.list* but not *repos/dde_linux/src/drivers/usb_hid/spec/arm_v7/source.list* on my side.

ssumpf, 2024-01-24:
@chelmuth: Okay and thanks.

chelmuth, 2024-01-25:
> Thanks, merged to staging.

I missed to update genode-allwinner, genode-imx, and genode-world yesterday. Sorry for the noise.

chelmuth, 2024-01-25:
depot archive bin/x86_64/wireguard does not build

```
    LINK     wireguard
/usr/local/genode/tool/23.05/bin/../lib/gcc/x86_64-pc-elf/12.3.0/../../../../x86_64-pc-elf/bin/ld: /data/depot/checker/bin/x86_64/wireguard/2024-01-24.build/var/libcache/wireguard_lx_inc_dirs
/data/depot/checker/api/virt_linux/2024-01-24/src/linux/kernel/sched/build_utility.c:239: undefined reference to `cpu_sibling_map'
/usr/local/genode/tool/23.05/bin/../lib/gcc/x86_64-pc-elf/12.3.0/../../../../x86_64-pc-elf/bin/ld: /data/depot/checker/bin/x86_64/wireguard/2024-01-24.build/var/libcache/wireguard_lx_inc_dirs
```

ssumpf, 2024-01-25:
@chelmuth: I have push 1a79f0e as a fixup for wireguard to my staging branch.

chelmuth, 2024-01-25:
@ssumpf test/driver_time bails out with the same error for some reason...

ssumpf, 2024-01-25:
> @ssumpf test/driver_time bails out with the same error for some reason...

@chelmuth: Seems like a forgot the commits for wireguard and driver_time, since they will be adapted with lxip again. Will do.

ssumpf, 2024-01-25:
@chelmuth: d540b93 on my staging branch fixes the driver_time test.